samtap/fang-hacks #91: New version of Fang Camera?

eterpstra, 2017-04-06:
Hi all,

Today I received my second camera. But is is a slighty different. The reset button has changed from a small button to a clip what you need to press. Also the blue led keeps blinking and doenst stop.

Therefore I have not managed to install the hack. I've tried it but with no succes. When i insert the SD card, i here the same sound with my other camera (probably that the sdcard is mounted?)

But no luck signing in to the webpage.

http://192.168.0.178/cgi-bin/status gives error:

404 Not Found
The requested URL /cgi-bin/status was not found on this server.

http://192.168.0.178/cgi-bin/hello.cgi gives:

System Volume Information
time_lapse
record
log_34CE00D52300_s.txt

Anyone who can confirm this?

eterpstra, 2017-04-07:
anyone who can confirm this?

samtap, 2017-04-07:
Haven't heard any other reports, but perhaps you're the first to try on this new version? Can you find and post the firmware version? 

Based on the hello.cgi output it appears the snx_autorun.sh and bootstrap folder are wiped, but the "System Volume Information" also tells me the sdcard has been used in a Windows PC, maybe you formatted it by accident, or the new version auto-formats sdcards when you insert them...?

eterpstra, 2017-04-07:
Hmm I will test it tomorrow.. The firmware is the same as my other cam. 3.0.3.56. 

The strange thing is that the blue led keeps flashing...but the camera is working fine in the MI home app. 

I will try to format the SD card again tomorrow and reset the camera go factory defaults. Many this helps..

eterpstra, 2017-04-08:
Tried to reset the camera but the blue led keeps flashing.. The camera is working though in the mi home app...

Verry strange.

danzig666, 2017-04-12:
I just got the same new version of the camera. 
The hack works just fine, the blue led keeps flashing though. 
[image]

eterpstra, 2017-04-12:
Strange... Dont know what I am doing wrong then... :S

danzig666, 2017-04-12:
Did you try it with another sd card?

eterpstra, 2017-04-12:
Yeah I tried 2 SD cards...  Wel.. maby reset it again :)

eterpstra, 2017-04-12:
You waited untit the camera keeps flashing blue and insert the SD card?

danzig666, 2017-04-12:
I just inserted the card, didn't care for the led.

absinte, 2017-04-19:
I also have the new version, the led is still blinking, the SD card mounted like @eterpstra but 404 error if I check the status Page.
@danzig666 You did no do anything just as if it was the old one ?

Maxxxel, 2017-04-20:
Also got the "new" Version, Blue light keeps blinking, i can access the Cam with Mi Home App.
I insert SD card but nothing happens.

404 on Status Page and Hello.cgi shows "OpenDir faile"

danzig666, 2017-04-20:
I used Win32 Disk Imager too

samtap, 2017-04-20:
Based on all comments, I think it's safe to conclude the new version works identically.
I'm keep seeing the same stuff that people run into, so please consider doing a writeup on the wiki when you hit certain issues so others can benefit as well. I'm trying to do my part i.e. most common pitfalls are mentioned in FAQ.

closed 2017-04-20 by samtap