samtap/fang-hacks #96: Turn Off IR LED

ageorgios, 2017-04-12:
How can I turn off the IR LED?

JordyGitHub, 2017-04-12:
I believe you can stop it by going to http://<ip>/cgi-bin/scripts and stop/disable ir-control

ageorgios, 2017-04-12:
I stop/start the 21-ir-control	but the IR LED is not switching

mhdawson, 2017-04-19:
Was it dark when you stopped the service ?  I'd try:
* rebooting the camera
* making sure the camera is in a brightly light spot
* verifying the ir led is off (which it should be when the camera is in bright light)
* use the scripts management page to disable the ir control

Otherwise if you can ssh to the camera you could try running the the same logic as the ir script that controls the led/filter which seems to be:

```
gpio_ms1 -n 2 -m 1 -v 1 # filter movement enabled
                        gpio_aud write 1 0 0    # disable ir led and latch the filter in the correct position
                        gpio_ms1 -n 2 -m 1 -v 0 # filter movement disabled
                        echo 0x40 > /proc/isp/filter/saturation
```
from `/media/mmcblk0p2/data/usr/bin/fang-ir-control.sh`

bbbenji, 2017-04-19:
Please delete your duplicate post. https://github.com/samtap/fang-hacks/issues/97

samtap, 2017-04-20:
Closing, see #99

closed 2017-04-20 by samtap

vedranius, 2021-03-31:
The actual solution haven't be provided.
I have the similar problem.
I want to Xiaofang camera to totally disable nightvision mode, and that it doesn't even try to get into it.
And I can't find a way to do that permanently.
I've disabled the script for infrared but that works only when it's day and camera is turned on then. But if camera has been restarted when it's dark then it still goes in infrared mode.
Any solution to this?
Removing IR LEDs manually? Or there's a script/hack change that I can do in a code? 
Tnx!