samtap/fang-hacks #97: Turn off status LED

ageorgios, 2017-04-12:
Is there a way to turn off the status LED?
so as to have a completely dark camera?

bbbenji, 2017-04-19:
Either turn off IR or physically disable the red LED's.

ageorgios, 2017-04-20:
Thanks of answer, I mean the status orange led that is turned on when the camera is plugged in.

samtap, 2017-04-20:
I was able to control the orange LED with some builtin command (forgot the name) that allows to enable/disable/blink the LED, but was anyone able to control the blue LED?

demod-au, 2017-05-19:
I found this the other night can be done if you compile a tool, or easier -  can be done via the command line:  
echo 1 > /sys/class/gpio/gpio4/value - will turn blue "wifi" led off (its reversed for some reason)
echo 0 > /sys/class/gpio/gpio4/value  - will turn it back on

davidjb, 2017-08-19:
Commands from @demod-au didn't work for me on 3.0.3.56 (`/sys/class/gpio/gpio4` doesn't exist), but this did the trick:

```
# Turn off
gpio_led -n 3 -m 1 -v 0

# Turn on
gpio_led -n 3 -m 1 -v 1
```

These commands output a variety of `sh: write error: Device or resource busy` errors after the first time I run `gpio_led`, but the LED still turns on/off fine.  There's also the `gpio3_blink` command, which runs the above intermittently, and is how I found the commands.

samtap, 2017-08-19:
The gpio has to be exported first. I've created a small tool the manage the led.

[xf_led_ctl.zip](https://github.com/samtap/fang-hacks/files/1236299/xf_led_ctl.zip)
```
Usage: ./xf_led_ctl

XF LEDs Control creates a named pipe '/tmp/xf_ledctl', LEDs can be controlled by writing a string "MODE/DELAY" to this pipe
 MODE: 0=off, 1=orange, 2=blue, 3=both
 DELAY (optional): delay in milliseconds (default: 0, no blinking)

Examples:
 Disable both leds:
  echo "0" > /tmp/xf_ledctl
 Blink orange led, every 500ms:
  echo "1/500" > /tmp/xf_ledctl
 Enable blue led without blinking:
  echo "2" > /tmp/xf_ledctl
 Blink alternate blue/orange, every second:
  echo "3/1000" > /tmp/xf_ledctl
```

closed 2018-03-14 by ageorgios