samtap/fang-hacks #98: crontab

ageorgios, 2017-04-12:
since there is no crontab, is there a way to schedule a daily reboot?

bbbenji, 2017-04-19:
Why would you want to reboot it daily?

samtap, 2017-04-20:
I'll enable cron in busybox, should be straightforward.

thanme, 2017-05-11:
I've just finished messing around with this :)

I basically created a startup script for crond (called 03-crond)

The start bit looks like this:

start()
{
  echo "Starting crond..."
  mkdir -p '/var/spool/cron/crontabs'
  ln -s /media/mmcblk0p2/data/etc/crontab /var/spool/cron/crontabs/root
  crond
  echo "$!" > "$PIDFILE"
}

Then you put what you want in /media/mmcblk0p2/data/etc/crontab

Just note that anything you update with crontab -e won't save on reboot because it recreates  /var/spool/cron/crontabs/root on every edit. There's probably a work around, but this works just fine :) I suppose you could copy the crontabs/root file back on stop or something.

closed 2018-03-14 by ageorgios

wiziwk, 2018-03-24:
@samtap  I am new to busybox. Could you elaborate how? Tq.

Powerbless, 2019-11-27:
> I've just finished messing around with this :)
>
> I basically created a startup script for crond (called 03-crond)
>
> The start bit looks like this:
>
> start()
> {
> echo "Starting crond..."
> mkdir -p [...]

thank. PLEASE NOTE that you must add a "start" at the end of your script(without "") else the command would not start. And after that you have to
chmod -R 777 /media/mmcblk0p2/data/etc/scripts/03-crond