samtap/fang-hacks #99: IR LED on demand

ageorgios, 2017-04-15:
Is there a way to turn on the IR LED at night only when there is a demand for rtsp feed and when nobody is connected to be turned off?
It would be great for baby monitoring.

bbbenji, 2017-04-19:
Not at the moment. Maybe you can write something like that:) Although I don't see why you are looking for such a function.

ageorgios, 2017-04-20:
There is a discussion whether it is ok to have IR LED turned on facing sleeping babies.
I dont know how can I check from a bash script if there is a rtsp client connected? If I knew I could try to write the script to control the IR LED according to if there are connected clients.

samtap, 2017-04-20:
I feel there should be some options in the webif to control the IR LED behaviour, but I think the script you're describing is very specific to your usecase. The image is setup so anybody can add their own scripts, if you come up with something cool that doesn't interfere with other features I'd be happy to add it to the image so others can benefit. Just create a PR on Github when you have something to share, instead of opening issues ;-).

You should be able to monitor the rtsp-server log to determine if a client is connected. Look at the [fang-ir-control.sh](https://github.com/samtap/fang-hacks/blob/master/data/usr/bin/fang-ir-control.sh) script for the commands to control the led. Also make sure your script doesn't use any bash-isms (device runs ash, the almquist shell).

mhdawson, 2017-04-30:
Related to https://github.com/samtap/fang-hacks/issues/87 I am planning to do something along the same lines in terms of IR control.  I'll want the IR LEDs to only be one while pictures are being taken.

Venex, 2017-05-17:
I would also like this, either automatically when starting the stream, or by switching a button (is such a thing available for TinyCam?)
The cloud server does support this 'kind of'. You can turn the device on and off using the Mi Home app, when off, the IR disables

closed 2018-03-14 by ageorgios