samtap/fang-hacks #100: Is this possible to run the base linux and install the driver?

sequencer, 2017-04-18:
I don't wanna get my device get the xiaomi's system in it :(
so it there a uboot and bootstrap to startup a linux, and add drivers to it?

bbbenji, 2017-04-19:
If you are worried about data being sent to Xiaomi, please disable the cloud services.

Also please close the duplicate post you made.

sequencer, 2017-04-20:
Sorry, #101 must be the problem of the network.
I would like to compile my own program and use the full tool-chain in xiaofang.
I hope to build a firmware like openwrt to totally be under control.
Is this possible.
I have didn't seen any SDKs in Google for this device.

ZeroPoints, 2017-04-20:
Check this thread for SDK and a Fedora distro already setup for compiling for this environment
https://github.com/fritz-smh/yi-hack/issues/118

sequencer, 2017-04-20:
Alright, thank you!

samtap, 2017-04-20:
Please use #63 to discuss, or better: create a fork since it is currently not in my goals to create a flashable firmware. Once you have something to share we can merge.

closed 2017-04-20 by samtap