samtap/fang-hacks #102: SDCard during boot. blocks booting with 3.0.3.56

SanderSchutten, 2017-04-18:
Hi.,

I've just recently bought two Xiaofang camera's. I really like them, especially because of it's low price.
I think more tweaking can be done to make the device even more powerfull.

I've read that one should should not boot with the sdcard in it. It indeed blocks the booting of the camera. Booting without SDCard and afterwards placing the SDCard does work. 

But i don't want to eject the SDCard everytime the device boots. Especially i will mount the device somewhere out of my reach. I started this thread to eventually resolve this issue with the sdcard blocking the boot.

I suspect that the following reasons can might be blocking the booting with an SDCard:
* In the bootloader there is a check for an SDCard and just hang. As this issue didn't happen before. It probably is caused by updating the firmware to 3.0.3.56.
* With the SDCard it tries to boot from the SDCard.

Does anybody know which bootloader is used by the camera?
Does anybody know which device is normally used by the camera to boot from?

I've created dump files of the mtdblocks 0 through 5 with:
dd if=/dev/mtdblock0 of=/media/mmcblk0p2/backup/dev_mtdblock0.img bs=4096
I'm hoping that one of these devices is the boot-device. As then maybe we would be able to add an extra partition for booting of the sdcard.

bbbenji, 2017-04-19:
You only need to boot without SD the first time, before enabling hacks. After which it will boot and use the SD just fine without having to remove and reinsert it.

SanderSchutten, 2017-04-19:
Thank you for your reply..
That's strange that with my sdcard and camera combination it doesn't work like that. Everytime when booting with the SDCard in it, it halts with:
* no wireless connectivity
* the blue led continuesly on (does not blink).

I've used the SDCard image v0.2.0 from this repository with a 16GB SDcard. I didn't enlarge the partion yet. Did you enlarge the partion? Are you also using the image v0.2.0 ?

bbbenji, 2017-04-19:
Yes I expanded the partition. I am using v0.2.0 with the latest commits since that release, but that should not make a difference. Are you sure you enabled the hacks?

SanderSchutten, 2017-04-19:
I've just now expanded the partition also and did a reboot. I think that was causing the issue.
It did reboot with the SDCard in it... Yay!

I do get a message:
   Error: Nothing is mounted on /media/mmcblk0p2

If i click the button "manual mount" i get messages:
   Mounting /dev/mmcblk0p2 to /media/mmcblk0p2...
   Device is present (filesystem: ext2)
   mount: mounting /dev/mmcblk0p2 on /media/mmcblk0p2 failed: Device or resource busy 
   Failed to mount!

These last messages might have caused by myself, because i've been fiddling with the snx_autorun.sh script.

Thank you for giving the confirmation that it should've worked. I didn't expect it to be caused by not expanding the SDCard partition.

thanks!

SanderSchutten, 2017-04-19:
Btw... I'm trying to create a nice admin web interface for this device. I'll present my files as soon as possible for every to use.

bbbenji, 2017-04-19:
You are getting that warning because you need to wait for the resize to complete. It might take a long time so be sure not to reboot during this process or it may fail!

I started work on a nice admin page myself, maybe we can work together:) Looking forward to a pull request from you.

SanderSchutten, 2017-04-19:
uh oh.. :-D
Ejecting the SDCard from the device probably is not a good idea :-S

bbbenji, 2017-04-19:
You don't want to touch anything until that message goes away. Also if your issue is resolved, please close it.

SanderSchutten, 2017-04-19:
I'll be closing this issue and will get in touch with you about working together on an admin interface.
thx!

closed 2017-04-19 by SanderSchutten