samtap/fang-hacks #104: Camera crashes when IR is enabled (New camera version)

eterpstra, 2017-04-22:
Hi all,

So i have finally managed to install the hack on the new version of this camera. Everything is running fine during the day. However, when it is getting dark and the nightvision  / IR leds needs to be activated the camera crashes.  After this, the camera is in a bootloop. It is only working again when i place the camera in a light area.

Without the hack, the camera is working fine in the Mi Home app (with nightvision on)

I already tried a different SDcard and changed the IR script (updated version)

garathX, 2017-04-22:
Same here, the camera is always crashing when IR activated, I have to reboot it each time I want to see a picture ... and I don't know how to turn off IR ...

DavidRayner, 2017-04-22:
I've got the new camera version and applied the hack yesterday. I'm fairly certain I didn't have the issue you mention last night. I manually prepared my sdcard instead of using the 0.2.0 image.

eterpstra, 2017-04-22:
Is it possible to create an image with youre fresh install?

garathX, 2017-04-22:
I tried to disable IR and the camera still crash, I thought it was IR but it's not.

samtap, 2017-04-22:
Anyone know a place where I can buy the cam and am guaranteed to get the new version?

eterpstra, 2017-04-23:
Hi samtap,
I bought mine from AliExpress but ofcourse it is not garanteed that you will receive a new version.
 
You want to test with the new version?

dmbekker, 2017-04-23:
I've bought mine from banggood.com. Got the new version. Doesn't seem to crash when IR-leds are activated.  Flicked the light in my room several times. 

http://www.banggood.com/Xiaomi-XiaoFang-1080P-110-F2_0-WiFi-IP-Camera-Night-Vision-8X-Zoom-p-1095214.html

garvarma, 2017-04-30:
Hello @eterpstra,

In first post you say that you use the new version of camera with app Mi Home. Do you use it in Android or iOS? because i think that in iOS doesn't work this camera.

thanks

eterpstra, 2017-05-02:
@garvarma 
Hi,

I am using it with Android.

RaymondSchnyder, 2018-05-03:
I have the same problem, is there a solution?

garathX, 2018-05-05:
Yes there is a solution.
I put another firmware : https://drive.google.com/file/d/0B46sxK2orIewQzlSR042RWwzZGM/view 
from a french hack http://domo-attitude.fr/camera-xiaomi-xiaofang-flux-jpg-domoticz-jeedom
And my camera **NEVER crashed again !** It's even one of my most stable device !

RaymondSchnyder, 2018-05-08:
@garathX i love you

garathX, 2018-05-08:
Very happy if it resolved your problem too 😊

RaymondSchnyder, 2018-05-11:
@garathX not exactly. I couldnt test it, as i totally bricked my camera using another hack.
But if i get it running again, i will test this.

RaymondSchnyder, 2018-05-15:
Can someone help me with the serial connection? Im connecting GRND, RX and TX. It does not work, the connection gets immediatly closed. Do i have to connect the camera to usb power? do i need to connect all parts together and then wire it up? pls help

RaymondSchnyder, 2018-05-25:
Okay now i have the correct cables and chip to connect the camera to usb. It is yellow, so it boots up. But i get no output on windows/linux. Do i have to attach all parts of the camera or only the board with the serial connectors? because now i get no output. Maybe i should have more patience?