samtap/fang-hacks #105: curl

utya1988, 2017-04-22:
Hello. Who can upload curl andssl lib for this camera? Thanks.

mals, 2018-01-07:
Just wondering if there's been any progress on this? If not, can someone provide a guide on how to setup a cross compiler to build curl?

samtap, 2018-01-07:
Curl was added to the xf-data branch, I apologize for not having a ready-to-use image yet (work in progress!) but if you can't wait look here: https://github.com/samtap/fang-hacks/tree/xf-data (please don't ask exactly which files to download etc. if you're not comfortable with it just wait for the sdcard image)

mals, 2018-01-08:
Thanks Patrick. I also managed to find it in the SNX SDK. Thanks for your
help!

pir2, 2019-05-06:
any updates on this branch?