samtap/fang-hacks #110: Feature request: rsync or LFTP utility

farhanito, 2017-04-28:
hi, 
firstly I'd like to thank you for this awesome hack. i choose to use the cloud app, disable RSTP, and enable the rest functionalities. 

what i wanted to do is to sync/mirror recording files between camera's SDCard and remote storage (i.e NAS). 
i can run LFTP on a raspberrypi to do so, but i wonder if this can be done directly between the camera and the storage. 

thanks

mel525, 2017-04-28:
Also interested in such functionality. 
If it is possible, the better way might be to store video files directly on network share (smb). Without usage of SD card for intermediate storing, and lftp for file transfer.

samtap, 2017-04-28:
Scp is available, I'll see if I can include the other utilities in a future release

vaidyasr, 2018-02-02:
Rsync for xiaofang is available here: https://sourceforge.net/projects/nmtcsi/files/rsync_xiaofang.zip