samtap/fang-hacks #112: USB Ethernet?

milesburton, 2017-04-29:
Is it possible to use the USB output for ethernet or otherwise?

I monitored lsusb/dmesg and I noticed no traffic - is it purely to daisy chain cameras?

samtap, 2017-04-29:
It's only for power, data lines are not connected. Possibly you can simply attach some wires and get a fully functioning usb port. But I also remember reading somewhere in the datasheet that the embedded hub only supports a single device (wifi chip is also attached to usb bus).

milesburton, 2017-04-30:
Thanks @samtap that's interesting. I may have to pick up another and see if it's possible to add another usb hub. Do you know of any tear down photographs?

DavidRayner, 2017-04-30:
There are some linked in the [Wiki](https://github.com/samtap/fang-hacks/wiki/External-Resources).

closed 2017-05-08 by milesburton