samtap/fang-hacks #114: Reduce the bandwidth video stream

Pierinopower, 2017-05-01:
Hi,is it possibile to reduce the bandwidth video stream? This because I have NAS outside home and the streaming saturates my upload bandwidth. Thanks

DavidRayner, 2017-05-01:
Take a look at [this](http://bobbyromeo.com/technology/xiaomi-smart-1080p-wifi-ip-camera-rtsp-streaming-hack/#resolution) blog post. It describes how to adjust the resolution of the camera but you can just change the -b value to adjust the bitrate of the RTSP stream.

tamamma, 2017-05-02:
It would be nice have different streams like the Xiaomi Ant that have three different outputs:
rtsp://192.168.1.11:554/ch0_0.h264 (Video HD)
rtsp://192.168.1.11:554/ch0_1.h264 (Video SD)
rtsp://192.168.1.11:554/ch0_2.h264 (audio only)

mayekk, 2017-05-22:
Hello,

First of all, try 720p instead of 1080p:

snx_rtsp_server -W 1280 -H 720 -Q 10 -b 2048 -a >$LOG 2>&1 &

-Q - quality (leave 10),
-b - bitrate (try without -b - camera will adjust bitrate to picture details - VBR I think). If there are a lot of details or moving objects in picture bitrate will raise. Static picture is about 100-200Kbps.
-a (audio) - if You don't need audio, don't use this option.

Try to run this in ssh shell (and connect to rtsp unicast):
snx_rtsp_server -W 1280 -H 720 -Q 10 -b 2048 -a

You will see current bitrate,
then, kill the snx_rtpd_server, wait about 20-30 seconds and run:
snx_rtsp_server -W 1280 -H 720 -Q 10

connact to rtsp port and look at ssh terminal for bitrate info.

and connect

1043717432, 2017-05-31:
@Pierinopower  Hi, how does the faculty access SynologyNAS?

nykoo, 2017-06-16:
Hi,
is there a way to make the snx_rtpd_server to run in 720p (snx_rtsp_server -W 1280 -H 720 -Q 10) vy default ?
I don't know what to edit and how to edit it :/ (ssh, ftp)
Thanks for your answer

mayekk, 2017-06-16:
nykoo: You have to edit the file (one line):

/media/mmcblk0p2/data/etc/scripts/20-rtsp-server

ssh to camera, open file via vi editor and edit the line or
ftp - download file to local computer, edit it, upload and replace original
( after uploading by ftp You have to login by ssh and "chmod a+x 20-rtsp-server" )

nykoo, 2017-06-20:
@mayekk Thank you very much works like a charm :)