samtap/fang-hacks #115: Running RTSP hack and Mi home app at same time

bronzerain, 2017-05-02:
Can't seem to run the hacks for RTSP and the MI home app at the same time. Is this just how it is? I would like to be able to use the Mi cloud a long with the my RTSP server.

tamamma, 2017-05-02:
It would be nice, with the Xiaomi Ant this is possible. I've tried keeping cloud service "1" but the Mi app does not see my cam online

bronzerain, 2017-05-02:
I noticed that to. I then have to reboot the Xiaofang camera and then i can see it as online in the MI home app. When I apply the hack from the fang hacks menu the Mi home app shows it as offline. Also if I shut off the IR LED's from the Mi home app then turn back on again after applying the fang hack. The Ant camera seems to be a better camera but the wifi is weak. I am running the L version of software on the Ant camera with RTSP running.

bronzerain, 2017-05-02:
The Mi cloud lets  you save up to seven days of motion clips for free. I did not sign up for it yet. So for some reason your RTSP server stops working when you are away you can still see the motion detection alerts on the Mi cloud.

bronzerain, 2017-05-02:
Another cool thing you can do with these cameras is hook two of them together via their USB ports and get 180 degree plus view with image correction through the Mi home app. But I still want RTSP to my own server.

samtap, 2017-05-02:
For now you can't use mi home and rtsp simultaneously. I'm working on a solution but no promises.

bronzerain, 2017-05-02:
Thanks samtap for all of your work.

roger-, 2017-05-02:
@bronzerain 

> Another cool thing you can do with these cameras is hook two of them together via their USB ports and get 180 degree plus view with image correction through the Mi home app. But I still want RTSP to my own server.

Really? Have you tried this?

bronzerain, 2017-05-03:
https://www.youtube.com/watch?v=FmyjekO5SDU 
This video explains how.