samtap/fang-hacks #118: Does camera record to data partition?

layanto, 2017-05-07:
Just tried this hack today, rstp working great. 

A few questions:
Does the camera still record to data partition - like with the Mi Home app? If so, is this motion detection recording or recording all the time. 
Also how to access this data partition in Windows? I can only see 100MB of the 32GB SD card in Windows.

samtap, 2017-05-07:
You'll need to disable rtsp if you want to record with mi home. As far as I know, motion detection recordings are only stored in the cloud. Manual recordings are stored on the vfat partition. You can trick it to record to the ext2 partition but if you need to access the files offline on Windows, you're better off extending the vfat partition using some kind of partition tool on Windows.

closed 2017-05-07 by samtap

layanto, 2017-05-07:
If I don't want to use Mi Home and want to keep rtsp, can the camera record continuously?
If recording is stored in vfat, what is the purpose of the resize fs that took a long time?

samtap, 2017-05-07:
Someone did make a version of the rtsp server that records continuously, not included in the image but I remember reading it somewhere. I don't personally use the mi home app, that's why the partition layout is not helpful if you want to use it for recording. You can change it to suit your needs.