samtap/fang-hacks #119: Request: Documenatation for Newbies

TylonHH, 2017-05-07:
Hi,
I'm clever enough to set up an 3D Printer with alot of hacks, digitalize my power meter and get running 4 raspberry with different projects. All this was possible of the well documented projects.
I got running the MiFang Hacks, but now? The only benefeit now is that I have an rtsp stream and disabled cloud cam. But I see alot of other services runnung, but no wiki how to use them or for what they are.
Also to set up the timestamp is very time eating.
Did I missed somewhere a Wiki or tutorial?
I dont a step by step doc, but a list with all the services and how to setup up would be nice.

keith721, 2017-05-07:
Have you reviewed [the Wiki](https://github.com/samtap/fang-hacks/wiki)?

samtap, 2017-05-07:
If you look around there's a lot of information but I agree for newbies it is difficult to put info together if you run into issues. Luckily for most it works out of the box. If you don't recognize the services that are running you likely don't need them. People tend to overlook that currently the hack is mainly running a mostly functional rtsp server. None of the other features provided by Mi Home are implemented yet. I prefer to spend my time on that, and also spend a lot of time on discord chat helping out where I can. This is an open community and the wiki is open to edits by anyone, so my hope is users will contribute!

Toomix, 2017-05-10:
I found a video, which helped me a lot

https://www.youtube.com/watch?v=eiUdSjby57M

samtap, 2017-05-10:
Created https://github.com/samtap/fang-hacks/wiki/External-Resources#guides
There's a very extensive tutorial out there that also has some stuff about the ir filter but I can't find it right now :(

closed 2017-05-14 by TylonHH