samtap/fang-hacks #121: Cannot change mac address, please advice

CrazySerGo, 2017-05-09:
Hi.
Could you advice how to change mac address on camera, please?
I tried "ifconfig wlan0 hw ether  XX:XX:XX:XX:XX:XX" but received next error:
"ifconfig: SIOCSIFHWADDR: Operation not permitted"
Command with "su" didn't help as well.
Thanks in advance.

samtap, 2017-05-10:
You probably have to bring the device down first before you can change any of its settings.

CrazySerGo, 2017-05-10:
Your answer means "No", am I right? :)  or there is a chanse that I can change mac during the boot if put script on sdcard?
P.S. One guy said me that it may be issue with driver, because no such functionality. Is it possible?

samtap, 2017-05-10:
It means yes, but to make it permanent you'll indeed have to change it during boot before the device is up. 
For testing you could try a command sequence like '/media/mmcblk0p2/data/etc/script/01-network kill_network; ifconfig wlan0 down; ifconfig wlan0 hw ether xxx;  /media/mmcblk0p2/data/etc/scripts/01-network start'  that will complete eventhough your (ssh) shell is dead after the first command (you be able to see any errors or other output).
I believe it is supported, some devices have a little flash chip to make it permanent but the software method should work in any case, as far as I know.

closed 2017-05-28 by samtap