samtap/fang-hacks #122: New firmware version 3.0.4.9

samtap, 2017-05-10:
Apparently a new firmware version is available. Please report any info or issues here.

gmruiz, 2017-05-10:
Seem to be a problem with the new firmware and Mi Home app for users outside of China. When you connect the app show a message of not possible to use outside of mainland China.

samtap, 2017-05-10:
@gmruiz According to other reports, that seems to be the case on older firmware already, if you have one of the newer cameras (that have a qr code on the base).

poom, 2017-05-10:
I'm outside China. I cannot connect to camera but can access setting/recorded video

Now It cannot monitor anything... 

@samtap Can I downgrade firmware to 3.0.3.56?

samtap, 2017-05-10:
@poom Should be possible, let me know how it works out you're in uncharted territory: https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-(factory-reset)#via-sdcard

BluPix, 2017-05-11:
3.0.3.56 already don't work outside mainland china, i have tested this procedure. It's possible to find somewhere little older firmware?

yuyat, 2017-05-12:
3.0.4.9 (info) 
[image]
[image]

EmilioBravo, 2017-05-12:
Hi, I'm about to buy a couple of this cameras, but I'll be using them outside mainland China. Should I wait? @yuyat 3.0.4.9 works outside China? Where can I download it?

BluPix, 2017-05-12:
No, 3.0.4.9 doesn't work outside china, 3.0.3.56 doesn't work too.

EmilioBravo, 2017-05-12:
@BluPix so if I buy one now I wont be able to use it? Is there any firmware that I can flash to make it work? Thanks again!

guerrierodipace, 2017-05-12:
Italy 3.0.4.9 works normally with original firmware and mi home , camera without qr code

BluPix, 2017-05-12:
New model with MAC address starting 34 don't work and it looks like critical factor is camera ip address when it's registered to mi home app.  More info for example at https://xiaomi.eu/community/threads/xiaomi-xiao-fang-lincam-1080p-security-camera.37129/ or on 4pda.

jak0lantash, 2017-05-12:
Anyway a firmware downgrade could be done to work around this issue? I can't use the camera... http://imgur.com/a/iSYzy

dotromain, 2017-05-13:
Hi guys,
I have the new version apprently (QR code + Mac adresse begining with 34:) and it works well with the firmware 3.0.4.9 in the mi home app.

st3venhall, 2017-05-13:
Cant you just set your region to mainland china within settings and get it to work (even if youre not in china)? I'm on 3.0.3.56 and live in the US, and couldn't get the camera to work with the region set to United States, but then I set the region back to mainland china and all was well. I set the language to english so i could relatively navigate the menus without to much chinese craziness.

jak0lantash, 2017-05-13:
Doesn't work. Both my mi account and mi home app are set to Mainland China, still the same error, on iOS and Android.
My camera has a QR code at the bottom and the MAC address starts with 34.

https://xiaomi.eu/community/threads/xiaomi-xiao-fang-lincam-1080p-security-camera.37129/page-4#post-367754

st3venhall, 2017-05-13:
This article helped me out quite a bit
http://bobbyromeo.com/technology/xiaomi-smart-1080p-wifi-ip-camera-rtsp-streaming-hack/

"For some reason, I wasn’t able to install Mi Home through Google Play as it prompted me that none of my devices were compatible. OK then, whatevers… For this reason, I decided to sideload the app via ADB. I went to apkpure and downloaded the app manually, then installed it via ADB. I won’t get too much into detail about ADB, google if unfamiliar.

adb install MiHome_v4.0.11_apkpure.com.apk
Next, we shall set up the camera.

Open the app with whichever finger you’re most comfortable with and select Mainland China, if you select another region you may run into install problems later.

Power on the camera.

Use the included “paperclip poking device” to depress the setup pin on the bottom of the camera. The camera will make an audible sound and you’ll hear a voice speaking a foreign language that I don’t quite understand.

After a few seconds, a pop up will appear on your phone or tablet, click on OK to begin the pairing process, however, you’ll need to sign in first with a valid Xiaomi account. Create a Xiaomi account and sign in and click next.

Select your WiFi network and fill in your WiFi password.

You’ll see a QR Code on your screen, slowly point it in the vicinity of the camera’s field of view so it can read the code. If successful, you’ll hear some more audible words – probably telling you that you’re about to do something bad, really bad, so stop it, damn it!

Wait a little while to finish the setup. If you get a timeout while you see the percentage, hit “try again”.

Now click on the your camera, it should say “online”. In the top right corner you’ll see an ellipse icon, click it and go into “General Settings” and “Check for updates”.

Begin upgrade and wait some more."

jak0lantash, 2017-05-13:
Doesn't work. Still can't be used outside of mainland China.

TDPFlorida, 2017-05-13:
I'm a little confused...will the hacks work if the firmware is updated? I'm currently at 3.0.3.56 (and the hacks work fine)...can and how do you upgrade/change/load a firmware to any given revision? If you can, where do you find these files/firmware revisions?
Thanks

rsenn1, 2017-05-14:
I tried using a location spoofing app, but I'm still getting the error about not being in China.

TDPFlorida, 2017-05-14:
If I can't upgrade the firmware (because the hack will not longer work) is there a way I can disable the annoying popup to upgrade the firmware every time I select the camera?

samtap, 2017-05-14:
The hack will work fine regardless of the firmware version.

TDPFlorida, 2017-05-14:
Thanks...the hacks seem to work fine with Firmware 3.0.4.9 but I'm trying...
Change firmware Via SDCard

1.Download firmware file "upd_isa.camera.isc5.bin.extracted" from: firmware v3.0.3.56 
2.Extract the file and place the files in the root of your SDCard on the first (FAT) partition "mmcblk0p1"
3.Rename "0.elf" to "FIRMWARE_660R.bin"
4.Push setup button while powering up the device. The device should auto reboot after flashing.

I open "upd_isa.camera.isc5.bin.extracted" and extract ALL files to a FORMATED SDCARD. Then rename the "0.elf" to "FIRMWARE_660R.bin" and have tried BOTH holding down the setup button AND toggling it while it reboots. 

When the camera comes on the status is a SOLID yellow for a few minutes (I think its SOLID yellow when the firmware is upgrading) but when I look on my phone the firmware is still 3.0.4.9...What am I doing wrong?

Thanks

aropunto, 2017-05-15:
I have the new model of the camera, with the button setup (without hole), and the mac is 34:XX:XX:XX... with the bidicode.
I put the firmware v3.0.3.56...., like you say and i press the setup until 20 seg. and i format the SD in FAT mode, not FAT32 and i can do it.
But the Fang Hacks doesnt work in my camera. I follow the instructions and i make the SD with the 0.2.0.img and win32Diskimager.
 I read that is due be the new model. The light is always flashing blue but doesnt stop. The camera is in my wifi network (the bue light is flashing) but when i put http://device-ip/cgi-bin/status my webbrowers says that cant access to that page by a DNS adress error.
Any solution?
Someone has the firmware 2.8..... ? and the camera maybe work with mi home.
Please help.

jak0lantash, 2017-05-15:
I have the new model too, with the button (not hole), MAC is 34: and QR code on the bottom and the hack worked without difficulty. The F/W is 3.0.4.9.
I connected the camera to mi home as normal, then was prompted with the message that the camera cannot be used outside of China. After that, I used Win32DiskImager to write the img file onto an SD card.
And the hack worked.

aropunto, 2017-05-15:
Opps. 
Your led is solid blue or it is flashing blue all the time? when you insert the SD?

TDPFlorida, 2017-05-15:
aropunto...I am unable to format a SDCard in FAT, only FAT32, ezFAT and a few others. So you were able to change the FIRMARE?

aropunto, 2017-05-15:
I changed from version 3.0.4.... to 3.0.3... because i thought that it was the reason that the Hacks doesnt work but i read thata the hack works with that firmware version too.
Maybe FAT32 works to downgrade the firmware but in my camera not and I tried with FAT and works. 
put the sd and turn on with the setup button near 20s. the orange led is solid all time and wait about 5 min and the camera restart and the orange led is flashing.

TDPFlorida, 2017-05-15:
Thanks aropunto...what software did you use to format the SDCard in FAT (not FAT32?) The hack words in both versions for me too but I just want to be able to change the firmware with possible future issues.

oberlon, 2017-05-16:
I got my camera today. Initialized it via Mi Home, updated to 3.0.4.9 and got the same message in Mi Home app (outside of China). After this I prepared the SD Card (wrote img via linux) and inserted the card while camera was on. After the click sound I can run http://camera/cgi-bin/status and install fang-hacks. Had some trouble resizing the SD Card but rtsp is working.

samtap, 2017-05-16:
Boa is a lightweight webserver: http://www.boa.org/

oberlon, 2017-05-16:
Thx. Was to lasy to google it, sry:)
Updated my first comment. Seems that everything is working well.

VolaLazioVola, 2017-05-16:
I have done the hack on a camera with Mac 34: and QR code at the bottom and firmware 3.4.9. It did work well.
I see services up. I need to test RTSP.
I have only a question. When I do create the microsd with Win32DiskImager I get a very small partition (like couple of hundreds of mb). I tried to use the button to increase the second partition but failed. I get a message on the status page.
The problem is that camera might be able to save files on the microsd but if the partition is too small then there is not much space.

Sixerito, 2017-05-16:
Can you do step by step tutorial for a normal users please?

Thank you so much!!!

VolaLazioVola, 2017-05-16:
not sure I can put links... here the one I did follow...

[http://bobbyromeo.com/technology/xiaomi-smart-1080p-wifi-ip-camera-rtsp-streaming-hack](url)

rsenn1, 2017-05-16:
The problem is the LED never turns solid blue.

augstenk, 2017-05-17:
I have the same issue - firmware version 3.0.4.9 (new camera with code on the bottom) and Mi Home cannot work with the camera (although recording to SD card works and camera can connect to internet) - it says I'm not in China. Seller is totally unhelpful. Pretty annoying. I do not like the Mi Home app or the fact that everything goes through China server, but still, it would be nice to try it and then maybe go for the hacks. 

Unfortunately, I cannot even do the downgrade of the firmware - can someone help? I put the files on SD card (unzipped), but when I power the device and push the setup button, nothing happens. For the ssh/serial... I have putty, but I do not understand how to connect to the camera with serial (I cannot use ssh as it is not yet hacked). Does anyone have a serial port these days? Lats time I used this, it was in 2007. Also, the camera does not have serial port... and if I use usb to microusb, nothing happens, obviously.

joemilan67, 2017-05-17:
For hacking RSTP looking here: https://www.youtube.com/watch?v=eiUdSjby57M
For find serial port look here: https://www.youtube.com/watch?v=TsuCepL7b5U

aropunto, 2017-05-17:
Finally i can do the hack.
I think that the FAT partition only you can do it in a small SD (8Gb) with windows.

If you install Mi Home version 4.0.11 the messange of mainland china disapears and works.

Resabiao, 2017-05-18:
@aropunto Could you please specify the steps you followed to get the "mainland china" message disappeared? 
Which firmware/MiHome versions did you used?
Thanks in advance.

augstenk, 2017-05-18:
Thanks for the guidance, joemilan67.

Resabiao - Actually he is right - I do not get the message in Mi Home version 4.0.11 and earlier (even with the new firmware), but also the video is not shown - it is stuck at step 1/3 of setting. I can see all the information from camera (IP address for example), but not the video. I assume the issue is the same- the camera cannot connect to Xiaomi server where the app is taking stream from.

Force2611, 2017-05-18:
I also have the new version of the camera. I downgraded the cam to 3.0.4.9 via SD Card. I also used the older version of MiHome which is available via .apk. But I am not able to connect to the camera after the final set up. The camera is shown as "online", but if i want to check Live-Picture.. it stuck in the monitor screen at step 1/3 "Connecting". Is there a solution for it? P.S. With the newest updated version i also have the "outside china" problem. Thank you.

augstenk, 2017-05-19:
"I downgraded the cam to 3.0.4.9 via SD Card."

3.0.4.9 is actually the latest firmware - the one that is causing all the issues. You should downgrade to  v3.0.3.56 or lower. 
I have the same issue (I haven't managed the downgrade yet) - next time I have some time, I will try the hack, then downgrade firmware and I will try to use both Mi Home and RTSP running. 
I wonder did someone used this camera with Synology NAS? I know there are 2 IP camera licences free, but I was under impression those cams have to be supported by Synology, which I think this one won't be (as it is designed only for chinese server streaming).

Force2611, 2017-05-19:
Sry. My fault. The current firmware on my camera is 3.0.3.56

gielk, 2017-05-19:
I have Version 3.0.3.56 on my xiaofang  but also here not able to use outside of china. Just received it. why did the block is ?

TDPFlorida, 2017-05-19:
augstenk OR anyone, how do you downgrade the firmware? I am on F/W 3.0.3.56 and F/W 3.0.4.9 and the hacks work on BOTH (I'm in US, running MiHome v3.16.0 in mainland China) but I like some of the features on F/W 3.0.3.56 and would like to downgrade the one I have on F/W 3.0.4.9 not to mention in the future I might need to downgrade.
I have tried all the things on this thread at different times and including:
https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-(factory-reset)#via-sdcard
via the SDCard and none seem to work (EXAMPLES include FAT/FAT32/exFAT 4GB SDCard the sample . Can someone POST a complete procedure to do this using the SDCard?
Thanks

Sagonips, 2017-05-19:
Hi everyone,

The cameras do work on the latest firmwares outside of China.
I have just received my first 4 xiaofang cameras and  I updated one of them to 3.0.4.9 and the other three are still on 3.0.3.56.
I noticed everyone saying the latest firmware and even the previous one doesn't work outside of China but I am in Australia and all of these cameras including the one with the latest firmware work perfectly for me.
You just have to change your location to China in the Mi App.. I don't know if this only works for Australia or not but give it a go. I was unable to connect to the cameras at all until I changed my location.

Let me know

Resabiao, 2017-05-19:
@Sagonips. Which are the  2 first digits of the mac address?

Sagonips, 2017-05-20:
@Resabiao 34

Sagonips, 2017-05-20:
I've just realise someone else already mentioned changing region to China before sorry I didn't read the entire thread. It's weird that it doesn't seem to be working for others.

bussekas, 2017-05-20:
i get also this camera.

with fw 3.0.3.56 it connecting 1 and blue is blinking. nothing more do. with fw 3.0.4.9  still same state in Europa Estonia.

What i should try more?

Liorkatz, 2017-05-21:
same here, i got the camera with the new version ( 34, and push button ).
In the app ( IOS )  theres 2 camers one with ( ) and one without .. first i installed the one with the ( ) the camera **worked** !! 

then i install the other type witout the ( ), enter the wifi password scan the barcode with the camera then it ask me to update , and then BOOM the camera **not working** .. 

i tried to **downgrade**  .. not helping 
i tried to install the first camera in the app ... not helping

plz help...

naut1co, 2017-05-22:
Hi guys, 

i've got the same issue.... same camera with the MAC address in 34, and push button. 
First I upgrade my cam and got the message...
then I downgrade it and now i'm in fw 3.0.3.56

On my Ipad I'm able to configure the connection in direct.
my translate motor say : camera fang (direct)

cam generate a wifi named isa-camera-isc5_miapA1F1. (maybe the name change)
so with a direct connect i 'm able to connect on the cam....
i got an orange blinking led 

maybe it can help some dev

TDPFlorida, 2017-05-22:
Can someone tell me how they downgraded the firmware?
Thanks

krystiag, 2017-05-22:
The problem is the LED... never turns solid blue..

bussekas, 2017-05-23:
With Hack led is turning solid blue. with orginal firmware it does not do.

is there any good way how to use CAM over internet? Right now i use http://ivideon.com/ but is there any better solution?

naut1co, 2017-05-23:
bussekas I tried to flash it with hack but without succeed.... 
is your cam a new model?

bussekas, 2017-05-23:
I get it two weeks ago. it seems to be latest version of it. I use hack and with it i can use ip camera as it should be but for viewing out of home i use ivideon. it was first application what seems to be easyst to use but if there is another and better solution..then i'm intrested.

with orginal fw it give me only error mainland china. with seller i ask and he told to make vpn over china but this is not some ok solution for me in europa, estonia.

rsenn1, 2017-05-23:
bussekas, krystiag means you cannot even apply the hack because the LED never turns solid blue.  I am trying the VPN method right now but so far no luck.

naut1co, 2017-05-24:
I tried Vpn and fake GPS does not work.....

You can recognize new version with the push button and not a hall to reset cam 
all the test i make I was not able to flash it

blautob, 2017-05-24:
I can confirm that my cam ( new version with the push button and not a hall to reset cam) has been succesfully hacked.

The blue led however is blinking constantly.

[56 more replies]