samtap/fang-hacks #123: Only maniland China: how to unlock camera for EU?

walterkaos, 2017-05-12:
Dear users, I bought a Xiaofang camera without paying attention to the country. Now, my camera can only operate from China IP when connected. Since I want use the camera from Europe, this is banned! How can I update firmware unlocking region ban?

jak0lantash, 2017-05-12:
I have the same issue

gmruiz, 2017-05-13:
the same for me! Please, hack for this.

milesburton, 2017-05-14:
On all but one of mine, you can still install the chinese setup app. Go through the setup process and install the mod via TF card.

Dont worry too much if it says "mainlaind china only" or some such

samtap, 2017-05-14:
New camera's are blocked from using mi home cloud outside China. You can still use the app to connect the camera to wifi and then apply the hacks by inserting sd-card and using the status web-page. I've got one of these new cams and confirmed this works, also with the latest 3,0,4,9 firmware.

Please report here if you have a different experience. Include mi home version, iOS/Android, firmware version and camera model (pinhole or lever reset button)

jak0lantash, 2017-05-14:
Applying the hack worked flawlessly for me after initial setup on 3.0.4.9 via mi home, despite the camera being geoblocked ("only used in mainland china") by Xiaomi.

sfornengo, 2017-05-16:
It seems there is a hack for the same problem on an other Xiaomi camera:
https://diy.2pmc.net/solved-xiaomi-xiao-yi-ant-home-camera-can-used-china/
Is someone able to transpose to XiaoFang ?

milesburton, 2017-05-16:
Good spot, I'll look into this when I get a chance

ChavezD, 2017-05-17:
So for me i was able to use the mi home app (4.0.11, android) with my new camera (MAC 34:xx...+QR on bottom+button instead of this needle push thing there), but im still on FW 3.0.3.56.
Does this "workaround" still work if i update the camera to 3.0.4.9?

However I'm going to flash the hack this evening, the Mi Home app isn't that great imho.

wahaha2017, 2017-05-17:
I hacked the camera and got the log.txt. when I used the usa vpn it didn't work, "mainland china only", log is wahaha.txt; when I used the china local net it worked, log is chinangb.txt. I can't find the differences between them. 
 
[wahaha.txt](https://github.com/samtap/fang-hacks/files/1008190/wahaha.txt)
[chinangb.txt](https://github.com/samtap/fang-hacks/files/1008191/chinangb.txt)

wahaha2017, 2017-05-17:
Connect Server
g_stTutkUserServerInfo.nLoginFlag = 0XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX   nLoginRet  -10
<><><><><><><><><><><><><><><><><><><><><><><><>88888888888888888888:TUTK LOGIN ER UNLICENSE

how can fake the response ???

samtap, 2017-05-17:
I don't really have any interest in circumventing the region check, the goal of this project is to create an open-source alternative firmware that doesn't require any of the Xiaomi stuff. 

But for those interested to hack the region block:
- It appears the old version is accepted regardless of cam firmware and mi home version. These can be recognized by not having a QR code on the base. The wifi chip is sourced from some Shenzen supplier and the MAC address does not match the MAC address in device.conf (28:XX...) used for identification with cloud services.
- The new version has a wifi chip from Xiaomi with MAC address that matches the one set in device.conf (34:XX...). Now the interesting bit: device.conf does not exist on a virgin camera. It is created by iCamera when it starts, way before a wifi connection is up or Mi Home is involved. (iCamera is the executable running on the cam doing all the work).
Ergo: If you can make a new cam appear like an old cam, by providing a device.conf with a 28:XX MAC, it will be accepted by Mi Home.
- The MAC address on wlan0 is set by iCamera when it starts so spoofing it before it starts is useless (i.e. run 'strings iCamera |grep ifconfig'). I believe from that point forward, the MAC address is stored in memory of iCamera, spoofing after it starts has no effect (wifi works fine with a different mac, device.conf still contains 34: MAC).
- Each time you press the setup button (required for pairing with Mi Home), device.conf is rewritten in case you messed with it. It writes the 34 MAC regardless of the MAC used by wlan0 at that time. A device.token is written when Mi Home starts the pairing process, it is different each time. When making device.conf inaccessible, iCamera enters an infinite loop and doesn't proceed with pairing. Messing with it too much during pairing simply causes the pairing process in Mi Home to timeout and you have to start from scratch.
Ergo: iCamera has a way to retrieve the MAC address which is not simply reading ifconfig HWaddr. It is also unlikely stored in some kind of file as that would require them to flash each cam with a different file as they are produced, possible but unlikely (and I could find no evidence of such file). It is not provided by Mi Home or cloud services either since it's used to configure wlan0 before any pairing is initiated.
- Interestingly, the MAC is stored in the 'serial' field of the USB device descriptors provided by the wifi chip. And iCamera happens to link with libusb....! I can't think of any other reason they would need libusb besides doing accessing the wifi device. I've added some debug probes in libusb in the most likely places: functions that return the serial descriptor. Those didn't get hit. But there're still many other calls that are more low-level and could equally be used to retrieve the serial.

Conclusion: If somebody would build a custom libusb that rewrites the serial in case it starts with 34 before returning it to caller, it might trick iCamera into thinking it is running on a old cam.

wahaha2017, 2017-06-09:
how can I get the  the open-source alternative firmware now?

nykoo, 2017-06-16:
same :( 
Cam FW 3.0.3.56 and iOS 3.17.0 Mi app

idaadi, 2017-06-19:
@samtap  wouldn't it work to hard code the Mac directly in the Driver for rtl8188E , recompile it for arm and replace the original one ? do you think the 28..... mac address needs to be the actual mac of a Camera they produced( they have a DB with all their own macs ) or not ?  How could the community motivate you to circumvent the geo block ? 
Regards

samtap, 2017-06-19:
I'm not interested in using the Xiaomi cloud stuff, there're many other things that need to get done... Since my previous post I noticed the mac is also stored in nvram so that may also play a role. I don't think the wifi driver is involved since I was able to spoof it and not get around the region block, and the older version has a 'fake' mac that doesn't match the hw mac but is used to identify with xiaomi cloud. I don't think they have a db with blocked mac, it's the mi home app that accepts or rejects the cam. If you use an older version of the app and block location services the cam is accepted (or, that's what I've been told).

santianton, 2017-07-03:
I have three cameras with MAC 34 and none of them works with the fang-hacks...

walterkaos, 2017-07-03:
Guys, me too. There's no solution. I have sent back the product to chinese online shop in order to be refunded.

zg2302vi, 2017-07-13:
There is EU solution https://www.ismartalarm.com/devices/cameras/spot/isa00013.html
Only caveat is it costs 2x > 
https://www.leroymerlin.fr/v3/p/produits/camera-connectee-ismartalarm-e1500580033
Bought one discounted for 49 eur

carloslebreiro, 2017-07-13:
Is it possible to "export" the firmware of that camera? It would be great if we could upgrade the firmware of our xiaofang to the ismartalarm original firmware

TweedleMB, 2017-07-13:
My configuration is Camera (MAC start with 34) firmware 3.0.3.56 downgradet from this maual - https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-(factory-reset)#via-sdcard 

Android Mi Home - version 4.0.11 :) from this link www.apkmirror.com/apk/xiaomi-inc/mihome/mihome-4-0-11-release/mihome-4-0-11-android-apk-download/

everything works.

Damadgeruk, 2017-07-13:
The above worked for me, 3.0.3.56 showing, getting video on mi home 4.0.11 tho motion sensing is patchy. Blue led flashing.

Jumpertrekker, 2017-07-14:
@TweeldeMB I have the same configuration but the app sees the the camera only if on the same wifi. Over 3g the app doesn't see the cam. Is it the same for you? Does anyone know what tcp/udp port uses the camera? Maybe I just need to better setup my router.
Thanks!

mp3llll, 2017-07-17:
@TweedleMB  and that work fine via HOME app also via 3G?

Damadgeruk, 2017-07-17:
Works over mobile data for me on Mi Home app.

ktanrtp, 2017-07-18:
@TweedleMB , I have installed 4.0.11 Mi Home App and my cam's firmware is 3.0.3.56 with hacks applied. I can do rtsp streaming. 
But I still cannot connect to my camera using MiHome App, even after disabling "stop-cloud" and "rtsp-server" services by following [FAQ](https://github.com/samtap/fang-hacks/wiki/F.A.Q#after-applying-xiaofang-hacks-does-the-camera-still-work-with-mi-home-app). The app is stuck at "Connecting (1/3)"

Are you saying that you are using the 4.0.11 Mi Home and 3.0.3.56 smoothly without applying the fang-hacks?

Thanks

J450NC, 2017-07-19:
Be gentle first post, and not a programmer, I have three of these camera's 3x MAC code 34 ****** .
They all behave differently !
The first doesn't have a press button for the reset - it has a hole - this works with the MiHome app and I even managed to upgrade to 3.2.0.30 with no dreaded 'only works in China error' - still working no errors.
The second has a push button for the reset and I upgraded the software believing from my previous experience with the first camera this would be ok - I got the dreaded 'only works in China error'
That brought me here to this thread and website - although I do not profess to be a programmer I can follow the SD format protocols and writing image files or working out what goes in a root directory etc.
I created a downgrade microSD card to 3.0.3.56 and flashed the second camera. It now has 3.0.3.56 firmware but still reports 'only works in China error' and has the option to upgrade to 3.2.0.30 in the firmware update section. It took several attempts but I think the key is the duration of holding the reset button after reapplying power as mentioned above.
The third camera also has MAC code 34 ***** but I have never upgraded the firmware when prompted and this one works with the MiHome app with no errors.

Summary:
Camera 1 - earlier vrs no button to reset (has a hole) MAC code 34 - currently 3.2.0.30 - working
Camera 2 later vrs button reset MAC code 34 - currently upgraded to 3.2.0.30 then flashed to 3.0.3.56 - it has region code error
Camera 3 later vrs button reset MAC code 34 never upgraded - currently 3.0.3.56 - working

My question/s are:
Is the second camera now flashed to a point where the China region error cannot be removed ?
If I use the fang hack would the MiHome app stop its remote functionality - appreciate that is the whole point to not use MI server to bounce/stream data - I'm looking at converting all three ultimately but want to try to recover camera number 2 so that if I want to revert to 'factory settings' I have the knowledge how to.
I am using IOS vrs of MiHome 3.19.0 set to mainland China

With the different permutations I have it is quite easy to see why members are having difficulty as I have three cameras all behaving differently all with MAC code beginning 34 *****

Has anyone had any success in the email link when you get the 'China Region Error' 
Can anyone translate what this page is saying ? 
It 'appears' to request you take a photo of the base of the unit and email that to an address to request international use, has anyone tried this ? 
 
My experience, so far, is that once the firmware upgrade is applied, although the downgrade can be applied the region error is retained ( see my comments above re camera 2 and camera three differences) I will try flashing camera 2 a few more times to see if it can be returned to factory settings and post my findings.

munineo, 2017-07-20:
Hi all dudes!

I'm trying to downgrade my xiaofang cam, but I don't know how time must press the button to know when it is flashing the firmware... Is there and led colour blink to detect that it is flashing or how to know when it is rebooting after flashing?

Other question, how to know witch firmware is in cam? I haven't serial usb cable...

J450NC, 2017-07-20:
To Downgrade - Start camera, wait, scan with app to pair to wifi, wait, when nothing new is happening insert SD card and hold button down .... keep holding it down, voice says something in Chinese, keep holding button down, when she's finished speaking wait until the orange light stays on then let go of the button BUT DO NOTHING ELSE at this point leave the power on and WAIT once it is doing nothing new remove SD card and cycle the power. That's how I have been flashing mine to revert to lower firmware.

munineo, 2017-07-20:
Thx @J450NC downgrade done and cam with 34 MAC visible from Mi Home

munineo, 2017-07-20:
No I can't over 3G @Jumpertrekker

J450NC, 2017-07-21:
OK, I've tried discussion/email follow up with MiHome pop-up email for region error, here's the reply:

亲爱的用户：
非常抱歉，您的申请未通过审核。
如果您依然希望在大陆地区之外使用小方，请严格按照要求提供正确的资料。
请提供在小米官网/小米商城/小米天猫旗舰店/小米京东旗舰店等正规官方渠道购买的带有订单号的截图；
小方底部的MAC地址拍照示意图。
感谢您对小方智能摄像机的理解与支持。
(Translation)
Dear customers,
We appreciate your business. Unfortunately, your device unlock request was not been approved.
However, you still can use your Spot in Mainland China.  If you still want to unlock your Spot and use it outside of Mainland China, please follow the Device Unlock Guide resubmit you request.  
Please provide the purchase records of MI.com、JD、MI Tmall or others formal official store.
Image of bar code. 
Thank you for choosing our products.
.................................................................................................................................................

Needless to say I purchased from a cheaper alternative source, I've tried reapplying with bill of sale barcode photo etc but not holding out much hope. So fanghack is probably the only way to go unless someone can bypass the region / mac 34 lock out.

munineo, 2017-07-21:
They give me same answer... Bought at Gearbeast...
I've other one without 34 MAC address and the newest with 34 MAC in which I must downgrade it to see on Mi Home.

ojgolan, 2017-07-25:
did any1 try to flash the xiaomi xiaofang with the ismart spot firmware?
Since there are too many similarities (even the root password is ismart12) did anyone manage to set this up?
As this may be the way to clear this crap of the limitation with the global branding.

snoerenberg, 2017-07-26:
Hi,
is someone having the ismart spot firmware or the actual xiaofang firmware version 3.2.0.30 as binary?
I already had a look with IDA Disassembler into the flashing part (1st part of binary) and I would like to ask if someone tried such a "cross-flash" and if there is any problem or error message.

So I would be able to change the existent ismart firmware to be flashable on xiaofang as well. I think there will be some model_id check which must be patched.

ojgolan, 2017-07-26:
@snoerenberg If you manage to do this, this will be superb.
Both cameras look the same, expect the ismart is supposed to be 720p and has 4 IRs instead of 2.
It also costs 4 times the amount of the Xiaomi Xiaofang...

munineo, 2017-07-28:
The spot with MAC starts with 34 doesn't work right with MiHome apk... not always connects when I try to access it on the same WIFI, thats a pitty.
Hope you've got good news with your effords

RylicR, 2017-07-30:
Hi guys,
I tried everything to downgrade my XiaoFang but nothing work... I really don't know what to do. Someone can explain more precisely what exactly to do? I tried : 
- Turn off, put card, hold button, Turn on, the light is orange fix, then i listen the "dudumdudum xabanheyhey" and light blink, then light turn off, i unhold button, 30 sec after light is blinking. Then when i connect, stil not downgrade.
- J450NC solution : i turn on, clic on button to have "dudumdudum xabanheyhey", connect to MiHome, light turn Blinking blue, insert the SD card, listening "klink klink", hold button and this done the usually "dudumdudum xabanheyhey", still hold then light turn fix orange... For 30sec then blanking again as usually... After connection it's stil the same FW...
I really don't know what to do... Please help, explain what to do and what could happened :( TY

J450NC, 2017-08-06:
Pair as normal - scan barcode. (This is no different to manufacturers instruction)
Put in SD card and hold button in - voice says something - don't let go yet
You should only have a constant yellow/orange light on - constantly on. 
Now you can let go - wait until nothing new is happening. Take out SD card.
Cycle the power.
If it is still not a lower firmware your SD card probably isn't set up correctly.

UPDATE:
Now another of my cameras has China mainland only msg.
I think any use of the MiHome app will soon be closed out for us Euro users - only solution will be a different firmware apk hack for the MiHome - region in MiHome App is set to 'Never' allowed location access but there is a default there and all update info on app page says to use location which should include 'locale region' - fanghack will be only option. Does anyone know how to fake a location in iOS - might be another option for those that want to use MiHome.

ilkdostun, 2017-08-07:
I had the same problem before. I have a camera with Mac-adress starring with 34 (the one with the plastic reset button). I got it to work using the Mi Home app version 4.0.8 (you can download it on Google if you search for Mi Home 4.0.8 apk). Install the app and then unselect "update plug-ins", set region to "Mainland China" and change language to English. DO NOT UPDATE CAMERA OR APP.

When this is done you have to reset the camera to the firmware version 3.0.3.56. You can do this by holding in the plastic setup button for 12-15 seconds. The status-led will tur off or solid orange. Wait until the led starts flashing orange. Now the camera is reset to the firmware it came with. 

Do the setup as usual and make all the steps on the screen.

When you are done, after a while, the camera status-led will start blinking blue. Now the camera will work on WiFi LAN only. Unplug the camera and leave it unplugged over night. plug the camera back in and wait for a couple of minutes and the status-led will turn solid blue. This means that the camera is connectet to the cloud server and now can be accessed via your phones 4g/3g or on a different WiFi than the cameras.

I don't know why you will have to wait over night for the camera to start working via cloud (4g/3g). But that's how it is.. I have tried this with 3 cameras and it works on all of them.

I hope i helped someone!﻿

devilweb77, 2017-08-10:
The reset the camera to the firmware version 3.0.3.56 holding in the plastic setup button for 12-15 seconds doesn't work.
I still have firmware 3.2...

J450NC, 2017-08-12:
Use the SD slot with 3.0.3.56 software on it to downgrade then.

adrianmihalko, 2017-08-12:
Guys, is there any hostname/address which blocks firmware upgrade checks?

Jumpertrekker, 2017-08-13:
@ilkdostun I'm almost there... I can access to some functions also from 3g
but still no streaming. Do you also get the streming to work on 3g?

schmurtzm, 2017-08-15:
Hi,
For information, I've got the right version for IOS.

On Android you can use the [version 4.0.11](https://apkpure.com/mihome/com.xiaomi.smarthome/download/60617-APK?from=details%2Fversion) (but that you already know).
On IOS you can use versions 3.11 and 3.12 only. Inferior versions (3.10) does not have the XiaoFang in the device list and superior versions (3.13) have the lock to mainland china.

To get the right version on IOS it's not easy : 
- you're jailbroken : install "App Admin" from cydia, go to app store and chose version 3.12 of Mi Home
- you're not jailbroken, follow one of these tutorial (not easy) : [This one](https://medium.com/@iosight/how-to-legally-download-any-previous-version-of-an-app-store-app-through-itunes-9b45559b8357), [other one](http://www.idownloadblog.com/2015/12/25/how-to-download-older-versions-of-ios-apps/) and [video](https://www.youtube.com/watch?v=KWrGh2d7hFg). After that backup the .ipa file with itunes to be able to restore it later.

Willian-Zhang, 2017-08-22:
> To Downgrade - Start camera, wait, scan with app to pair to wifi, wait, when nothing new is happening insert SD card and hold button down .... keep holding it down, voice says something in Chinese, [...]

The downgrade guide in wiki didn't work, this worked for me

Resabiao, 2017-08-22:
The guide I followed: https://www.andryou.com/2017/07/11/setting-xiaomi-xiaofang-security-camera/.
Resulting:
When sd card keep inserted, blue light fix, hacks (rtsp) working , mi home 4.0.8  not getting image from camera ( lan neither 4g).  Camera firmware:  3.0.3.56.
SD card out  (and re start): on lan, mi home fully functional. On 4g no video stream, no connection and getting some alerts from movement detection, but downloading video error.
Thinking on sending  this time consuming piece of trash to hell....

Andreaux, 2017-08-28:
Does anyone know if the form (link in the error message page) where you could send in a proof of purchase and a photo of the mac address of the camera to have it unlocked for international use works? I can't read chinese so I can't tell, but I presume it's meant to be that...
Anyone?

Andreaux, 2017-08-28:
On the other hand, does anyone know how it would be possible to spoof the mac address of a camera to change it to not beginning with 34 or so...?

ilkdostun, 2017-08-29:
i tried the way **Resabiao** mentioned and it made the camera downgraded. Thanks

Andreaux, 2017-08-29:
Downgrading is not the issue here. I managed to do that as well, it's the mac address check that we should somehow get around...

AdrianKielbowicz, 2017-09-01:
Does anyone have a definite answer if this camera currently works outside of mainland China? I thinking of buying one but don't want to end up with a useless brick!

Resabiao, 2017-09-01:
Unless you have a lot of free time, and want to invest it on learning how to deal with this camera, DO NOT buy it, its a lottery

AdrianKielbowicz, 2017-09-01:
Thanks for that @Resabiao.

I will look at other options then :)

singhvivek73, 2017-09-05:
I am able to make it work by changing the locale in MI Home to China, no other change was required.

kulokoy, 2017-09-06:
Hi everyone, just want to share how I managed to get around on viewing the camera outside your personal/home wifi network connection.

1. You need an android device(tablet/phone/tv box) - stand as a host.

2. Install teamviewer android host app on the dedicated host device. - to connect and remotely view the camera inside your network unattended.

3. Register to teamviewer to save the device host id.

4. Run the teamviewer host app to host device prior to remote connection.

5. Login to your teamviewer account to connect to host device and remotely. Make it sure that the host device is connected to your personal home network.

Hope this help.

pedropcruz, 2017-09-11:
really cant do anything with this camera... And i am starting to be pissed of!

Can anyone can just explain well? I cannot connect my camera to my wifi neither!

Many thanks!

Andreaux, 2017-09-12:
If your wifi has mac address filtering, maybe that's the problem. It turned
out mine had an other mac address than was on the sticker and it couldn't
connect to WiFi.

joarno, 2017-09-22:
hi guys
TweedleMB managed to made the camera works somehow. Maybe he could help me or someone else

I followed step by step the procedure from 
https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-(factory-reset)#via-sdcard

joarno, 2017-09-22:
hi guys
TweedleMB managed to made the camera works somehow. Maybe he could help me or someone else

I followed step by step the procedure from
https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-(factory-reset)#via-sdcard

even from here:
https://www.andryou.com/2017/07/11/setting-xiaomi-xiaofang-security-camera/

mi home app is 4.0.11 as advised.
mac address of the camera starts with 34
Impossible to downgrade firmware to 3.0.3.56

I am obviously doing something wrong.
Could someone who succeeded to downgrade the fw 3.2.0.30 to 3.0.3.56 detail me step by step his process. Exactly what you have done to do it. !because sometimes I see different answers that the partition must be named mmcblk0p1, sometimes a folder should be named mmcblk0p1.
I am beginning to be confused and I tried every single possibilities.

Thanks for the help.

toncsiking, 2017-11-18:
Ok, my Dear Friends!

So, I bought a Xiaomi camera, model ISC5 from Ali express.
Went trough all the disappointing methods You have, to get it alive.
I try to sum all my experienses – here they are:
Here’s my solution – I don’t swear it helps for everyone, but for me it works…mostly…but only at my home (I mean the mobile phone has to be on the same Wifi as the camera, doesn’t work from other places, Wifi, or mobile net  :( )
1.	Use/download the old application of MiHome (https://apkfree.com/download/mi-home-4.0.11) – it is easy: erase/delete the new app from Your mobile, and just download the old one from here, and let it go…
2.	Downgrade the camera firmware – it was a little difficoult for me – but at the end I won.
So I took an 8 G micro SD card (deleted and formatted) and copied the old firmware to the card with my computer– just as [You can get it from here with a little help](https://drive.google.com/open?id=1kC0BgM52i3ow15x4DWLNaa0oDsRtXyev).
After I inserted this SD card into the camera which was "out of power"
While pushing the reset button on the bottom of the camera  I plugged the power cable, and then I still held the button for about 20 seconds
I waited about 2 minutes and then I went through the pairing procedure (You know very well ):
1.push the reset button for a second
2.wait the chinese voice 1
3.show the qr code of the mobile screen to the camera
4.chinese voice 2 sounds
5.wait about 60 seconds until it’s done (maybe You have to repeat it a few times, I mean the pairing method)

I think it’s important to mention:
to disallow the phone application to refresh itself automatically You have to switch it out!
(go to Playstore, find  MiHome, from the menu select do not refresh)
and later You never refresh anything in the application

Hope it helped :)

Tamas

[198 more replies]