samtap/fang-hacks #131: How to capture RTSP Stream?

sbrinkmann, 2017-05-17:
I've tried to capture the RTSP with openRTSP or ffmpeg, but without success as you can see in the attached log. Does anyone has a working solution to capture the RTSP on linux?

```
# openRTSP -4 -w 1920 -h 1090 -f 10 -P 3600 -I 192.168.42.105 "rtsp://192.168.42.124/unicast" > foo.mp4
Created new TCP socket 3 for connection
Connecting to 192.168.42.124, port 554 on socket 3...
...remote connection opened
Sending request: OPTIONS rtsp://192.168.42.124/unicast RTSP/1.0
CSeq: 2
User-Agent: openRTSP (LIVE555 Streaming Media v2017.04.26)

Received 152 new bytes of response data.
Received a complete OPTIONS response:
RTSP/1.0 200 OK
CSeq: 2
Date: Wed, May 17 2017 19:51:05 GMT
Public: OPTIONS, DESCRIBE, SETUP, TEARDOWN, PLAY, PAUSE, GET_PARAMETER, SET_PARAMETER

Sending request: DESCRIBE rtsp://192.168.42.124/unicast RTSP/1.0
CSeq: 3
User-Agent: openRTSP (LIVE555 Streaming Media v2017.04.26)
Accept: application/sdp

Received 743 new bytes of response data.
Received a complete DESCRIBE response:
RTSP/1.0 200 OK
CSeq: 3
Date: Wed, May 17 2017 19:51:05 GMT
Content-Base: rtsp://192.168.42.124/unicast/
Content-Type: application/sdp
Content-Length: 580

v=0
o=- 1494960745014391 1 IN IP4 192.168.42.124
s=LIVE555 Streaming Media v2014.07.04
i=LIVE555 Streaming Media v2014.07.04
t=0 0
a=tool:LIVE555 Streaming Media v2014.07.04
a=type:broadcast
a=control:*
a=range:npt=0-
a=x-qt-text-nam:LIVE555 Streaming Media v2014.07.04
a=x-qt-text-inf:LIVE555 Streaming Media v2014.07.04
m=video 0 RTP/AVP 96
c=IN IP4 0.0.0.0
b=AS:956707
a=rtpmap:96 H264/90000
a=fmtp:96 profile-level-id=644029;sprop-parameter-sets=Z2RAKawsqAeAIn5U,aO44gA==
a=control:track1
m=audio 0 RTP/AVP 8
c=IN IP4 0.0.0.0
b=AS:956707
a=control:track2

Opened URL "rtsp://192.168.42.124/unicast", returning a SDP description:
v=0
o=- 1494960745014391 1 IN IP4 192.168.42.124
s=LIVE555 Streaming Media v2014.07.04
i=LIVE555 Streaming Media v2014.07.04
t=0 0
a=tool:LIVE555 Streaming Media v2014.07.04
a=type:broadcast
a=control:*
a=range:npt=0-
a=x-qt-text-nam:LIVE555 Streaming Media v2014.07.04
a=x-qt-text-inf:LIVE555 Streaming Media v2014.07.04
m=video 0 RTP/AVP 96
c=IN IP4 0.0.0.0
b=AS:956707
a=rtpmap:96 H264/90000
a=fmtp:96 profile-level-id=644029;sprop-parameter-sets=Z2RAKawsqAeAIn5U,aO44gA==
a=control:track1
m=audio 0 RTP/AVP 8
c=IN IP4 0.0.0.0
b=AS:956707
a=control:track2

Unable to create receiver for "video/H264" subsession: getsockname() error: Bad file descriptor
Unable to create receiver for "audio/PCMA" subsession: getsockname() error: Bad file descriptor
```

```
# ffmpeg -i rtsp://192.168.42.124/unicast -c copy -map 0 -f segment -segment_time 300 -segment_format mp4 "capture-%03d.mp4"
ffmpeg version 2.8.11-0ubuntu0.16.04.1 Copyright (c) 2000-2017 the FFmpeg developers
  built with gcc 5.4.0 (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.4) 20160609
  configuration: --prefix=/usr --extra-version=0ubuntu0.16.04.1 --build-suffix=-ffmpeg --toolchain=hardened --libdir=/usr/lib/arm-linux-gnueabihf --incdir=/usr/include/arm-linux-gnueabihf --cc=cc --cxx=g++ --enable-gpl --enable-shared --disable-stripping --disable-decoder=libopenjpeg --disable-decoder=libschroedinger --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmodplug --enable-libmp3lame --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-librtmp --enable-libschroedinger --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxvid --enable-libzvbi --enable-openal --enable-opengl --enable-x11grab --enable-libdc1394 --enable-libiec61883 --enable-libzmq --enable-frei0r --enable-libx264 --enable-libopencv
  libavutil      54. 31.100 / 54. 31.100
  libavcodec     56. 60.100 / 56. 60.100
  libavformat    56. 40.101 / 56. 40.101
  libavdevice    56.  4.100 / 56.  4.100
  libavfilter     5. 40.101 /  5. 40.101
  libavresample   2.  1.  0 /  2.  1.  0
  libswscale      3.  1.101 /  3.  1.101
  libswresample   1.  2.101 /  1.  2.101
  libpostproc    53.  3.100 / 53.  3.100
Guessed Channel Layout for  Input Stream #0.1 : mono
Input #0, rtsp, from 'rtsp://192.168.42.124/unicast':
  Metadata:
    title           : LIVE555 Streaming Media v2014.07.04
    comment         : LIVE555 Streaming Media v2014.07.04
  Duration: N/A, start: 0.000750, bitrate: N/A
    Stream #0:0: Video: h264 (High), yuv420p, 1920x1080, 120 tbr, 90k tbn, 180k tbc
    Stream #0:1: Audio: pcm_alaw, 8000 Hz, 1 channels, s16, 64 kb/s
[segment @ 0x1f6a60] Codec for stream 0 does not use global headers but container format requires global headers
[segment @ 0x1f6a60] Codec for stream 1 does not use global headers but container format requires global headers
[mp4 @ 0x150ca0] Codec for stream 0 does not use global headers but container format requires global headers
[mp4 @ 0x150ca0] Codec for stream 1 does not use global headers but container format requires global headers
[mp4 @ 0x150ca0] Could not find tag for codec pcm_alaw in stream #1, codec not currently supported in container
Output #0, segment, to 'capture-%03d.mp4':
  Metadata:
    title           : LIVE555 Streaming Media v2014.07.04
    comment         : LIVE555 Streaming Media v2014.07.04
    encoder         : Lavf56.40.101
    Stream #0:0: Video: h264, yuv420p, 1920x1080, q=2-31, 120 tbr, 90k tbn, 90k tbc
    Stream #0:1: Audio: pcm_alaw, 8000 Hz, mono, 64 kb/s
Stream mapping:
  Stream #0:0 -> #0:0 (copy)
  Stream #0:1 -> #0:1 (copy)
Could not write header for output file #0 (incorrect codec parameters ?): Invalid argument
```

ZeroPoints, 2017-05-19:
I did write this script a while ago for ffmpeg if you really want.
https://github.com/ZeroPoints/fang-hacks/blob/master/data/etc/scripts/22-ffmpeg-record

keith721, 2017-05-19:
I'm attempting to use VLC 2.2.4 Weatherwax on Windows 10 to view/monitor the RTSP video stream from this camera.  Zenmap shows port 554 open, but nothing ever comes through.  VLC simply says it cannot connect.

Initiating SYN Stealth Scan at 21:09
Scanning Fang001(192.168.1.61) [1000 ports]
Discovered open port 22/tcp on 192.168.1.61
Discovered open port 554/tcp on 192.168.1.61
Discovered open port 80/tcp on 192.168.1.61

Everything has gone fine until this point.  Writing the SD card, loading the hacks, configuring the network, disabling the cloud apps, etc.   But VLC refuses to open the RTSP stream. :(

Your input can't be opened:
VLC is unable to open the MRL 'rtsp://192.168.1.61:554/unicast'. Check the log for details.  

Just wondering if I've missed anything obvious, thanks for any advice you might be able to provide.

YoureNowOnTV, 2017-05-19:
keith721

You don't need the port in the RTSP address for VLC.

In your case it'd be:

rtsp://192.168.1.61/unicast

keith721, 2017-05-19:
Thanks for that tip, but it doesn't matter whether I include the port number or not, VLC still won't open the stream.  I've logged into the camera and run netstat, and port 554 is definitely open and listening.  For what it's worth, this is one of the newer cameras with this MAC prefix: 
HWaddr 34:CE:00:XX:YY:ZZ

sbrinkmann, 2017-05-20:
@ZeroPoints I've tried the ffmpeg which you've provided through the link. It works but it also doesn't capture the sound :( 

So if anyone figured out how to use ffmpeg to record the RTSP stream including sound please let me know 😎

gerardovf, 2018-12-22:
@sbrinkmann I use this script in my Raspberry Pi (I get sound too):
[savertsp_xiaofang_stream](https://gist.github.com/gerardovf/e80d27a6669d2ea02a14bd5ed10d74e3#file-savertsp_xiaofang_stream)