samtap/fang-hacks #132: WLAN Settup via SD-CARD

a-zimmerer, 2017-05-19:
Quick question. I want to update my WLAN settings fast via SD-Card.
I have seen there is a wlan config file under /tmp directory.
Do I just have to edit that one and is it possible to just edit
the a wlan config file on the fat partion and copy that file with 
every boot onto the device. Maybe someone can point me in the right direction :-)

a-zimmerer, 2017-05-20:
Okay, i managed to code it by myself.
WLAN-Configuration is now fast and 
easy without Mi Home App and QR-Code
bullshit. 

I have uploaded all files and a little maunal:
https://mega.nz/#!FWoFQJ6a!fGu5laM8eow0uScZM6uerXr7DyuOCoLv-O5r2ikW-AA

Have fun with it :-)

prysme01, 2017-05-30:
it would be great !! I hate having to use apps to configure IOT :) 👍

puxxi2003, 2017-06-12:
Hi,
i bought a new Cam with 34.. MAC. :-(
I dont want to use the android app. In fear it updates the firmware and then things wont work.
Cam was/is on Firmware 3.0.3.56
So i used "a-zimmer"-files to setup the WLAN.

I connected the Cam to Power, LED was blinking blue, put in SD, waited for the sound and power off.
Pulled SD out, power the Cam on, LED was still blinking blue, but i saw in my router that the Cam was connected to my network.
Now i could connect to Cam and apply the hack.

**Many Thanks
So you could use Cam with 34..MAC, without ever using the app.
With the use of "a-zimmer" changes, you could not only update WLAN, you could install the whole hack.  :-)**

Kab14523, 2017-07-13:
Hi puxxi2003, 
did you change any code from the "a-Zimmer" files to make it install the fang-hacks? I used fang-hacks 0.2.0 and copied the "a-Zimmer" files in the FAT portion of the SD card. I used it on a out-of-the box xiaofang. It did connect to the WLAN but the IP/cgi-bin/status page or Telnet isn't available. However, IP/cgi-bin/hello.cgi does respond. I tried it with 2 different SD cards and repeated flashing but had no success.

denizk, 2017-07-29:
Just a hint for others that might struggle with the wifi connection. 
Make sure your editor doesn't add a newline after your wifi password and ssid.

```sh
echo -n "YOUR_WIFI_PASSWORD" > .wifipasswd
echo -n "YOUR_WIFI_SSID" > .wifissid
```

Worked like a charm on a couple 34xMAC cams. Thanks @a-zimmerer !

prysme01, 2017-07-31:
Could you please explain the step by step procedure to setup the wifi and password without the application ? it doesn't work for me.