samtap/fang-hacks #133: Questions regarding re-init of camera using SD card?

keith721, 2017-05-19:
https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-(factory-reset)

Many thanks for the wiki pages, but I can't tell that this SD card re-initialization is functional.  I downloaded the firmware, extracted it, copied it to the FAT partition, and pressed the reset button while applying power to the camera.  Because there's no visual or sound clues, I have two questions:
1) How long should I hold the reset button after applying power?
2) How long should I wait for the device to complete copying the firmware and reboot?

Thanks in advance for your help! :)

radepol, 2017-05-20:
check this out https://github.com/samtap/fang-hacks/issues/17

keith721, 2017-05-20:
That got my second camera back to working, THANK YOU!!!  I renamed the rescue config template file and it seems to have restored connectivity.  What I was doing before the situation arose, was changing the network config from Cloud to Wireless Client.  It went well, until I tried to reboot.  At that point, it was only a yellow LED and no other responses. :(

radepol, 2017-05-20:
http://bobbyromeo.com/technology/xiaomi-smart-1080p-wifi-ip-camera-rtsp-streaming-hack/

keith721, 2017-05-20:
There's a lot of different (and very helpful!) information on that web page.  To which particular item should I direct my attention?

keith721, 2017-05-20:
The same rescue config template file cleared up my first camera, as well.

closed 2017-05-20 by keith721