samtap/fang-hacks #135: snx_autorun.sh is never executed

xbyte99, 2017-05-20:
So, I unfortunately received a batch of the new cameras with the QR code on the bottom, which give a "this product is only used in mainland china" message after initializing them in the Mi Home (Android) app even with region set to Mainland China, regardless of the cam's firmware version.
Now before throwing them into the bin, I thought of giving your nice hack efforts a try, so I could at least receive an RTSP stream.
Lacking a current Linux system, I used [Win32DiskImager](https://sourceforge.net/projects/win32diskimager/) to write fanghacks_v0.2.0.img to an 8GB Micro SD-Card. Contrary to your description, the first partition is in FAT32 format instead of FAT16 (vFAT), but I guess that's what the image contains.
However, it seems snx_autorun.sh never gets executed after the insertion of the SD card. After some seconds the speaker burps some squeaky sound, signalling me that the card is recognized but accessing /cgi-bin/status on the cam still gives me the 404 not found error, while the cgi-bin/hello.cgi at least delivers a blank CGI output page.
My first try was with the already updated firmware version 3.0.4.9. As this didn't work, I did a full reset (pushing setup button for 15 seconds) reverting to the original firmware version 3.0.3.56. Did the basic setup with Mi Home again, then inserting the SD card, to no avail. Even tried it with the few updated post 0.2.0 files from the repository. And then tried to format the first partition with FAT16, placing the necessary files on it again afterwards.

And by the way, the blue LED keeps flashing, as it is never able to connect the Mi cloud, your description says to wait for it to wait until the flashing blue LED to stop (turn blue constantly), which for obvious reasons will never happen. I simply gave the cam some minutes until inserting the SD-Card, no luck though.

So I'm stuck and clueless. Am I missing something, or doing something wrong? What else can I check/do?

Any hint is very much appreciated.

CraigTRichards, 2017-05-22:
+1 on this. Bought 2 cameras and am having the exact same issue outlined above.

chris4269, 2017-05-23:
+1 on this. Bought 1 camera and am having the exact same issue outlined above.
help please

samtap, 2017-05-23:
- vfat is just how I'm used to name it, it doesn't care if it is fat16, fat32 or even exfat.
- The card is mounted by the OS regardless of hacks and cloud apps so it is quite robust. You should hear a hammer sound (bang bang).
- The hello.cgi is part of the default firmware so it's always there. It lists the contents of the fat partition on the sdcard, so it should show snx_autorun.sh and bootstrap and sd_record and some logfile (the latter two created by the cloud app).
- The 0.2.0 image has been used successfully by many, it's the easiest way to get going, no need to mess with anything else like firmware versions. It's been proven to work on latest cam.
- It seems in the new version the led keeps blinking. Ignore it. You can insert card as soon as it gets online.

There some people reporting issues where they don't even hear the hammer sound. That could be caused by a broken sdcard slot. Other than that I could not find any issues with the newer version so it should work fine.

CraigTRichards, 2017-05-23:
So I have re-formatted the card in fat32 and both devices mount it no problem. They will not mount when applying the image to the card with win32DiskImager. Strange...

Jono400, 2017-05-25:
Just wonder if you had any joy with the above " CraigTRichards ? looking to buy one off these cams myserlf but want to make sure everything would work smoothly with the hack !
is it still better to try get an older version off this camera or do they both require the hack to work in the uk ?

mukundv, 2017-05-26:
My issue is somewhat similar. I get the following at `http://device-ip/cgi-bin/hello.cgi`
`snx_autorun.sh`
`bootstrap`
`time_lapse`
When I go to 'http://device-ip/cgi-bin/status' i get a 404.

I used Etcher to image a 4GB card.

Any help is appreciated.

Jono400, 2017-05-26:
You guys having the issues with the camera hacks can I ask if it's the older or newer version off camera?

mukundv, 2017-05-26:
Turned out to be a dodgy microSD card. Ran `diskutil repairvolume /Volumes/SD` on the card and it fixed it. 
This time when I inserted the card, there was a sound and all good from there on.
[image]
[image]

@Jono400 as you can tell from the picture, it is a new camera and the firmware = 3.0.4.9

Jono400, 2017-05-26:
Okay thanks that's encouraging to hear ! will order one off the cameras from china and see what turns up at least I know that the hack works and it's not just a brick ! hopefully with a little help from you guys I can get the hack carried out !

xbyte99, 2017-05-26:
I can also confirm that with another SD card it worked for me now. In the end I didn't use Win32Imager anymore because it seemed to not to be working reliable on my system. I used a Live Ubuntu USB stick in the end to handle partition creating and extracting the files from the github archive.
Worked for all three cams now.
Thanks samtap for all your efforts!

closed 2017-05-26 by xbyte99