samtap/fang-hacks #136: How to access camera from the Internet and how to record on a NAS

leas1968, 2017-05-21:
Is it possible to access this camera from the Internet? Do I need to use port forwarding to access using VLC and how? 
Secondly. How can I record on a NAS?

keith721, 2017-05-21:
If you've not hacked the camera yet, I'd strongly advise you to block it from accessing the internet.  The default Xiaomi Xiaofang apps are heavily cloud-based, and communicate with servers outside the US.  If you have hacked the camera, I'd still advise you not to port-forward, because there is not yet any authentication whatsoever in the snx_rtsp_server program.
The best thing to do is setup a VPN into your LAN, and access the camera locally from within your LAN.  Otherwise, have it stream to a NAS or NVR and access the videos from there.  In short, allowing cameras to face directly to the internet (via port forwarding) is a very bad idea.  The recent Mirai and other botnets are constant attackers of IOT devices.

leas1968, 2017-05-21:
Thanks, Keith721. I live in Greece and purchased two of these cameras and can't use the mi home app because it says this product is only for mainland China:-( About more than a month ago I experienced a burglary in my apartment and decided to install cameras inside and outside to track movement and unwanted visitors! I am kind of disappointed because Xiaomi advertises this camera can record on Xiaomi cloud. It would be really helpful to record on the cloud or in any cloud services.

 I have also installed 4.0.11 version of MI Home but I see no video stream!  I have hacked one of them using the fang-hacks and can watch stream via VLC rtsp://device-ip/unicast but only from my PC inside my LAN. I tried the same through my Android phone but it freezes the device for about  a minute and does nothing!

Ok. I am not going to port forward the camera from my router to access it from the Internet.

Then, how can I record on a NAS? How can I setup a motion detection trigger and create a 1 minute video for example on my NAS? I have Zyxel NSA 325 V2 that I can access it from the Internet.

Thank you!

keith721, 2017-05-21:
You should install the NFS package from Zyxel and configure an NFS mount point for the camera.  From the camera, you should mount an NFS path pointed at that NFS mount point.  Then, use the FFMPEG program to write camera output to the NFS path.

leas1968, 2017-05-21:
Thank you. I will try it! Already installed the ffmpeg on my Zyxel. Very exciting! Thank you!

keith721, 2017-05-21:
ffmpeg is also available on the SD card from fang-hacks:

    /media # cd /media/mmcblk0p2/data/test/ffmpeg
    /media/mmcblk0p2/data/test/ffmpeg # uname -a
    Linux Fang001 2.6.35.12 #27 Thu Dec 22 18:48:16 PST 2016 armv5tejl GNU/Linux
    /media/mmcblk0p2/data/test/ffmpeg # ls -la
    drwxr-xr-x    2 root     root          1024 Jan 25 16:30 .
    drwxr-xr-x    5 root     root          1024 Jan 25 16:30 ..
    -rwxr-xr-x    1 root     root      16945088 Jan 25 16:30 ffmpeg
    -rwxr-xr-x    1 root     root      16889760 Jan 25 16:30 ffprobe
    -rwxr-xr-x    1 root     root      15315136 Jan 25 16:30 ffserver
    -rw-r--r--    1 root     root           140 Jan 25 16:30 hls_stream.sh

leas1968, 2017-05-21:
Thank you. I am kind of confused now. How can I mount an NFS path in my camera pointed to the NFS mount point on the Zyxel to write camera output? And how do I trigger a motion detection alarm in the camera to start recording?

keith721, 2017-05-21:
Look at the script named hls_stream.sh in the directory shown above.  Instead of using localhost as the DNS name, use the local IP address of your camera.  Then execute ffmpeg on the Zyxel NAS to pull the stream from the XiaoFang snx_rtsp_server:
    ffmpeg -i rtsp://IP-ADDRESS/unicast -c copy -hls_time 10 -hls_list_size 6 -hls_wrap 10 -start_number 1 /NAS/PATH/filename.m3u8

leas1968, 2017-05-21:
If I understand correct I have to run the ffmpeg on the Zyxel NAS using a telnet session and login as root. I did and didn't work. I get a message Segmentation fault (core dumped). I replaced IP-ADDRESS with the ip address of the camera and /NAS/PATH/ with the actual NFS path

keith721, 2017-05-21:
If you're running ffmpeg on the NAS, then you don't need NFS, because the storage is local to the NAS.  The only reason to use NFS is if you run ffmpeg on the camera, to direct the output to an NFS share on the NAS.

leas1968, 2017-05-21:
Ok. Thank you Keith721. But I need to make another dumb question since I am new to this. So, I edit the hls_strea.sh and edited as bellow. 

./ffmpeg -i rtsp://192.168.1.102/unicast -c copy -hls_time 10 -hls_list_size 6 -hls_wrap 10 -start_number 1 /192.168.1.10/i-data/c5bf47ad/nfs/xiaomientrance/stream.m3u8

I tried to run it directly through telnet and root user but I get an error "No such file or directory." The NAS has a password also. How do I pass it in the script?

keith721, 2017-05-21:
You can't use an IP address in the output path.  You should use a literal device name and directory path that are local to the server.   When you telnet into the NAS, and issue the command 'pwd', what is the output?  That will be your default working directory.  Then perform a 'df' command.  That will show all your logical devices.  Finally, perform a 'mount' command.  That will show all your mountpoints.  Somewhere between all that, we should be able to construct a valid path for you.

leas1968, 2017-05-22:
When I telnet into the NAS I login as root and the pwd command shows /root. When I issue the df command I get the following output:

Filesystem      1K-blocks       Used Available Use% Mounted on
/dev/mtdblock6      48896      44300      4596  91% /zyxel/mnt/nand
/dev/sda1          497829     490380      7449  99% /zyxel/mnt/sysdisk
/dev/loop0         138829     122660     16169  89% /ram_bin
/dev/loop0         138829     122660     16169  89% /usr
/dev/loop0         138829     122660     16169  89% /lib/security
/dev/loop0         138829     122660     16169  89% /lib/modules
/dev/ram0            5120          8      5112   1% /tmp/tmpfs
/dev/ram0            5120          8      5112   1% /usr/local/etc
/dev/ram0            5120          8      5112   1% /usr/local/var
/dev/mtdblock4      10240       1532      8708  15% /etc/zyxel
/dev/md0       1922351368 1033932680 888418688  54% /i-data/c5bf47ad
/dev/md0       1922351368 1033932680 888418688  54% /usr/local/zy-pkgs
/dev/md0       1922351368 1033932680 888418688  54% /etc/zyxel/zy-pkgs
/dev/md0       1922351368 1033932680 888418688  54% /usr/local/apache/htdocs/adv,/pkg
/dev/md0       1922351368 1033932680 888418688  54% /usr/local/apache/web_framew
ork/data/cache
/dev/mtdblock4      10240       1532      8708  15% /usr/local/apache/web_framew
ork/data/config
/dev/md0       1922351368 1033932680 888418688  54% /usr/local/mysql
/dev/md0       1922351368 1033932680 888418688  54% /usr/local/apache/htdocs/adv
,/res/imdb_poster

Then the mount command gives me this:

rootfs on / type rootfs (rw)
/proc on /proc type proc (rw,relatime)
/sys on /sys type sysfs (rw,relatime)
none on /proc/bus/usb type usbfs (rw,relatime)
devpts on /dev/pts type devpts (rw,relatime,mode=600)
/dev/mtdblock6 on /zyxel/mnt/nand type yaffs2 (ro,relatime)
/dev/sda1 on /zyxel/mnt/sysdisk type ext2 (ro,relatime,errors=continue)
/zyxel/mnt/sysdisk/sysdisk.img on /ram_bin type ext2 (ro,relatime,errors=continu
e)
/zyxel/mnt/sysdisk/sysdisk.img on /usr type ext2 (ro,relatime,errors=continue)
/zyxel/mnt/sysdisk/sysdisk.img on /lib/security type ext2 (ro,relatime,errors=co
ntinue)
/zyxel/mnt/sysdisk/sysdisk.img on /lib/modules type ext2 (ro,relatime,errors=con
tinue)
/dev/ram0 on /tmp/tmpfs type tmpfs (rw,relatime,size=5120k)
/dev/ram0 on /usr/local/etc type tmpfs (rw,relatime,size=5120k)
/dev/ram0 on /usr/local/var type tmpfs (rw,relatime,size=5120k)
/dev/mtdblock4 on /etc/zyxel type yaffs2 (rw,relatime)
/dev/md0 on /i-data/c5bf47ad type ext4 (rw,noatime,barrier=0,data=writeback,usrq
uota)
/dev/md0 on /usr/local/zy-pkgs type ext4 (rw,noatime,barrier=0,data=writeback,us
rquota)
/dev/md0 on /etc/zyxel/zy-pkgs type ext4 (rw,noatime,barrier=0,data=writeback,us
rquota)
/dev/md0 on /usr/local/apache/htdocs/adv,/pkg type ext4 (rw,noatime,barrier=0,da
ta=writeback,usrquota)
nfsd on /proc/fs/nfsd type nfsd (rw,relatime)
/dev/md0 on /usr/local/apache/web_framework/data/cache type ext4 (rw,noatime,bar
rier=0,data=writeback,usrquota)
/dev/mtdblock4 on /usr/local/apache/web_framework/data/config type yaffs2 (rw,re
latime)
/dev/md0 on /usr/local/mysql type ext4 (rw,noatime,barrier=0,data=writeback,usrq
uota)
/dev/md0 on /usr/local/apache/htdocs/adv,/res/imdb_poster type ext4 (rw,noatime,
barrier=0,data=writeback,usrquota)

I am attempting to run the ffmpeg command from a telnet session from the camera

keith721, 2017-05-22:
Okay, first things first.  You need to establish whether you will a) run ffmpeg from the NAS, and pull the video from the camera's stream, OR b) run ffmpeg from the camera, and push the video to the NAS's storage.  Frankly, the easiest way will be option (a), because option (b) requires creating an NFS mount point on the NAS, and mounting it on the camera.  Why do those things if it's not necessary?  That would be unnecessarily complex.

So, attempt to run the ffmpeg command from a telnet session to the NAS, pulling the video from the camera and storing it on the NAS storage.  If the camera's IP address is 192.168.1.102, then the ffmpeg command to pull the video from the camera is:
    ffmpeg -i rtsp://192.168.1.102/unicast -c copy -hls_time 10 -hls_list_size 6 -hls_wrap 10 -start_number 1 
and all we're missing is the final argument, the destination/output filename on the NAS storage.  Looking at your command outputs above, my guess for this value is:
    /i-data/c5bf47ad/nfs/xiaomientrance/stream.m3u8
So, just append that to the end of the command, and you should execute this command on your NAS:
    ffmpeg -i rtsp://192.168.1.102/unicast -c copy -hls_time 10 -hls_list_size 6 -hls_wrap 10 -start_number 1 /i-data/c5bf47ad/nfs/xiaomientrance/stream.m3u8

That should create at least one file in the directory, if the path is okay.

leas1968, 2017-05-22:
Thank you. I already run this ffmpeg command on the NAS and I get an error. Segmentation fault (core dumped), Perhaps I should uninstall and reinstall the ffp package on the NAS?

leas1968, 2017-05-22:
Ok. And another question relative to the thread. How can I create a VPN to access the camera from the Internet? Do I need a router that supports VPN?

keith721, 2017-05-22:
Generally, yes, you would need a home router that provides VPN services.  Most Asus routers do this.

Regarding the segmentation fault, you'd probably find something on the NAS forums for your device.  It's very likely there's another kernel module that needs to be loaded for ffmpeg to execute properly.

leas1968, 2017-05-23:
Thank you Keith. I will search this issue and try again.

keith721, 2017-05-23:
I also remembered [this very helpful web page](http://bobbyromeo.com/technology/xiaomi-smart-1080p-wifi-ip-camera-rtsp-streaming-hack/#remote_ffmpeg) which may help clarify it for you.

leas1968, 2017-05-26:
Thanks. The above command actually worked after updating the ffmpeg in the NAS using slack -a. But now as I understand this command continue until you close the telnet session to the NAS and there is no cycling recording. How do I specify 20 GB recording limit for example and then start over again? 

Thanks for the link. It is very useful. I will have a look!

keith721, 2017-05-26:
https://video.stackexchange.com/questions/17267/how-to-limit-file-size-with-ffmpeg

leas1968, 2017-05-27:
Thank you Keith. You are super helpful!

a1q123456, 2017-06-01:
@keith721 I can't mount nfs on camera, it gives me `Connection Refused` error message, but I can mount nfs using same command on my OS X with no error, could you please give me some advice?

keith721, 2017-06-01:
Use the 'lsmod' command on the camera to check which kernel modules are installed.  Chances are, you need to use the 'insmod nfs' command (or something similar) to load the nfs filesystem kernel module.

1043717432, 2017-06-03:
@a1q123456  Hi, did you succeed? I also want to connect Xiaofang to SynologyNAS. What are your good ways?

viorel-anghel, 2017-06-10:
xiaofang to synology is easy. have xiaofang with gang-hacks and streating rtsp. then, on synology install video surveillance application and add camera-> brand "user defined" (scroll up in that list) ->IP, port 554, source path=/unicast

leas1968, 2017-06-12:
Ani propositions for a VPN router? What about the TP-Link Archer D2 v1?  It supports IPSec VPN

maragelis, 2017-06-12:
Φιλε βαλε microtik οτι καλύτερο.  https://routerboard.com/RB2011UiAS-2HnD-IN
The best router you can buy.

viorel-anghel, 2017-06-12:
to access the camera from the internet, a less secure solution but much easier than VPN would be to change the URL "/unicast" to a random password-like string. then you can do port-forwarding on the router. to do this, on a camera with fang-hacks applied, you need to edit the file ...data/etc/scripts/20-rtsp-server and change the parameters for snx_rtsp_server startup adding something like "-u s3cr3tstr1ng; and  eventually -P 33554 for another port (to the line "snx_rtsp_server -W 1920 -H 1080 ... etc")

please be aware of the security considerations.

leas1968, 2017-06-12:
Mikrotik RB2011UiAS-2HnD-IN is a little bit pricey to use it just for two cheap cameras and I believe too much. I will examine it anyway.  A secure solution is important when it has to do with privacy. I am looking for a replacement of my ADSL2+ PSTN router with a VPN ADSL2+ PSTN router.  Unless of course the next version of the fan-hacks provides an authentication method.

J450NC, 2017-07-21:
Can I get some assistance with fang hack remote access pls.
I've set up VLC on iOS (phone) and rtsp with local address works fine.
I've set router to port forward local address to 80
When I try and stream http://private ip :80 I get nothing.
I've tried no firewall on the router - no joy 
I've tried putting the address in dmz of router - no joy
I've tried website to see if port is open - reports as open 
Anyone any ideas ?

viorel-anghel, 2017-07-21:
rtsp is on port 554 udp, not 80

J450NC, 2017-07-21:
I have rtsp working. See second line of post.
I'm asking for assistance for remote access http

louis-lau, 2017-07-30:
@J450NC There is no http access, only rtsp. You can port forward rtsp and use your public ip.

DON'T DO THIS if you don't have authentication set up on your rtsp server, for instructions look [here](https://github.com/samtap/fang-hacks/issues/6#issuecomment-314530036). [This](https://github.com/samtap/fang-hacks/issues/6#issuecomment-314940230) is currently the latest build and is working fine for me,

nightrope, 2017-11-24:
I have an issue getting the camera running on synology (DS218play DSM 6.1.4-15217 Update 2). I followed the instructions of viorel-anghel , which are:

> have xiaofang with  fang-hacks and streaming rtsp. then, on synology, install video  surveillance application and add camera-> brand "user defined"  (scroll up in that list) ->IP, port 554, source path=/unicast

If I click test, I get an ok sign. But when clicking next, I get the error message "failed to load camera capabilities values"
Any advice??

olskar, 2018-02-23:
I am having the same issue as @nightrope  :(

sanderjohan, 2018-04-10:
I also have the same issue with the error message. It will pass the connection test but when I hit the next button I get the "failed to load camera function" message.

Anybody an idea how to solve this?

gijsje, 2018-05-04:
I do see the same issue "failed to load camera function"
Any solution for this yet?

Edwin-Dijk, 2019-07-01:
> to access the camera from the internet, a less secure solution but much easier than VPN would be to change the URL "/unicast" to a random password-like string. then you can do port-forwarding on the [...]

@viorel-anghel Can you perhaps show me the edited line? I have no clue where in the startup method to put these parameters? Is this even remotely safe by the way? I know that it prevents the standard bots that check for the unicast but it is obviously brute force-able but this would have to be specifically targeted at your public ip