samtap/fang-hacks #137: File contents in /etc/config/*

ril3y, 2017-05-23:
Not quite an issue but I am working on getting the camera connected and working without using the MiHome app.  Can someone how has already activated their camera give me the file contents of all files in 

/etc/config/*
ls -lash as the files are hidden.  And of course you can edit your info so you don't paste your wifi creds here :)

Thanks

keith721, 2017-05-23:
/etc/config # ls -lash
     0 drwxr-xr-x    2 root     root           0 Oct 10  2016 .
     0 drwxr-xr-x   15 root     root           0 May 20 11:27 ..
     0 -rw-r--r--    1 root     root           2 Oct 10  2016 .Change_IQ_bin
     0 -rw-r--r--    1 root     root         137 May 18 20:45 .auto_record_info
     0 -rw-r--r--    1 root     root         172 May 17 20:36 .camera_encyid
     0 -rw-r--r--    1 root     root          73 Oct 10  2016 .nasConfig
     0 -rw-r--r--    1 root     root         129 May 20 11:14 .user_config
     0 -rw-r--r--    1 root     root          13 Oct 10  2016 .wifipasswd
     0 -rw-r--r--    1 root     root          14 Oct 10  2016 .wifissid
     0 -rw-r--r--    1 root     root           1 Oct 10  2016 .wifitype
     0 -rw-r--r--    1 root     root          38 May 17 20:39 appVer
    /etc/config #
    /etc/config # cat .Change_IQ_bin
    97
    /etc/config # cat .auto_record_info
    [INFO]
    enableTime=1495209599
    durTime0=1440
    startMin0=0
    startHour0=16
    date=1111111
    fullTime=1
    loopType=1
    segNum=1
    type=1
    switch=1
    owner=1
    /etc/config # cat .camera_encyid
    j9YQ7g+KpukGvKehhApeTSrbU+FOD6Ji/5E8yIkGNpYRFuuIR6PfslR5FJwmJ22jCoHbSEXR6JAmu1RZ8p+JttpTuOseRwIfFk7r9wfqwO37SP0/AoOJT8h5CUzI8+lpY/BOfMMJeNm6uNkGli1yphxzwwH3bPFFuoFy3FFsiZo=
    /etc/config # cat .nasConfig
    [NAS]
    transTime3=25
    transTime2=25
    transTime1=17
    transTime0=8
    validFlag=0
    /etc/config #  cat .user_config
    [VA_PARMS]
    Video_OSD=y
    NEW_OSD=1
    
    [ALARM]
    Alarm_Audio_Switch=y
    Alarm_Motion_Switch=y
    
    [CAMERA_INFO]
    P2p_UID=88888888888888888888
    /etc/config # cat .wifipasswd
    XXXYYYZZZAAABBBCCC
    /etc/config # cat .wifissid
    MySuperSecretSSID
    /etc/config # cat .wifitype
    5       <=== WPA2-PSK

closed 2017-06-09 by ril3y

350d, 2017-09-30:
Hello! So, it works without Mi Home App or not already? Mi Home app stop working in iOS 11 :)