samtap/fang-hacks #138: Connect to Cam with 3g

DerMilchShake, 2017-05-23:
Hey,

i hacked the cam and can watch the stream on tinycam, when i am connected to the same wifi.
But if im connected to a other wifi or 3g, it isnt possible.
How can i fix it?
I already tried it to release ports, but it doesnt work. I do it maybe wrong.
Can anyone help?
The cam is connected to a FritzBox, but my connection is on IPv4 and it use DS-Lite, maybe there is the problem?

Sorry for my english, it isnt so well, but i need help.

samtap, 2017-05-23:
You should ask your ISP how to forward IPv4 ports. Likely they'll have to change something on their end.

caffeineflo, 2017-05-24:
Fellow German here, so I know your problem... (although I'm no longer subject to it)

This is certainly not a problem with this project or the hack but an issue where your DS-Lite connection can't really forward ports as the IPv4 address you've been given comes back through IPv6.

Either get a real IPv4 address or do it properly through IPv6

samtap, 2017-06-06:
No other feedback, so I'm closing this as it has nothing to do with the cam itself.

closed 2017-06-06 by samtap