samtap/fang-hacks #139: ir-control script change day/night mode too late

korkin25, 2017-05-24:
Hi,

First of all - thank you for your hacks!!! It's great!!!:)

I have zoneminder working with camera.

the problem is ir-control script switch to night mode too late... 
I have second camera (with stock firmware) very near from first camera. same light and etc.
I see stock camera is already switched to night mode, but hacked camera is not.... then I see a lot of noise then zoneminder alarms.

I looked into your script but dont understand values for gpio_ms1/gpio_aud...

At the moment I corrected script to "always night mode", but I dont like it.

thank you!

samtap, 2017-05-28:
There's probably a setting to configure the light-sensor threshold but I can't find it anywhere :(. I don't fully understand the gpio tools either.

bronzerain, 2017-06-06:
I would like to have the IR LED's permanently disabled. But I still want the night vision to work. In one of the posts, i think it was about the fang overheating, there was a statement that the camera had to sense the LED's turn on before the IR filter can be moved so the filter is not used at night/low light conditions. If this is the case, then I will have to manually black out the LED's so they are still sensed and the IR filter can be moved to the off position.

bronzerain, 2017-07-10:
I took the camera apart and just covered the two IR LED's with tape. Now at night I can use the camera through a window without the glare from the IR LED's. I could not figure out the gpio_ms1/gpio_aud script so I could have a more elegant software code solution. 
But for right now the tape solution works for me.

timsayshey, 2017-09-15:
Is there no way to disable the IR LEDs via the ir-control script? 🤔

samtap, 2017-09-15:
gpio_aud write 1 0 0    # disable ir led and latch the filter in the correct position

You'll have to modify the script to your liking, but do keep track of power usage to make sure it won't get too hot over time. Or wait until I've finished the new script that supports config parameters to change behaviour.

danpowpow, 2019-01-22:
I have XiaoFangs in two locations which are plenty bright, but the camera/sensor itself is in slight shadow. As such they are mostly always in 'night mode', and their effectiveness is reduced as a result.
@samtap is your above post still valid? ie are you still working on a script that supports config parameters to change behaviour? 
If not can you guide me to where I might manually edit the threshold level of the IR sensor?