samtap/fang-hacks #140: Busybox Problem

darkblackside, 2017-05-24:
Hey, 

i've copied the files to the SD-Card correctly and i hear the smith-sound when i plug in the card.
But all the services are not working. I can visit the status-page.

On pages like /cgi-bin/action?cmd=expandfs; /cgi-bin/scripts and /cgi-bin/action?cmd=showlog

I get a message of BusyBox:
`BusyBox v1.22.1 (2017-01-13 21:01:25 CET) multi-call binary.
BusyBox is copyrighted by many authors between 1998-2012.
Licensed under GPLv2. See source distribution for detailed
copyright notices.

Usage: busybox [function [arguments]...]
   or: busybox --list[-full]
   or: busybox --install [-s] [DIR]
   or: function [arguments]...

	BusyBox is a multi-call binary that combines many common Unix
	utilities into a single executable.  Most people will create a
	link to busybox for each function they wish to use and BusyBox
	will act like whatever it was invoked as.

Currently defined functions:
	[, [[, addgroup, adduser, arping, ash, awk, basename, blockdev,
	bunzip2, bzcat, bzip2, cat, chgrp, chmod, chown, chpasswd, chroot,
	clear, cp, crond, crontab, cut, date, dc, dd, delgroup, deluser,
	depmod, df, dhcprelay, diff, dirname, dmesg, dnsd, dnsdomainname, du,
	dumpleases, echo, egrep, env, expr, false, fdformat, fdisk, fgrep,
	find, flash_eraseall, free, fsck, fstrim, fsync, ftpd, ftpget, ftpput,
	getopt, getty, grep, groups, gunzip, gzip, halt, hd, hdparm, head,
	hexdump, hostid, hostname, httpd, hwclock, id, ifconfig, ifdown, ifup,
	inetd, init, insmod, install, iostat, ipcrm, ipcs, kill, killall,
	klogd, linuxrc, ln, logger, login, logname, logread, losetup, ls,
	lsmod, lsof, lspci, lsusb, lzcat, lzma, md5sum, mdev, mkdir, mkdosfs,
	mke2fs, mkfifo, mkfs.ext2, mkfs.vfat, mknod, mkswap, mktemp, modprobe,
	more, mount, mt, mv, nc, netstat, nice, nohup, nslookup, ntpd, passwd,
	patch, pgrep, pidof, ping, pkill, pmap, poweroff, printenv, printf, ps,
	pstree, pwd, pwdx, readlink, reboot, reset, rm, rmdir, rmmod, route,
	run-parts, sed, sendmail, setserial, sh, sha1sum, sha3sum, sleep, sort,
	start-stop-daemon, strings, stty, su, sulogin, swapoff, swapon, sync,
	sysctl, syslogd, tail, tar, tcpsvd, tee, telnet, telnetd, test, tftp,
	time, top, touch, tr, traceroute, true, tty, ubimkvol, ubirmvol,
	ubirsvol, ubiupdatevol, udhcpc, udhcpd, umount, uname, uniq, unlzma,
	unzip, uptime, users, usleep, uudecode, uuencode, vi, vlock, watch, wc,
	wget, which, who, whoami, whois, xargs, yes, zcat - /etc/fang_hacks.sh: Running startup scripts
BusyBox v1.22.1 (2017-01-13 21:01:25 CET) multi-call binary.
BusyBox is copyrighted by many authors between 1998-2012.
Licensed under GPLv2. See source distribution for detailed
copyright notices.`

Is there a problem with busybox?

samtap, 2017-05-24:
You probably screwed up the symbolic links to busybox while copying files manually. Use the sdcard image instead.

darkblackside, 2017-05-24:
Such a simple error, working now!

closed 2017-05-24 by darkblackside