samtap/fang-hacks #145: 02-ntpd script does not start automatically

spacenode, 2017-05-27:
02-ntpd script does not start automatically. It seems that the script is unable to resolve time.google.com at boot.  When I replace time.google.com with the IP 216.239.35.0 in NTPD_OPTS then the script starts automatically.

samtap, 2017-05-28:
Maybe a temporary DNS issue with google, or your DHCP server does not provide a DNS server to the cam. Can you resolve other domain names?

closed 2017-06-03 by samtap

Nokius, 2017-07-29:
I had a similar issue as @spacenode for me it works well after I added a `sleep` >30.

davidjb, 2017-11-19:
I've started experiencing the same issue because the network isn't up yet at the point of `ntpd` being started.  The error that occurs is:
```
Starting ntpd...
ntpd: bad address 'time.google.com'
```
and `ntpd` exits at that point.  If you do as @spacenode did and use the IP address, `ntpd` stays running because it doesn't fail during the DNS lookup.  In my testing, however, `ntpd` didn't sync to network time even after a few minutes; I didn't look into why.

My solution was to delay starting `ntpd` until the network is up.  My changes at https://github.com/samtap/fang-hacks/commit/be1af3128c0d430bd028ea5dd873a4ade5dbe1d4 check `ifconfig` for receiving an IP address and then run `ntpd` thereafter.  This will only work on a DHCP setup so otherwise, a long `sleep` like @Nokius solution is an option (but may need tweaking if your network takes > 30 secs to come up.