samtap/fang-hacks #147: Can't disable cloud applications

bartwell, 2017-05-29:
Hi @samtap!
First, thank you for this project.
I get a problem, I can't disable cloud applications because checkbox is disabled. As I understand, I need to change connection type in Network. But:
1, When I try to change Wireless mode to Wireless client, there is no one available network in list (from smartphone I see 4 available WiFi networks).
2. When I try to change it to Access Point, it changed. But after reboot doesn't works. Just yellow led lighting and AP doesn't created. Thank's for recovery feature :)

Help please. What I can to do in this case?

Some additional info:
```
# cat /etc/config/appVer
[VER]
AppVer=3.0.4.9
IqBinVer=3.0.4.9
```

Contents of /tmp/hacks.log:
```
Wed Dec 31 21:00:05 GMT 1969 - /etc/fang_hacks.sh: Executing script (enabled: 1)
Wed Dec 31 21:00:05 GMT 1969 - /etc/fang_hacks.sh: Waiting for cloud apps...
Wed Dec 31 21:00:06 GMT 1969 - /media/mmcblk0p1/snx_autorun.sh: Running (device: mmcblk0p1)
Wed Dec 31 21:00:06 GMT 1969 - /media/mmcblk0p1/snx_autorun.sh: Created /tmp/www/cgi-bin
Wed Dec 31 21:00:10 GMT 1969 - /etc/fang_hacks.sh: iCamera is running!
Mon Oct 10 13:00:23 GMT 2016 - /etc/fang_hacks.sh: Boa webserver is running!
Mon Oct 10 13:00:23 GMT 2016 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/action -> /tmp/www/cgi-bin/action
Mon Oct 10 13:00:23 GMT 2016 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/func.cgi -> /tmp/www/cgi-bin/func.cgi
Mon Oct 10 13:00:23 GMT 2016 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/network -> /tmp/www/cgi-bin/network
Mon Oct 10 13:00:23 GMT 2016 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/scripts -> /tmp/www/cgi-bin/scripts
Mon Oct 10 13:00:23 GMT 2016 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/status -> /tmp/www/cgi-bin/status
Mon May 29 19:29:25 GMT 2017 - /etc/fang_hacks.sh: Running startup scripts
Mon May 29 19:29:27 GMT 2017 - /etc/fang_hacks.sh: Finished

![boot](https://cloud.githubusercontent.com/assets/1823669/26563432/a704f40c-44da-11e7-984b-525c516ab004.png)

![network](https://cloud.githubusercontent.com/assets/1823669/26563433/a70d90c6-44da-11e7-9eef-6215167aa746.png)

```

samtap, 2017-05-31:
Are there any weird characters in SSIDs of the networks in your area? There's a bug in the cgi scripts i.e. for SSIDs with spaces.
If you know your way around ssh you can manually configure it instead of using the webif. It involves changing wpa_supplicant.conf in /media/mmcblk0p2/data/etc and setting the wifi client mode in /etc/fang-hacks.cfg

bartwell, 2017-05-31:
No, here is no networks with spaces or non-latin/digits symbols. 

But thanks a lot, I successfully changed this configuration files using rescue telnet.

closed 2017-05-31 by bartwell