samtap/fang-hacks #150: Any way to physically remove/disable IR LEDs

bronzerain, 2017-05-30:
I do not need the IR LED's. I want to use the camera behind a window. Issue # 61 has pictures of the camera tear-down. The Yi camera had a IR ring in it and it could be disconnected. Has anyone taken apart the fang camera lens to see if the LED's can be removed/disabled. If removed will the camera still work?

EdByrnee, 2017-06-02:
Why not use the hack to disable the IR remotely?

bronzerain, 2017-06-05:
I think I would have to do this every time it gets dark and then disable it when it gets light. I just want the IR LED's off permanently so the camera can turn on its night vision on its own. I thought that the LED's still have to be turned on for the night vision to work. For some reason this was the order that the camera firmware needed. I am using the camera behind a window, so you see the LED's in the picture.

samtap, 2017-06-06:
Duplicate #99

closed 2017-06-06 by samtap