samtap/fang-hacks #152: 20-rtsp-server	NOK

bartwell, 2017-05-31:
Hello!
I get a problem: I can't run RTSP server. I always get "NOK" when I start it. Please, help me to run RTSP server.
Here is screenshots of configuration and logs:

[image] [image] [image]

**Process list**:	
```
  PID USER       VSZ STAT COMMAND
    1 root      1164 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [events/0]
    5 root         0 SW   [khelper]
    8 root         0 SW   [async/mgr]
  209 root         0 SW   [sync_supers]
  211 root         0 SW   [bdi-default]
  213 root         0 SW   [kblockd/0]
  223 root         0 SW   [khubd]
  226 root         0 SW   [kseriod]
  231 root         0 SW   [kmmcd]
  241 root         0 SW   [cfg80211]
  262 root         0 SW   [rpciod/0]
  269 root         0 SW   [khungtaskd]
  270 root         0 SW   [kswapd0]
  316 root         0 SW   [aio/0]
  323 root         0 SW   [nfsiod]
  332 root         0 SW   [crypto/0]
  399 root         0 SW   [mtdblock0]
  404 root         0 SW   [mtdblock1]
  409 root         0 SW   [mtdblock2]
  414 root         0 SW   [mtdblock3]
  419 root         0 SW   [mtdblock4]
  424 root         0 SW   [mtdblock5]
  429 root         0 SW   [snx-spi.0]
  438 root         0 SW   [zd1211rw]
  485 root         0 SW   [usbhid_resumer]
  518 root         0 SWN  [jffs2_gcd_mtd4]
  570 root         0 SW   [isp]
  591 root         0 SW   [mmcqd]
  648 root       796 S    /tmp/boa
  734 root      1160 S    /sbin/getty -L ttyS0 115200 vt100
  735 root      1152 S    /sbin/syslogd -n -m 0
  736 root      1148 S    /sbin/klogd -n
  737 root         0 SW   [RTW_CMD_THREAD]
  740 root      1088 S    wpa_supplicant -B -i wlan0 -c /media/mmcblk0p2/data/
  748 root      1164 S    udhcpc -i wlan0 -p /var/run/udhcpc.pid -b
  751 root      1152 S    ntpd -p time.google.com
 1840 root      1148 S    /usr/sbin/telnetd
 2035 root      1164 S    {status} /bin/sh /tmp/www/cgi-bin/status
 2041 root      1164 S    {status} /bin/sh /tmp/www/cgi-bin/status
 2042 root      1144 S    cat
 2051 root      1152 R    ps
```
**Mounts**:	
```
rootfs on / type rootfs (rw)
/dev/root on / type cramfs (ro,relatime)
none on /proc type proc (rw,relatime)
none on /sys type sysfs (rw,relatime)
none on /proc/bus/usb type usbfs (rw,relatime)
dev on /dev type tmpfs (rw,relatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,relatime,mode=600)
/dev/mtdblock4 on /etc type jffs2 (rw,relatime)
tmpfs on /tmp type tmpfs (rw,relatime,size=36956k)
lock on /var/lock type tmpfs (rw,relatime)
log on /var/log type tmpfs (rw,relatime)
run on /var/run type tmpfs (rw,relatime)
spool on /var/spool type tmpfs (rw,relatime)
tmp on /var/tmp type tmpfs (rw,relatime)
mq on /var/mq type tmpfs (rw,relatime)
media on /media type tmpfs (rw,relatime)
/dev/mmcblk0p1 on /media/mmcblk0p1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=cp437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p2 on /media/mmcblk0p2 type ext2 (rw,relatime,errors=continue
```
**Contents of /tmp/hacks.log**:
Wed Dec 31 21:00:05 GMT 1969 - /etc/fang_hacks.sh: Executing script (enabled: 1)
Wed Dec 31 21:00:05 GMT 1969 - /etc/fang_hacks.sh: Cloud apps are disabled
Wed Dec 31 21:00:05 GMT 1969 - /etc/fang_hacks.sh: Mounting /media/mmcblk0p1
Wed Dec 31 21:00:05 GMT 1969 - /etc/fang_hacks.sh: Starting boa webserver...
Wed Dec 31 21:00:05 GMT 1969 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/action -> /tmp/www/cgi-bin/action
Wed Dec 31 21:00:06 GMT 1969 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/func.cgi -> /tmp/www/cgi-bin/func.cgi
Wed Dec 31 21:00:06 GMT 1969 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/network -> /tmp/www/cgi-bin/network
Wed Dec 31 21:00:06 GMT 1969 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/scripts -> /tmp/www/cgi-bin/scripts
Wed Dec 31 21:00:06 GMT 1969 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/status -> /tmp/www/cgi-bin/status
Wed Dec 31 21:00:06 GMT 1969 - /etc/fang_hacks.sh: Failed to find hacks in /media/mmcblk0p2/data!
Wed Dec 31 21:00:06 GMT 1969 - /media/mmcblk0p1/snx_autorun.sh: Running (device: mmcblk0p1)
Wed Dec 31 21:00:06 GMT 1969 - /etc/fang_hacks.sh: Mounted /media/mmcblk0p2/data
Wed Dec 31 21:00:06 GMT 1969 - /media/mmcblk0p1/snx_autorun.sh: Created /tmp/www/cgi-bin
Wed Dec 31 21:00:06 GMT 1969 - /etc/fang_hacks.sh: Running startup scripts
Wed Dec 31 21:00:13 GMT 1969 - /etc/fang_hacks.sh: Finished

DavidRayner, 2017-06-01:
You must enable and run the stop cloud service before the RTSP service can be used.

bartwell, 2017-06-01:
Hi @DavidRayner! When I try to start 00-stop-cloud script I get `Cloud is disabled` message.

[image]

DavidRayner, 2017-06-01:
Whoops, I didn't see that you disabled cloud apps. I'm not sure what is causing the problem in this case, sorry

bartwell, 2017-06-01:
I've found a solution. Here is no permissions to execute snx_rtsp_server. I successfully fix it with this command and now RTSP works well:
```
chmod 766  /media/mmcblk0p2/data/usr/bin/snx_rtsp_server
```

And same with other scripts:
```
chmod 766  /media/mmcblk0p2/data/usr/bin/fang-ir-control.sh
```

I also set execute permissions for ftpd and busybox, but it doesn't work. It looks like missed arguments for busybox, but I am not sure.
```
chmod 766 /media/mmcblk0p2/data/bin/busybox
chmod 766 /media/mmcblk0p2/data/usr/sbin/ftpd
```

@samtap it looks like a bug. Is it possible to add auto checking permissions for scripts? And probably fix bug with busybox... Anyway thank you for your work.

samtap, 2017-06-01:
@bartwell All files you mention are executable by default so there's no bug. Did you flash the image or clone the repo and create your own sdcard? Usually what happens is people edit scripts on Windows and the execute flag is lost in transmission over (S)FTP. And sometimes sdcards get corrupted, which may be why you can't change attributes on busybox and ftpd?

bartwell, 2017-06-02:
> Did you flash the image or clone the repo and create your own sdcard?
Yes, I just create two partitions on SD card and copy files using Explorer on Windows 10.

> why you can't change attributes on busybox and ftpd?
I am changed it successfully. But it still doesn't works. I am trying to launch it manually `/media/mmcblk0p2/data/usr/sbin/ftpd` but it doesn't works (it shows BusyBox help).

samtap, 2017-06-02:
There're many things that can go wrong if you manually create a card. I suggest you try the image instead.

bartwell, 2017-06-02:
My bad, you are completely right. Thanks a lot.

closed 2017-06-02 by bartwell