samtap/fang-hacks #153: Can't display the time on rtsp stream .....

hoangthaibmvl, 2017-06-01:
`#!/bin/sh
PIDFILE="/var/run/rtsp-server.pid"

status()
{
  pid="$(cat "$PIDFILE" 2>/dev/null)"
  if [ "$pid" ]; then
    kill -0 "$pid" >/dev/null && echo "PID: $pid" || return 1
  fi
}

start()
{
  LOG=/dev/null
  echo "Starting RTSP server..."
  snx_rtsp_server -W 1280 -H 720 -Q 10 -b 5000  -a >$LOG 2>&1 &
  #OSD Date & Time
  snx_isp_ctl --osdset-en 1 --osdset-ts 1 --osdset-template 1234567890./-:Date --osdset-gain 2 --osdset-bgtransp 0x1 --osdset-bgcolor 0x000000
  echo "$!" > "$PIDFILE"
}

stop()
{
  pid="$(cat "$PIDFILE" 2>/dev/null)"
  if [ "$pid" ]; then
     kill $pid ||  rm "$PIDFILE"
  fi
}

if [ $# -eq 0 ]; then
  start
else
  case $1 in start|stop|status)
    $1
    ;;
  esac
fi

`
I config that like the introdution but it can't display the time pls help me
Thank you so much

samtap, 2017-06-01:
It probably can't find snx_isp_ctl because it is in the 'test' folder. Can you run it when logged in on ssh?
You can move it to a different location like /media/mmcblk0p2/data/usr/bin where it'll be found, or change the script to use the full path: /media/mmcblk0p2/data/test/snx_isp_ctl

hoangthaibmvl, 2017-06-02:
Hi when i run it show the error "/media/mmcblk0p2/data/etc/scripts/20-rtsp-server: line 37: snx_isp_ctl: Permission denied" so how can i fix that ?

DavidRayner, 2017-06-02:
run `chmod 777 /media/mmcblk0p2/data/test/snx_isp_ctl` in your SSH terminal.

hoangthaibmvl, 2017-06-03:
@DavidRayner i run the code you give me but it not work .....

hoangthaibmvl, 2017-06-03:
Ok i solved the problem thanks all

closed 2017-06-03 by hoangthaibmvl

takthetank, 2021-12-12:
@hoangthaibmvl I know I'm a little late bt how did you solve the `Permission denied` error?

hoangthaibmvl, 2021-12-13:
@takthetank it's a long time so I can remember exactly how I fix it. `chmod 777 /media/mmcblk0p2/data/test/snx_isp_ctl` you can try this command

takthetank, 2021-12-13:
I've tried it but it seems that I don't have access with ssh in mmcblk0p2 for some reason, so I get Permission Denied.

hoangthaibmvl, 2021-12-14:
@takthetank :D I can't remember how to do that. Sorry