samtap/fang-hacks #155: Web interface rework?

Protocol-IRK, 2017-06-03:
I remember reading something regarding a re-write of the web interface adding auth support and other nice features. I wondered what the progress with this was? If not I was going to take a bash at having a go with limited knowledge (any starting place/directions would be great) maybe try something like the openwrt/dd-wrt style interface?

samtap, 2017-06-03:
I have no intention to do a re-write myself but if you feel up to it go ahead and submit a PR. There've been a couple of others who wanted to do it but I haven't heard about any progress.

psmanek, 2017-06-10:
@Protocol-IRK what for? Current interface is good enough.

mkusz, 2017-08-08:
I'm interested in this topic and will try to find some time to do some in that topic in near feature (today I hack my first camera and definitely will work on it, because it has a huge potential).