samtap/fang-hacks #156: Hi, ask for help, I've processed it in order, but it's still the following problem. Many services don't open.

1043717432, 2017-06-03:
Hi, ask for help, I've processed it in o
[image]
[image]
[image]

rder, but it's still the following problem. Many services don't open.

Protocol-IRK, 2017-06-03:
See the yes button near top of logo, press that to expand the sd card partition and wait. Takes up to 30mins or more on some larger cards. Once it's complete it will tell you it needs a reboot, do it too early and you will cause problems which will mean you will need to reflash sd card.

samtap, 2017-06-03:
Please read the other issues or ask for support on discord. After flashing the image all services are supposed to work regardless of filesystem expansion. If it doesn't work try a different sdcard.

closed 2017-06-03 by samtap

1043717432, 2017-06-04:
ok