samtap/fang-hacks #161: Change/upgrade/downgrade firmware

TDPFlorida, 2017-06-07:
I'm currently at 3.0.3.56 and 3.0.4.9 on another camera living in the US and have my (mi home iOS) account set up to be mainland China (and the hacks work fine)...can and how do you upgrade/change/load a firmware to any given revision? If you can, where do you find these files/firmware revisions?

In looking through these threads, it seems this is possible and I have tried many of the things listed at different times and including:

https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-(factory-reset)#via-sdcard 

and none seem to work (EXAMPLES include FAT/FAT32/exFAT 4GB SDCard. 

Can someone POST a complete DETAILED procedure to do this using the SDCard?
Thanks

ZeroPoints, 2017-06-07:
You mean this? https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-(factory-reset)

TDPFlorida, 2017-06-07:
Yes, I have tried that but only the SD Card part.

TDPFlorida, 2017-06-07:
Download firmware file "upd_isa.camera.isc5.bin.extracted" from: firmware v3.0.3.56
Extract the file and place the files in the root of your SDCard on the first (FAT) partition "mmcblk0p1"
Rename "0.elf" to "FIRMWARE_660R.bin"
Push setup button while powering up the device. The device should auto reboot after flashing. 

#2 I may have messed this up but up put them on the first partition that I could see from windows. 

#4 I tried toggling the switch, holding it down and several other variations. The light came on solid Amber and stayed that way for several minutes so it seems like it was doing something but then I checked in the firmware was the same. 

Any suggestions?
Thanks

samtap, 2017-06-16:
I'm also unable to flash firmware automatically (with serial console you can see decrypt fail error, I think the 0.elf can't simply be renamed it needs something else...). But you should be able to use the ssh/serial approach: https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-(factory-reset)#via-ssh--serial

TDPFlorida, 2017-06-17:
That would be awesome but if its NOT something you can easily do with the SDCard then its probably over my head. I have yet to have anyone give me the steps to do this (using the SDCard) but eventually I'll figure it out:(

TDPFlorida, 2017-06-17:
I can connect via ssh (winscp) but where do I go from there?
Thanks

zekje, 2017-06-19:
if it could help you , i think the procedure is :

format a SDcard ( less or egual to 4gb ) in FAT ( more capacity need vfat or extfat ... )
put all file ( 0.elf , xxx.cramfs , xxx.cramfs.swap and cramfs directory ) on the SD
rename 0.elf to FIRMWARE_660R.bin

put the SD in the xiaofang
push the button and let it pushed will plug power
firmware update is automatic ( like 2 minutes ) 

if you have soldiered serial port , you can see the output ( see my post bricked with orange led for an exemple )

adrianmihalko, 2017-07-08:
Anybody knows what url needs to be blocked to prevent firmware upgrades?

blam86, 2017-11-01:
I have no luck in downgrading the firmware.

I held the reset button with the pre-programmed SD card inserted then plugged in the power cable but nothing significant happen. I held the reset button for 2 minutes.

I accidentally upgraded the FW to  3.0.4.9. I am wondering if the downgrade trick and Fang hack will work on units from later batch.

My units has MAC addresses start with 34 and the CMIIT ID is 2016DP5557.

Thanks

TDPFlorida, 2017-11-01:
Same here and I ended up screwing up the unit when I tried to open it up and hardwire it:(

samtap, 2017-11-01:
I've done it countless times and it never fails. You have to make sure to use the correct FIRMWARE_669R.bin and a working fat32 formatted sdcard, of course.

khaliullov, 2017-11-03:
@samtap 

Thank you for your great work!

I am also having issue while unbrinking my device.

```
U-Boot 2011.09 (Nov 24 2016 - 03:23:06)

DRAM:  64 MiB
MMC:   MMC: 0
SPI FLASH: 16 MB
In:    serial
Out:   serial
Err:   serial

Partition Map for MMC device 0  --   Partition Type: DOS

Partition     Start Sector     Num Sectors     Type
    1                 2048          204800       b
    2               206848          307200      83
reading FIRMWARE_660R.bin

10234928 bytes read
## Update firmware Starting  ...
ERROR: decrypt check failed
## Update firmware End, rc = 0xFFFFFFFF
ERROR: update failed

Partition Map for MMC device 0  --   Partition Type: DOS

Partition     Start Sector     Num Sectors     Type
    1                 2048          204800       b
    2               206848          307200      83
reading FIRMWARE_660R_F.bin
8 dma read ecc error, crc_err = 0x00000080

** Unable to read "FIRMWARE_660R_F.bin" from mmc 0:1 **
reading FIRMWARE_660R_F.bin

** Unable to read "FIRMWARE_660R_F.bin" from mmc 0:2 **
sd_update_fail: no FIRMWARE_660R_F.bin in the sd
ERROR: update FIRMWARE_F.bin from sd failed
Hit any key to stop autoboot:  0
roofsr size = 0x6d3070
rootsr size check failed
loadkernel 0x00007FFC 0x1;bootm 0x00008000
## Booting kernel from Legacy Image at 00008000 ...
   Image Name:   Linux-2.6.35.12
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4041920 Bytes = 3.9 MiB
   Load Address: 00008000
   Entry Point:  00008040
   Verifying Checksum ... OK
   XIP Kernel Image ... OK
OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.
```
Also there is some problem with serial connection: connected only TX and RX, when I am plugging GND - output stops. Why this can happen?
I think device does not respond (or receive) on my keystokes.
Please help.

khaliullov, 2017-11-03:
Also, I could not connect to Camera via MiHome. After scanning QR code in logs:
```
RTL871X: IW_SCAN_THIS_ESSID, ssid=TP-LINK_5EE504, len=14
```
if connect directly to device to open its hostspot - it does not offer DHCP IP address.

Any ideas?

samtap, 2017-11-07:
@khaliullov I've had that ERROR: decrypt check failed on one of my cams, never figured out what caused it. Could you post md5sum of your FIRMWARE_660R.bin, and what's the current fw version? You can get around it by using the ssh/serial method: https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-(factory-reset)#via-ssh--serial

The serial console should work fine with rx, tx, gnd. You need gnd, are you sure you're not using the other pad (3.3v) by mistake?

The log trace is just a scan of all SSIDs at your location. Sometimes Mi Home doesn't work, just try again and be patient... Reboot camera in between retries.

TDPFlorida, 2017-11-13:
Can FW 3.2.0.30 be downgraded to a FW that supports FangHacks? What are the FW's that FangHacks works on?

Thanks

samtap, 2017-11-13:
Yes, downgrade to anything lower, 3.2.0.30 changed the way sdcards are mounted.

TDPFlorida, 2017-11-14:
Just a clarification...downloading even the latest works?

samtap, 2017-11-14:
@TDPFlorida I don't understand, downloading the latest?? Fang-hacks, once applied, works on all versions. But to apply it initially, you need a version < 3.2.0.30.

TDPFlorida, 2017-11-14:
Sorry about the confusion but I was away from my records…

 

Four questions…

1.

I had a camera (not sure what FW but NOT  3.0.3.56) and downgraded it to 3.0.3.56 but am not sure EXACTLY what I did. I would like to upgrade it to the latest (I prompts me to optionally upgrade the FW to 3.2.0.30) and try it again and document EXACTLY how I do it BUT I don’t want to do that if the latest FW will NOT allow me to downgrade. Can I downgrade from 3.2.0.30?

 

2. This is what I have…

v3.0.3.56              hacks ok:)

v3.0.4.9                hacks ok:)

v3.0.4.20              hacks do NOT work:(

v3.2.0.30              hacks do NOT work:(

Is this correct?

 

3.

Does it matter if the camera’s MAC address begins with…

28:                          Downgrade…don’t remember if downgrading worked, When HACKS are NOT enabled and Wi-Fi OFF, the Camera live feed through iOS works fine.

 

34:                          Downgrade…don’t remember if downgrading worked, When HACKS are NOT enabled and Wi-Fi OFF, the Camera live feed through iOS DOES NOT.

 

PS I know the second part DOES NOT apply to the hack but I was curious if you had any info?

 

4.

I have the .bin file to downgrade to 3.0.3.56 but where can I find the .bin file to downgrade to 3.0.4.9?

 

Thanks

samtap, 2017-11-14:
1. Yes
2. For me it works on all versions. Some people reported issues but never responded with appropriate feedback so can't reproduce. 
3. 34 MAC is not supported by mi home outside china. It's not relevant for flashing. 
4. No idea, but why would you need it? Just stick with 3.0.3.56 when it works.

TDPFlorida, 2017-11-14:
Thank you for the speedy reply and answering all my questions.

 

1. Thanks

2. I’ve just always heard it’s better NOT to upgrade the FW         

3. on MAC 34…I am able to use iOS MiHome 3.11 or 3.12 in US (according to something I read in your forum…still looking to get the live working when not on same network as camera)

4. I just want to use the latest FW that works.

 

Thanks

djmanosv, 2018-01-05:
any good samaritan to help me activate continous recording ??? I have been trying houurs ! 
Fat32 , fresh 16 gb sd card ! What else to do  **** help me !!

spot320, 2021-01-01:
hi trying to downgrade firmware v5.6.5.45 to 3.0.3.56. to install a hack, the camera does not do anything while holding the setup button and continues to work as usual. Tell me, is there any way to downgrade the firmware version to 3.0.3.56?