samtap/fang-hacks #164: no reaction

dawiddudek, 2017-06-16:
i connect XiaoFang to local wifi and i put sd card to slot but camera doesn't react
I re-formatted the card - Without reaction

What does not work properly.

samtap, 2017-06-16:
If you don't hear the audio (bang bang) then the sd-card cannot be mounted, which is usually caused by a defective card, or corrupt filesystem.

dawiddudek, 2017-06-16:
Yes i don't hear audio (specyfic Bang Bang)

But when I put Card to Computer i can all operation on this Card (format,
copy/past, repartition)

What do You mean filesystem? I repartition this Card - come back to NTFS
filesystem and install again image hack and this not work.

How can i come back oryginal firmware or hard reset (I turn off camera and
push setup and turn on i kept pushed this button to first audio communicat
but this not work)

zekje, 2017-06-19:
NTFS ?

try the first partition of the SDcard in msdos/fat , and the data partition in Ext2

samtap, 2017-06-19:
Some people manage to break the sd slot. There's a card-detect pin, if bent the wrong way then card isn't detected. You can open it up and use a needle or some other tool to fix it. 
A working card with a fat partition should always play the "bang bang" sound on insertion regardless of the files on the card / hacks applied / cloud apps running etc.

dawiddudek, 2017-06-19:
Ok, i change sd and check what's going on. If then isn't work I open it and
check slot. thx

Pozdrawiam / Best Regards

*Dawid Dudek*

tel. (+48) 604 443 886

skype: dudek.dawid

gg: 447774
mail: dawid.dudek@gmail.com
------------------------------

Please consider the environment before printing this e-mail

dawiddudek, 2017-06-29:
ok, tell me when I upload hack and format sd and want reupload. when i put sd to slot camera doesn't reaction.
how can i delete hack and reupload hack?