samtap/fang-hacks #165: Display too dark sometimes

denierlexiese, 2017-06-16:
I run the cameras in my dark attic, and sometimes the display is too dark to see virtually anything. Other times it's fine.  My guess is it's the threshold between the attic getting too dark, but not dark enough for the IR to turn on.  But perhaps it's something else.  Anyone else dealt with the display being so dark, it's mostly just a black image?

samtap, 2017-06-19:
One of the remaining mysteries of the cam is how the LDR threshold can be configured, if it can be configured at all... Meanwhile it may be a solution to leave it in night mode permanently?

denierlexiese, 2017-06-19:
samtap.   Thank you for your feedback. Let me try to putting it in night mode permanently.  Do you have the command or script I need to update.  You can be brief. I read most the scripts for the camera, and install a few extra myself from.  Thank you for all your hadwork!!!

http://bobbyromeo.com/technology/xiaomi-smart-1080p-wifi-ip-camera-rtsp-streaming-hack/

samtap, 2017-06-19:
The LDR gpio is read here: https://github.com/samtap/fang-hacks/blob/master/data/usr/bin/fang-ir-control.sh#L18
So if you replace that line by DAY=0 it'll stay in night mode

denierlexiese, 2017-06-19:
Beautiful. Love it. Thanks!!!