samtap/fang-hacks #166: Bricked with orange fixed led

zekje, 2017-06-18:
Hi,

i Use Xiafoang hack without problems
last day , no video on my nas, i saw that rstpserver on xiaofang was NOK

i have tryed to disable cloud in cfg file and badaboum, xiaomi xiaofang is now bricked ....
i only have the orange led fixed :(

i have tried to soldier an usb port, and i saw this (reboot in loop)

`U-Boot 2011.09 (Nov 24 2016 - 03:23:06)

DRAM:  64 MiB
MMC:   SD Card not detect
mmci_host_init error - -1

SPI FLASH: 16 MB
In:    serial
Out:   serial
Err:   serial
GPIO[2] is high
Hit any key to stop autoboot:  0
roofsr size = 0x6d3070
## Booting kernel from Legacy Image at 00008000 ...
   Image Name:   Linux-2.6.35.12
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    3038112 Bytes = 2.9 MiB
   Load Address: 00008000
   Entry Point:  00008040
   Verifying Checksum ... OK
   XIP Kernel Image ... OK
OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.
Linux version 2.6.35.12 (fedora@localhost.localdomain) (gcc version 4.5.2 (SONiX GCC-4.5.2 Release 2011-12-06) ) #27 Thu Dec 22 18:48:16 PST 2016
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00057177
CPU: VIVT data cache, VIVT instruction cache
Machine: SONiX SN98600 Development Platform
Memory policy: ECC disabled, Data cache writeback
CPU: found ITCM 16k @ ffff4000, enabled
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
Kernel command line: console=ttyS0,115200 root=/dev/mtdblock2 init=/linuxrc mem=64M isp=10M vc=12M vo=0M mtdparts=snx-spi:768k(uboot),3072k(kernel),7168k(rootfs),4096k(rescue),1024k(etc),256k(userconfig)
PID hash table entries: 256 (order: -2, 1024 bytes)
Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Memory: 64MB = 64MB total
Memory: 36836k/36836k available, 28700k reserved, 0K highmem
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
    DMA     : 0xffa00000 - 0xffe00000   (   4 MB)
    vmalloc : 0xc4800000 - 0xe0000000   ( 440 MB)
    lowmem  : 0xc0000000 - 0xc4000000   (  64 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .init : 0xc0008000 - 0xc0026000   ( 120 kB)
      .text : 0xc0026000 - 0xc04ec000   (4888 kB)
      .data : 0xc050a000 - 0xc0535440   ( 174 kB)
SLUB: Genslabs=11, HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Hierarchical RCU implementation.
        RCU-based detection of stalled CPUs is disabled.
        Verbose stalled-CPUs detection is disabled.
NR_IRQS:96
Console: colour dummy device 80x30
console [ttyS0] enabled
Calibrating delay loop... 200.29 BogoMIPS (lpj=1001472)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 512
CPU: Testing write buffer coherency: ok
[... 332 lines cut ...]
sysctl net.netfilter.nf_conntrack_acct=1 to enable it.
ctnetlink v0.93: registering with nfnetlink.
xt_time: kernel timezone is -0000
IPv4 over IPv4 tunneling driver
GRE over IPv4 tunneling driver
ip_tables: (C) 2000-2006 Netfilter Core Team
arp_tables: (C) 2002 David S. Miller
TCP cubic registered
NET: Registered protocol family 10
lo: Disabled Privacy Extensions
tunl0: Disabled Privacy Extensions
IPv6 over IPv4 tunneling driver
sit0: Disabled Privacy Extensions
ip6tnl0: Disabled Privacy Extensions
NET: Registered protocol family 17
lib80211: common routines for IEEE802.11 drivers
VFS: Mounted root (cramfs filesystem) readonly on device 31:2.
Freeing init memory: 120K
hub 1-0:1.0: /run/media/fedora/software/SN986_1.50_P2P_TUTK_050a_20160921_1712/snx_sdk/kernel/linux-2.6.35.12/src/drivers/usb/core/hub.c 3351: state 7 ports 1 chg 0002 evt 0000
hub 1-0:1.0: port 1, status 0503, change 0000, 480 Mb/s`

but after flash/boot , it reboot in loop :(

Can anybody help me ?
did my cam is realy died ?

samtap, 2017-06-19:
Can you upload a log that shows the reboot loop? Or does it start immediately after the last line? It appears the filesystem is mounted and all looks well, I can only guess at this point but maybe it fails to load the init (linuxrc) process? But normally there should be at least a kernel panic or some other error before it reboots.

zekje, 2017-06-22:
sorry i'm not at home this week .

yes, the device reboot in loop as you can show in reboot.txt ( after USB lines it restart by Uboot .... )

i have tried to flash firmaware (see the second txt ) but after flashing ( seems to be successful ? ) , it reboot again .... did my firmwarefile is not good ?

did my webcam is out ?

diegomtnz, 2017-06-25:
> Use at your own risk! It is still possible to break the device, to the point where you'll have to solder some wires to get access to a serial console, or worse. When cloud apps are disabled and WiFi mis-configured or offline you will not be able to access the device, not even by removing the sdcard. The scripts that disable cloud apps and set the network mode are located on the device itself. But don't panic, recovery is possible by placing the right files in the bootstrap folder on the sdcard. If you do find any bugs please let me know!

> 
mi cam is beek when disable cloud too
i only have the orange led fixed

i go to ubuntu
in terminal 

> sudo nautilus

> 
go to media/mmcblk0p2/data/etc/wpa_supplicant.conf

And edit with my wifi data:

> ctrl_interface=/var/run/wpa_supplicant
ctrl_interface_group=0
ap_scan=1

network={
	ssid="XXXXXXXX"
	key_mgmt=WPA-PSK
	pairwise=CCMP TKIP
	group=CCMP TKIP WEP104 WEP40
	psk="XXXXXXXX"
	scan_ssid="1"
	priority=2
}
``
config your acces point in:
WPA-PSK
CCMP TKIP
In microSD and power on cam
the orange led continue fixed
but in your router go to lan status to see ip cam
http://device-ip/cgi-bin/status 
and you can config cam

zekje, 2017-06-30:
after many times on my xiaofang, i have found the problem ..... it's me :p 

USB to serial FTDI provide 3,3v , perfect for serial connection, but not for the webcam
i have disconnected 3,3v, and power by USB (from the same PC for the same ground ) , and ont he serial terminal , it not reboot in loop , and i can connect with serial on the embeded linux ( to modify cloud=1 )
i have flashed firmware to start from 0, and after reboot, the led flash orange ( not stay ), no sound becouse speaker not pluged :p 
i can make wifi connection to the xiaomi app on my phone , and install hack ( already present .... firmware flash will not reset this ) .
my webcam is now online ....

i have an other problem with rstp server seem to hang after a while,  but this issue is closed

closed 2017-06-30 by zekje

marceloxlo, 2018-01-05:
I've acidently put it in "Download mode" without sd card... now the leds stays fixed orange too :( Is it bricked? how can i solve it?