samtap/fang-hacks #172: Two parallel RTSP Streams possible

thomas-schwieters, 2017-06-25:
Hello,

I have 3 Xiaomi cams and a Raspberry Pi 3. I use the raspi to capture the RTSP Stream and save it as TS file.
I'm using basic shell scripts to capture the rtsp stream and first store it as TS file and then convert the files to mp4 files.
This is working quite well.

Now my problem is, that when I try to access the RTSP Stream (currently configured for FHD resolution with 25fps) via VPN, the connection is not working very reliable.
The problem is the upstream of my internet connection. It's too low to provide the stream in such a high quality.

Now I would like to get 2 RTSP Streams of the camera. One which I can use for recording (high quality) and one which I can access also remotely (lower resolution and lesser fps).
I have already tried around a bit:
- I tried to open the snx_rtsp_server twice on the camera, but unfortunately this is not possible
- I tried to use ffmpeg on my raspberry to retrieve the RTSP stream and re-stream the camera stream with lower resolution and fps. But unfortunately the rapberry seems to be to slow to handle this.

Do you have any ideas on how I could achieve my goal?
Is it maybe possible to capture images from the camera maybe twice a second additionally while running the snx_rtsp_server? I could then use these images for my remote access...

snjoetw, 2017-06-25:
Give Shinobe video a try, instead of bash script. I'm using shinobe to record 1080@20fps and it also outputs a low quality stream which I use to integrate with home assistant

thomas-schwieters, 2017-06-26:
Hi,

Thanks for the reply.

I don't think it'll run smoothly on the raspberry. I already tried ZoneMinder and EyeMotion. Despite of not liking the webinterface they both didn't run well. I guess you need a full pc in order to run such a heavy application. And I would like to avoid that, because of the power costs.
However, I'll give it a try.

I the Wiki --> Feature ideas section. I saw exactly, what I wanted to have:
- Sub-stream (dual stream 1080p30 + 480p30)

I guess that would be the best implementation for my requirements. I'm wondering if there is already an idea, how that could be implemented.

thomas-schwieters, 2017-06-28:
Is there some other idea?

I think the best way to achieve this is to exchange the snx_rtsp_server with some other rtsp server, who is capable of providing two streams.
Does anybody has an idea if there is such a server?

I could also try something on my own, I would just need some tipps.

thomas-schwieters, 2017-06-28:
Could you explain, how to compile/build the snx_rtsp_server for the camera?

I had a look at the code and I might be able to do some code changes on my own. Maybe i'm able to add a second rtsp stream

ageorgios, 2017-06-29:
@thomas-schwieters  check
https://github.com/samtap/fang-hacks/issues/6

thomas-schwieters, 2017-06-30:
@ageorgios thanks for a link. I'll have a look