samtap/fang-hacks #174: Record to camera SD card?

pjpalomaki, 2017-06-27:
Sorry, this seems like a stupid question, but how do you get the hacked camera to record to the inserted SD card (which also has the FengHacks)? I'm using tinyCam Pro to view and record stream onto my mobile, but seem to find an option to record to camera storage? Is there another way to trigger this (i.e. Linux command via SSH or something?)

thomas-schwieters, 2017-06-27:
Hello,

In the Wiki --> external resources there is this link [http://bobbyromeo.com/technology/xiaomi-smart-1080p-wifi-ip-camera-rtsp-streaming-hack/#remote_ffmpeg](link).
It shows, how you can record on the device itself or from a remote device.
From my experience, the remote recording works better than the on device recording. I have 3 cams and a raspi. The raspi does the recording for all 3 cams.

thomas-schwieters, 2017-06-27:
One addition:

For remote recording I changed the option "-c copy" to "-c:v copy -c:a ac3". This will convert the A-LAW audio to a TS compatible audio stream. The recommended option from the link has no audio.

pjpalomaki, 2017-06-27:
Thanks @thomas-schwieters, will try it now!