samtap/fang-hacks #175: Unknown state - brick?!

knowingme, 2017-06-28:
Hi!

First of all thanks to everyone sharing knowledge about the Xiaofang WiFi camera.
These are the facts: I own a 34-MAC version which was sold with the hack installed and w/ sd card. 

When I first booted the device the blue light was flashing. I thought because it was in use, the camera is looking for the wifi which it was connected. So I reset the camera (long press setup after boot, wait til orange light flashes again, power off, repeat this procedure). I remember the camera was telling something chinese permanently. So I tried to connect through mi home app and connect to my wifi. The connection failed, I couldn't scan the QR. 
After this procedure the camera is in a state which I cannot define. I tried many things, that are descriped here (flashing firmware, tmpl etc.). A big problem for me is the missing confirmation something worked out or not.

When I currently power on the camera the orange led lights up, than flashes shortly and then remains off. I know, when I push the setup button in the short time the orange led flashes, the chinese voice starts but promptly stops. I can go through mi home app, but I'm not capable of scanning the QR. The red led at the sensor are lighten up. 

Not sure if the camera is broken, the mi home app (sometimes) prompted me to connect to the camera. But denies to scan QR or connect overall. Does this mean wifi is working?

Atm I very desperate. Sorry to bother you but I really can't help myself.

samtap, 2017-06-30:
I wouldn't trust it if it was sold with hacks and sdcard installed. Who knows what was done to it before you got it... The only reliable way to debug issues like these is attaching a serial console and figuring out what state it is in so you can try to fix it.

louis-lau, 2017-07-30:
@knowingme I know I'm a month late but I just had this issue as well. Solid led on startup, then flashing shortly, then off.
I solved it by holding the camera over the qr code, powering the device on, and pressing the setup button in the short flashing period. Went on connecting fine after that.

Maybe you solved it already, in that case I'll just leave this here for anyone else with this problem.

jgyoldi, 2017-09-06:
@knowingme , I have exactly the same problem you describe. It happened after I tried to downgrade from  3.2.0.30. 
Then I tried what @louis-lau says without success. 
Then I connected serial but ... as @samtap indicates in post #185, they have changed root password. I am really desperated and I don´t know what to do. Is there any possible workaround to get back to life my XF?

jgyoldi, 2017-09-06:
@samtap if it can help, I have the log file of the loop.
@knowingme, did you try with @louis-lau solution? Did it worked for you?

knowingme, 2017-09-07:
I appreciate @louis-lau solution, couldn't try it because I desperatly sold the cam!

jgyoldi, 2017-09-07:
Thanks @knowingme. 
I tried @louis-lau method several times with no results. I don´t know what else can I do....