samtap/fang-hacks #176: Time is off by -x hours when ticking 'Disable cloud applications'

Pauperx, 2017-06-28:
Hi people, I have a question. I have the script up and running in Wireless Client mode. Once I click ‘Disable cloud applications’ on the Status page and I reboot the cam, the time on my cam screen is off by -2 hours, while on the Status page the time and timezone show correctly. Any ideas what might be going wrong here? Will I not be able to check the ‘Disable cloud applications’ box now?

Kind regards, Pauper

Pauperx, 2017-06-28:
Problem solved by applying this code: 

https://github.com/samtap/fang-hacks/issues/78#issuecomment-307811377

Thanks to kamel83! Great work, bud!

closed 2017-06-30 by samtap

madmax2, 2021-05-05:
@Pauperx I applied that code but why now my rstp stream stop working
and also I can't ssh into the camera.
Also the status page is now showing 404..

And idea how to fix this?