samtap/fang-hacks #177: Confirmed : unhackable version

gistsnippetss, 2017-06-29:
I have 3 units , the one with the famous mac starting with 34 simply refuse to comply .
Tried everything ,even upgrading to 3.04.20 and downgrading to 3.03.56,all clients ,several microsd cards,nope.
Sometimes we must accept that no means no.

wekii, 2017-06-30:
https://github.com/samtap/fang-hacks/issues/122#issuecomment-309304465

gistsnippetss, 2017-06-30:
More precise:
-3 units  ,2 infamous 34 mac
- 2 working  ( 1 +1 with 34 mac)
- tried all possible combinations on the 3rd , it was on 3.03.56 , tried with both 4.0.11 and the upgrade ,nothing ( the other 2 working) in the same conditions
- multiple cards used ,confirmed working in the other 2
- i even upgraded and downgraded a working cam ,only to be sure i am not missing something
-tried even the wlan hack ( worked on the other 2,thank you coder) ,nothing

My wife touched the not working one. This could be the issue #nojokeiamdeadserious

samtap, 2017-06-30:
I had no problems at all using the hacks on a version with 34 MAC, as commented on #122. Can you explain exactly what doesn't work? Does it not accept the sdcard (play the hammer sound?). Does it not run snx_autorun.sh? If there's anyone with a non working cam that has serial console access I could provide some more help, please mention me on discord so we can try to get to the bottom of this.

gistsnippetss, 2017-06-30:
As i said, i have another 34 mac , everything went smooth . The new one with 34 mac is different.
- i hear the sound 
- it reach the router for a second and get an ip but after the connection is closed 
- not responding to ping and not reachable in a browser,of course
- during setup i can briefly   see the video stream in ad-hoc mode , connected on it's wifi ( isa something )

I think something is dropping the connection immediately after ip asigning

samtap, 2017-06-30:
Maybe there's an issue with signal strength? It's not unusual for devices to get a connection up, and then dropped as soon as some data is flowing through. Of course it's possible for the cloud app to drop the connection on purpose but I feel it's unlikely. And if it does, we could still get around that by preventing the cloud app from starting. It's just not user friendly because you can't do it with the web-interface.
If you're familiar with shell scripting you can put stuff in snx_autorun.sh to debug. I.e. print wifi details to a logfile periodically, let it run for a whlie, check logs on sdcard with a pc.

gistsnippetss, 2017-06-30:
I tried even that , moving near the router ,the behaviour is indeed typical.
Another mention for the blinking blue light ,even in adhoc mode and having a stream ( no hack applied) the led is blinking blue , never steady.
Let's do the thing ,i'll post the log.

batmankule, 2017-07-11:
same for me. have the 34mac with button.
Tried everything.
cannot downgrade firmware (nothing happens after applying:
https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-%28factory-reset%29#via-sdcard)
cannot apply fang hacks - 404 page not found every time, tried 4 different micro-cards, ubuntu copy....
then i upgrade new firmare 3.2.0.30, now got the 1/3 everytime i tried to connect to camera
Have MiHome 4.0.11 all the time
i cant do anything...
its driving me nuts...

gistsnippetss, 2017-07-11:
I received another one ,also mac34 , everything worked smooth on this one.
I can only think that the one not working is a  'somehow ' defective unit.
What i learned playing with 3 working and 1 not working units :  
- always check everything , if the image is burned ok ( 3 out of 30 times it wasn't)
- some cards works  ,some not  ; sometimes the ones not working works ,and viceversa 
- sometimes ( way too often) is a server issue ,not yours ; try tommorow
- if the issue persist , most likely is not fang-related .

lordneon, 2017-07-14:
@batmankule 

Try editing snx_autorun.sh and putting a "sleep 30" on a new line at the beginning.

I just had to solve this problem on a mates camera, essentially the script run too early, by putting a delay in it fixes the problem. Where as on my camera it was fine without.

Kab14523, 2017-07-18:
After reading this issue I discovered that all 4 of my xiaofang cameras (which didn't work with the standard hack) have a mac starting with 34. However, with lordneon's idea of adding a sleep 30 I could make 3 of the work. I will try another SD card with the 4th at some point.

lec668, 2017-07-22:
Hi,
I received the cam yesterday wirh the 34 mac.
I upgraded to last firmware  3.2.0.30 and application.
No trouble registering to mi server, and using the cam locally or with the cloud from 3G.
Trouble is, the hack does not seem to work. 
I got the double sound, then nothing.
I was not able to access the status page, nor cgi-bin/hello.cgi
Reverting to the 3.03.56 firmware enabled access to hello.cgi and listed the files on SD.
But no way to acess status

lordneon, 2017-07-22:
@lec668 Whilst on firmware 3.03.56 try putting sleep 30 on a new line at the beginning of snx_autorun.sh see if that fixes the problem for you.

If you can see the files on the SD card with hello.cgi, it shows that the sd card is mounting correctly.

pointbob, 2017-07-27:
HI. followed xiaofang setup (3.0.3.56 firm/ 4.0.11) but now just flashes blue. Even unplugged it overnight. Cannot connect to public wifi OUTSIDE of home (don't have 4g). Only if on wifi router at home is working. IS this being blocked by xiaomi?? HELP. (I'm in usa)﻿

lec668, 2017-07-28:
Hack is working fine, thanks. firmware 3.03.56 
"Outside of China" message appeared a few days ago

SpoKeys, 2017-09-22:
Managed to modify the MAC by adding in snx_autorun.sh as second line

SpoKeys, 2017-09-22:
Managed to change the MAC by adding in the snx_autorun.sh as second line:

ifconfig wlan0 down hw ether 28:6C:05:F1:BC:6F && ifconfig wlan0 up

(please use a different MAC to prevent duplicates in their system)

I must say there are more identification issues that prevent the camera to login on their system.

samtap, 2017-09-22:
The MAC on the wifi interface is irrelevant. They get it from nvram and possible compare with libusb IDs.

divampo, 2017-09-26:
@lec668, Hi, i have mac34 with 3.2.0.30 firmware, but i cannot downgrade it to 3.0.x.x. I'm using approach with sd card, but after restarting, my camera doesn't want to react on qr-code. Can you explain how you did it with your device, please? Thanks.

TDPFlorida, 2017-11-17:
SpoKeys...Originally the blue light flashed on my MAC 34... camera and I could see it when my iPhone was on the same network. From what you said, I thought that meant the live stream was NOT on iCloud. So I tried what you suggested to change the MAC (to a MAC 28... I have and turned off) and now the light is SOLID blue. I tried to live view the camera on MiHome 3.12 iOS (WiFi on and OFF with VPN) but neither the camera I had already paired (MAC 34) or the MAC address (MAC 28) I used worked. Any suggestions anyone??? Thanks