samtap/fang-hacks #182: Night vision distance

dawiddudek, 2017-07-05:
Hi i have a problem with Night vision. in discription for XiaoFang is 9m but when my camera goes to night vision i don't see anything. My room has 5m width.
Do i can regulate IR power in config?

wekii, 2017-07-06:
It looks that  you have camera with broken IR lights.

dawiddudek, 2017-07-06:
At night IR light its on (I see two IR led that candle )
How do i can check IR it is ok?

wekii, 2017-07-06:
Do you hear "click" when camera turns on or off night vision?

dawiddudek, 2017-07-06:
Yes i hear specyfi click when camera goes to night Mode and 2 IR led is then blink
[image]

wekii, 2017-07-06:
Then I don't have any idea what could be a problem.
Is this camera hacked?

dawiddudek, 2017-07-06:
Yes it is hacked. I change script from #61 issues and change wireless Mode
to wireless client  and check at night.

samtap, 2017-07-06:
Based on the first pic, I'd say the IR LEDs are off but the 2nd pic clearly shows they're on... If power supply is lacking LEDs will be not as bright. But maybe you're expecting too much of it, two leds are really not capable to reach 9mtrs (but they should be able to light your room, at least a couple of meters directly in front of the lens)

dawiddudek, 2017-07-06:
I know that 9 meters is the manufacturer's declaration but in the room from camera to the wall I have 3 meters
It seemed to me that it would brighten up. 

Looking at screenshots of other users, it seems to me that something is wrong.

wekii, 2017-07-06:
Did you had the same problem before hacking the camera?

dawiddudek, 2017-07-06:
I don't know beacouse i can run camera outside China

Pozdrawiam / Best Regards
Dawid Dudek
————————————
Write from my mobile

wekii, 2017-07-06:
Maybe you meant to say that you **can no**t run camera outside China ?
In wich country do you live?

dawiddudek, 2017-07-06:
Yes Yes can not (Write on phone with dictionary)
I live in Poland :-)

wekii, 2017-07-07:
Maybe you should try to remove hack to see if there's any difference.
You can use camera in Poland, but only when you are connected on you local wifi.
https://github.com/samtap/fang-hacks/issues/122#issuecomment-309304465

For connection from outside of your LAN you need to have xiaomi router.

louis-lau, 2017-08-01:
@samtap There's definitely something going on related to the hacks.

This is a pic (from the rtsp stream) with cloud apps completely disabled:  
[image]

This is a pic (from the rtsp stream) with cloud apps disabled after boot with the stop-cloud script:  
[image]

As you can see with cloud enabled the picture becomes fuzzier, but brighter.

davidjb, 2017-09-09:
I'm seeing the same result with my cameras as @louis-lau's results.  With the hacks enabled the visible distance is incredible short, unfortunately making the cameras unusable in the dark.  Suggestions of things to try would be great; otherwise I'll start poking around in the camera.

BlaY0, 2017-09-19:
My guess is that there are some camera host driver setting exposed through **/proc/isp** which are changed on iCamera app start. Check if there are any differences in those proc handle files between cloud-on and cloud-off mode.

davidjb, 2017-09-19:
I diff'ed all the parts of `/proc/isp` over the last few days and found that `/proc/isp/af/w0-sum` notably different.  With cloud apps enabled, the value is around `0x450` and cycles between `0x300` and `0x500`.  However, with cloud apps *disabled*, this value is steadily increasing over time. 

Tonight, it started around `0xf4fa` and has moved upwards and occasionally downwards, now being `0x9512` and still going.   I wonder if this correlates with the fact that I'm visibly seeing the night vision getting worse the longer the camera is on.  With cloud apps *disabled*, the camera started from cold boot with excellent IR -- the majority of my room was visible, and now at almost 30 mins later, it's almost entirely black. I can see the change slowly happening on the video stream and the `AF W0-SUM` values keep increasing (eg `snx_isp_ctl --afget-sum -?`).  What does this value represent and is it possible to set or affect that value?  `snx_isp_ctl` only has the ability to get/poll the value.

Otherwise, in the diff, I saw that `/proc/isp/iq/nrn` changed from 1 (with cloud apps) to 0 (without).  Anyone know what `/proc/isp/iq` and `/proc/isp/iq/nrn` or `/proc/isp/iq/nra` are?  Changing the settings seems to have little to no discernible difference.  Also, `/proc/isp/osd/1/{height,width}` get adjusted but I doubt that's it; and the same with lots of various parts of `/proc/isp/md/*` for Motion Detection.

halfluck, 2017-09-21:
@davidjb i'm impressed with your progress, it really seems like you might find an answer to this annoying issue. 

I've been rebooting my cameras manually every night time, which i believe kicks in iCamera on boot, this works really well for me and the IR mode stays stable and vision is good.

Has anyone tried a simple bandaid hack to fang-ir-control.sh that does a /sbin/reboot on Day Stare Change? this would save me from manually rebooting.

Thanks Rob.

halfluck, 2017-09-21:
for anyone that wants a quick fix for now, this is what I am doing

`#!/bin/sh

echo "IR Reboot script started"

DAY="$(gpio_aud read 2)"
    if [ $DAY -eq 1 ]
    then 
        echo "Daytime"
        IR_ON=0
    else
        echo "Its Dark"
        IR_ON=1
    fi

while :
do
        DAY="$(gpio_aud read 2)"
        if [ $DAY -eq 1 ]
        then
                if [ $IR_ON -eq 1 ]
                then
                        echo "It was Dark Now Its Daytime so Reboot"
                        /sbin/reboot
#                        IR_ON=0
                fi
        else
                if [ $IR_ON -eq 0 ]
                then
                        echo "It was Daytime Now Its Dark so Reboot"
                        /sbin/reboot
#                        IR_ON=1
                fi
        fi
        sleep 3
done
`

samtap, 2017-09-21:
It's quite silly to reboot and rely on iCamera to setup the hardware. Can't anybody figure out what iCamera does and the ir script doesn't do? Run something like find /proc/isp -type f -print -exec cat {} \; will print all isp settings.

BlaY0, 2017-09-21:
There's already a difference in quality if you power-on your camera when there's already dark. If you power it on during the day or when the lights are on, then when turning the lights off you get more grainy/noisy image. That's even with iCamera initialization.

So hacks enabled, cloud enabled, cloud apps stopped and powering camera on in the dark will give you good results (a little bit blurry/fuzzy image but without noise) and low bandwidth usage (under 100 kbps in 720p) while rtsp streaming. More grainy/noisy image gives you bigger bandwidth usage tho.

I played a little bit in cloud disabled mode, fiddling with sensor exposure and gain (with ISP AEC and AWB disabled) and I actually managed to get more light but wasn't able to reduce noise (the noise was horrible) thus bandwidth went haywire (from let say 300 kbps to 2500 and even more on a bw 720p image).

The next thing I'm planning to do is stracing that iCamera stuff...

BlaY0, 2017-09-25:
OK, stracing didn't get me anywhere but  I have found few starting points after that...

That /proc/isp/iq/nrn is actually quite important. That iq handles control image quality sub module and nrn is actually enabling some sort of fuzzy/blury overlay so the image gets smoothened which is good for the small bitrate. As I said, more noise gives you way bigger bitrate which just means bigger file or more bandwidth used without any quality gain.

Then there are handles for auto exposure submodule in /proc/isp/ae dir and these are adjusting automatically depending on intensity of the light source(s). But you can still play around with for example gain and as soon as you change the value of it, you see the result. At that point ae submodule is disabled even tho the enable handle still shows 1. If you want ae back, you have to disable it again via handle and then reenable it.

frol, 2018-08-14:
I have the same problem, but I have two XiaoFang cameras and I clearly see that they have a drastic difference in IR lights.

Here is what I have:

* Two XiaoFang cameras:
1. non-banned (with a hole Setup button), which I own for over 2 years now, and it has poor night picture quality.
2. banned for use outside of China (with a push-button Setup button), which I got about a year ago, and it has great night picture quality.
* fang-hacks with Cloud apps auto-stopped after a boot for both cameras, and in fact, I even tried using one SD card for both of the cameras to ensure that I have identical setup.
* I have additionally applied patches for fang-ir-control from @davidjb (namely, `/proc/isp/iq/nrn` gets set to `0x1` for night mode).

The older camera produces the type of picture shown in the first message in this thread. When I replace the first camera with the second one, I get a perfect picture. Next, I powered on both cameras and the older camera started producing a great quality night vision picture as well and I can see that IR light goes from the second camera (I can see a flashlight effect on the video from my old camera when I turn my newer camera). The old camera still has its IR indicators on and even can light some items in a range of less than 1 meter (while the newer camera lights up the whole room which is ~3x4 meters).

It looks like a hardware problem and IR lights in my old camera just die. :(