samtap/fang-hacks #184: scp-client for uploading to sftp chroot

Nokius, 2017-07-06:
i like to upload from the camera to an sftp chroot. But the hack is not including a sftp-client just the server part.

I checked the issues and searched online about the SDK and some resources but I found no helpful  information.

Could someone tell me how to get/use SDK or compile the sftp client.

I would be really happy to get this done 

Many Thanks in advance.

samtap, 2017-07-07:
You can use scp. I'll see if I can include a proper client like lftp in a next release or can you recommend a different client?

Nokius, 2017-07-07:
@samtap Many thanks for the quick answer.

I didn't know lftp till a few minutes ago but it sounds good after reading the documentation.

What would be an ETA for a new release? I know it's a stupid question.