samtap/fang-hacks #185: Firmware 3.2.0.30 - No SD card found - Bug or "feature"?

spxak1, 2017-07-07:
Hi all,
both my cameras were update to newest firmware 3.2.0.30 and both now fail to see the SD card.
I have tried formatting, new cards, different sizes, to no avail.
As if the firmware disabled the SD slot. 
It also reset the time zone. As such I am back to stock and most annoyingly without the ability to record anything on the sd cards.

Anyone else affected?

smilemaxLee, 2017-07-07:
Same version(3.2.0.30) and same symptom.
It is not possible to boot via SD-Card....

samtap, 2017-07-07:
I've upgraded one of my cams to the new version 3.2.0.30.

This update definitely contains more content than the previous, so it'll take some time to figure out what's changed. Same as the previous updates, it's copied over the current contents in flash (isn't erased before the update). This means the fang_hacks.sh script is not deleted, and I also didn't notice any changes in the other scripts that are essential for hacks to work (sdcard mounting, rc.local etc.)

For me, everything still works as expected. Note this cam already had the fang_hacks.sh script installed. Some things I noticed:
- Couldn't use the cam from Mi Home anymore (offline). Had to re-add it and it took a number of tries. It gets to the last stage but times out. I also didn't get the notification of a new camera being detected after pressing the setup button, had to add it manually. After a couple of reboots suddenly the notification appeared and pairing was back to normal? Possibly just a cloud server issue, seen it before on previous firmware
- The hello.cgi script no longer displays the SDCard contents. I don't think the cgi executable itself changed but if I remember correctly it requires some file in /etc/SDCard(?) to list the path it will show on the web-page. I'll have to compare with a different non-upgraded cam to figure it out.
- The hammer (dang dang) sound no longer plays when a sdcard is inserted. But it's still auto-mounted like before.

If anything, it seems they just made things less convenient (by leaking less info) but nothing that really prevents us from running our homebrew stuff on it!

eftibor, 2017-07-08:
Nálam sem működik, de a "ding-ding" hallatszik...

eftibor, 2017-07-08:
It does not work, but the "ding-ding" is heard ...

spxak1, 2017-07-08:
Is there any way to downgrade to 3.2.0.20 until we figure something out? No SD card means no real surveillance right when I am about to leave for holidays. Thanks.

samtap, 2017-07-08:
@spxak1 Mi Home allows recording to sd so why would they disable that? If you still have a way to access the device (telnet) you could try to resolve it manually?

spxak1, 2017-07-08:
Hi thanks for the reply.
Well Mi Home says "SD Not found" so recording is out of the question. My devices are now stock (after having messed up with it, formatted cards, rebooted etc). All I get is the motion detection that records in the cloud. SD is not detected (on either device). And btw my two cameras eventhough both look the same could not be more different (mac addresses especially are too weird, also one has different mac address on sticker).

So I would hope flashing the old firmware to get them to record using Mi Home for the time I'm away would be nice and when I'm back I'll look into it in more detail. 
Thanks again.

DuanWeiye, 2017-07-10:
Hi, I bought a new xiaofang camera with firmware 3.2.0.30, tried to install hacks but failed.
This is what i did:
I've downloaded the 2.0 image and wrote it into tf card;
Insert tf card into camera after the blue light stopped blinking;
Wait;
Heard beep sound twice;
Access Http://<camera ip>/cgi-bin/status and got 404 error, http://<camera ip>/cgi-bin/hello.cgi got nothing(blank page)

Did I missed sone steps?
Thanks.

batmankule, 2017-07-11:
i dont thing so.
i have the same thing, always get 404 error, tried 4 different cd cards
from yesterday, camera stop working, got the 1/3 connection, no metter what i do..
can't, downgrade firmware, nothing helps.
looks like this firmware locks the camera outside China.
Anybody have the same experience or any suggestions?

azadaffif, 2017-07-11:
DuanWeiye, i am facing exactly the same problem. Please share if you the resolution.

spxak1, 2017-07-11:
OK this is very weird.
I followed the instructions and flashed the stock 3.0.1.56 firmware as shown in the wiki. 
I then proceeded to set the cameras up with Mi Home. The camereras are found, the QR code scanned but Mi Home always times out. I tried 20 times. I uninstalled Mi Home, reinstalled, same. 

The cameras appear to connect to the wifi after the QR code is scanned but the blue LED keeps flashing.

So I prepared a hack SD card, and both cameras now work with fang-hack. Everything works fine. 
The LED is always on Orange however, not blue. 

So am I right to assume that there is also an issue with Mi Home now? I cannot get stock behaviour and cloud apps won't work so I am back to rtsp + linux motion on my server. Works ok (-ish, motion loses the stream every now and then).

Still curious as to what happened though? And why is the LED orange?

samtap, 2017-07-11:
I have a cam that refuses to mount sdcard after the update now as well. It's the same version as my other cam, that works fine. I'm reading similar reports on forums, some say it works, some say it doesn't. Very weird..
Did anyone already manage to obtain the firmware update files? Or capture the urls used to download them?

torukmakto, 2017-07-12:
i dont thing so.i have the same thing, always get 404 error, tried 4 different cd cardsfrom yesterday, camera stop working, got the 1/3 connection, no metter what i do..can't, downgrade firmware, nothing helps.looks like this firmware locks the camera outside China.Anybody have the same experience or any suggestions?
--

Exactly same here. I think the firmware must be locked after 3.2.0.30 .

samtap, 2017-07-12:
I'm 100% sure it works, there just appear to be some differences that require changes in my scripts to work reliably. 
- First boot the cam without sdcard and connect with mi home. It doesn't matter if you get the 'not supported outside china' error as long as you get it connected to wifi. Even if the pairing process times out, often it is already connected to wifi but can't access cloud servers for whatever reason. Get the IP address from network info in Mi Home, or your router wifi client list.
- Try to access http://ip-address/cgi-bin/hello.cgi to make sure you got the correct address. It should load and show an empty page with 'CGI Output' in the title
- Insert the sdcard, wait a couple of seconds and try to access /cgi-bin/status page, it should load.

spxak1, 2017-07-12:
The above worked fine for me after I restored the original firmware. The camera now works with the hack rock solid (many many thanks for your work!).
Only problem so far is the Orange LED stays always on. Minor. 
Oh, and there's no way to go back to Mi Home. It always times out. As such the cameras will stay on that firmware forever. 

Thanks again.

P.S. BTW my two cameras, although "identical" they couldn't be more different. Even the field of view is different (one is wider) and the MACs are completely different (one is actually different to what it says on the sticker). 
Finally, one of them, when connected via rtsp to motionEYEos (raspberry pi 2) it can report sensing data (a temperature of sorts, usually around 50 C). The other one gives nothing. Sorry for the out of topic.

torukmakto, 2017-07-13:
I'm 100% sure it works, there just appear to be some differences that require changes in my scripts to work reliably.

First boot the cam without sdcard and connect with mi home. It doesn't matter if you get the 'not supported outside china' error as long as you get it connected to wifi. Even if the pairing process times out, often it is already connected to wifi but can't access cloud servers for whatever reason. Get the IP address from network info in Mi Home, or your router wifi client list.
Try to access http://ip-address/cgi-bin/hello.cgi to make sure you got the correct address. It should load and show an empty page with 'CGI Output' in the title
Insert the sdcard, wait a couple of seconds and try to access /cgi-bin/status page, it should load.

-----

In my case , I've never load the hack before and i did  exactly  your instruction but failed. It's not working

404 Not Found

The requested URL /cgi-bin/status was not found on this server.

torukmakto, 2017-07-13:
//spxak1  

Can u please show me the way how to restore original firmware ?  I think i need to downgrade my firmware to solve my issue.

batmankule, 2017-07-13:
torukmakto, try this, i have the same problem...
works for me, i downgraded it with success :)
https://www.andryou.com/2017/07/11/setting-xiaomi-xiaofang-security-camera/
bus still 404 for fang hacks :(
still trying..

torukmakto, 2017-07-13:
/batmankule thank you for your kind link.  Really appreciate it. 
i tried the instruction so many times on that site , but also did so many fails. I think i'm using  iphone couldn't downgrade the firmware because it requires android MI Home apk 4.x.x

spxak1, 2017-07-13:
torukmakto

just follow the instructions here: 
[https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-(factory-reset)](url)

Make sure the firmware files goes into your first partition of your fanghacks SD card (the small FAT partition), not just any SDCard.

After you restore the original fimware, try to get MiHome to show the QR code and make sure the camera scans it well and gives you the long message in chinese. Then check your router to find that the camera has actually connected to the Wifi configured in the first step of MiHome and record the IP. You will need this IP to access the fang-hack webpage. Ideally you'd want the router to assign the same IP everytime (setup your dhcp server for this).

Like I said this restores the hack and cameras work fine, but NOT with MiHome like before. MiHome times out

torukmakto, 2017-07-14:
batmankule / spxak1 

I REALLY REALLY THANK YOU SO MUCH YOU GUYS EXCELLENT HELPNESS AND GUIDANCE. 

I downgraded my firmware  to v3.0.3.56 successfully. I found out stupidly copy the "cramfs-root” folder

under "bootstrap" folder. I REALLY APPRECIATE IT !!!

Frankman2, 2017-07-15:
Sorry, i have read the hole thread. But I can't downgrad from 3.2.0.20 to 3.0.3.56. I make it like it is discribed in the instruction. But if I test it with ".../cgi-bin/status" I every time get "404 Not Found" ... and the Firmware Version shows in Mehome  still 3.2.0.20. I have no Idea!

spxak1, 2017-07-15:
Are you sure you are on the right IP? After your downgraded did you use MiHome to scan the QR code? Does the camera appear on your router? Do you use that IP?

Frankman2, 2017-07-15:
Yes, the IP it the right one and the camera appear on my router.

Frankman2, 2017-07-16:
Every thing seems ok with the downgrad ...  I can hear the "hammer sound" by the hack ... but I get every time "404 Not Found"  with ".../cgi-bin/status" ... It is hopeless ...

batmankule, 2017-07-17:
try with another sd-card...

Firehand81, 2017-07-17:
Same problem here, after camera updated automaticly to 3.2.0.20 (Mi Home had self-enabled autoupdate), I can't connect to camera with Mi Home or use Fang hacks.

Tried downgrading ~50 times now (3 different memory cards, 16/32/64 gb Kingston/Sandisk/Transcend), once even taped weight over reset button and left it there for few hours, just to ensure that it stayed down, but still, no hope. Also tried downgrading with fang hacks flashed to memory card and just fat/fat32/exfat/ntfs formatted, just to try it.

Starting to kinda give up and just check in few months if there is any progress with 3.2 firmware, since downgrade seems impossible for this one.

martinbottanek, 2017-07-17:
I've been finally able to downgrade :)

1. made a sd card with fang-hacks 0.2.0
2. copied the older firmware to the root of the sd card
3. cut power to the camera
4. inserted sd card
5. i've enabled power, WHILE having the setup button pressed (all other variations didn't start firmware downgrade)
6. kept it pressed for at least 1 minute - no idea how long is enough
7. paired the camera with Mi Home

It works fine ... except Mi Home (stuck at 1/3) and the blue light is always on (can't figure out how to turn off the camera via software)

Frankman2, 2017-07-17:
No ... Unfortunately it does not work for me.

samtap, 2017-07-17:
Upgraded another cam today, same as I reported before: V3.2.0.30 works fine with hacks. Except for the bang bang sound and hello.cgi (which are not part of the hacks).

J450NC, 2017-07-18:
Be gentle first post, and not a programmer, I have three of these camera's 3x MAC code 34 ****** but they all behave differently !
The first doesn't have a press button for the reset - it has a hole - this works with the MiHome app and even upgraded to 3.2.0.30 with no dreaded 'only works in China error'
The second has a push button for the reset and I upgraded the software believing from my previous experience with the first camera this would be ok - I got the dreaded 'only works in China error'
That brought me here to this thread and website - although I don not profess to be a programmer I can follow the SD format protocols and writing image files or working out what goes in a root directory etc.
I created a downgrade microSD card to 3.0.3.56 and flashed the second camera. It now has 3.0.3.56 firmware but still reports 'only works in China error' and has the option to upgrade to 3.2.0.30 in the firmware update section.  It took several attempts but I think the key is the duration of holding the reset button after reapplying power as mentioned above.
The third camera also has MAC code 34 ***** but I never upgraded the firmware when prompted and thid one works with the MiHome app with no errors.

Summary:
Camera 1 - earlier vrs no button to reset (has a hole) MAC code 34 - currently 3.2.0.30 - working
Camera 2 later vrs button reset MAC code 34 - currently upgraded to 3.2.0.30 then flashed to 3.0.3.56 -
has region code error
Camera 3 later vrs button reset MAC code 34 never upgraded - currently 3.0.3.56 - working
 
My question/s are:
Is the second camera now flashed to a point where the China region error cannot be removed ?
If I use the fang hack would the MiHome app stop its remote functionality - appreciate that is the whole point to not use MI server to bounce/stream data - I'm looking at converting all three ultimately but want to try to recover camera number 2 so that if I want to revert to 'factory settings' I have the knowledge how to.
I am using IOS vrs of MiHome 3.19.0 set to mainland China 

With the different permutations I have it is quite easy to see why members are having difficulty as I have three cameras all behaving differently.

samtap, 2017-07-18:
I don't really care for Mi Home so the 34 mac blocks don't bother me either. My stuff is designed to work from sd-card with any original firmware.

In my experience, flashing at boot by holding the reset button doesn't always work. I haven't spent enough time on it but using a serial console you can read the output and some firmware updates are rejected because of a decryption error. I've had better luck flashing after boot, by executing 0.elf from sd-card (which requires access via telnet/ssh). Unfortunately, 0.elf is not a complete firmware image but only the rootfs, so reverting to factory settings is difficult. Not being able to do a reliable factory reset is a big problem and leads to devices being in all kinds of different states.

For discussing downgrade and region block please use #123, keep this thread about fang-hacks on V3.2.0.30

J450NC, 2017-07-19:
Thank you for the reply samtap - Apologies, I have reposted there.
I will try a few more attempts and post my findings there.

b1g1an, 2017-07-19:
Just received my first cam, 34 MAC with button reset and 3.0.3.56 on it from factory. 

Took ages for 4.0.11 app to accept my login but once in all went fine, no region lock and could see footage in app, network light solid blue. Inserted sd into cam got the double chime and all good so far, using tinycam for now. Next stop add to Home Assistant.

Many thanks samtap.

pierangelof, 2017-07-20:
Hello everybody,
just to report that my camera (an old model) had no issues with the new firmware. The hack is still working (it was already installed when the update was performed) and the SD card is correctly mounted. The only problem I have experienced is that it is not possible to record from MyHome because the SD is not found. Anyway, the motion alarm and the screenshots are flawless working.

I hope it helps.

@samtap: thank you for your work!

varoufitsa, 2017-07-22:
Recently updated my cam(mac adress 34:...) to 3.2.0.30. Since then everytime I'm trying to access hack page an 400 runtime error appears. 
I tried every method described here, without success(cgi output runs perfectly). 
Any advice?

samtap, 2017-07-23:
@varoufitsa HTTP 400 is a request error, I don't see how the update could cause that. Maybe the sdcard got corrupted or did you make any changes to the cgi scripts?

I've captured some more logging of latest fw version and noticed cloud app sometimes gets into a loop where it disconnects the wifi for no reason other than that it can't reach some server (perhaps related to NAS recording?), and doesn't restart udhcpc client, so the cam doesn't get an ip. Pressing the setup button fixes it, but also gets into pairing mode...
So for now cloud apps seem quite useless, failing to detect sd-card so no recording, and unreliable wifi. I suggest disabling it (with my scripts, wifi works fine). If you're unable to access the webinterface you can use the rescue scripts on sdcard.

varoufitsa, 2017-07-24:
@samtap How can i run rescue scripts?

TOLIS35, 2017-07-25:
hi ... i am a new one ... yesterday i took to my hands 2 of this little things (new version)... i try to setup the first one and immediately the firm updated to 3.2.0.30 so the CHINA message appears (((( . then i try to setup the second one.... i disable the auto update and i can see that the fiem is 3.0.3.56  BUT the same message appears... the MIHOME connected with camera ... i can see the ip on my router , the blue light blinks all the time... i dont know what to do now....... 
From what i read now:
1) i must make the 2 partition SD 
2) put the hack inside the first partition 
3)connect the camera to  the CGI-bgin/staatus/server to enable the hack.....
am i right so far????

pointbob, 2017-07-25:
Hi. Long time follower of your posts. I bought a XF small cube camera but screwed up and updated firmware to 3.2.0.30 (most recent) I cannot get it to connect to older Mihome apk but on the current one it locks me out with the mainland china only restriction. the blue light on the led always blinks. Clearly i have to downgrade firmware. I've followed your info on your blog but its a disaster. I simply cannot downgrade. I've tried 7 times now. help.

I created the mmcblk0p1 folder in my micro sd card (see attached image) and have the firmware folder in the root folder. That seems weird to me because you'd think you want the firmware files outside of a folder..but anyways..am I doing it right???

https://image.ibb.co/nyMTH5/IMG_20170724_200258.jpg

So I power on my xf cube...install an older vers of the mihome apk (4.0.11)..proceed with setup and damn..it won't let me get a solid blue light even though it shows the device is online....and when i try to check my firmware it shows again the most recent firmware on the xf cube...3.2.0.30 

so brutal. especially since the app shows the device online but when i try opening it in the app it wont show an image and still flashes blue and hangs when connecting 1/3.

is there a foolproof way to downgrade the fw? can you please make a video on your cell phone and just put on utube..i'm sure many would appreciate..any help appreciated

pointbob, 2017-07-25:
"martinbottanek
I've been finally able to downgrade :)made a sd card with fang-hacks 0.2.0copied the older firmware to the root of the sd cardcut power to the camerainserted sd cardi've enabled power, WHILE having the setup button pressed (all other variations didn't start firmware downgrade)kept it pressed for at least 1 minute - no idea how long is enoughpaired the camera with Mi HomeIt works fine ... except Mi Home (stuck at 1/3) and the blue light is always on (can't figure out how to turn off the camera via software)
--
"
I'm stuck at 1/3d too and it shows that firmware is still 3.2.0.30 Are you sure you actually downgraded because I have a 2nd xf cam with older fw and it does not get stuck. Also you say you copied the firmware to the root of the sd card - you mean you took it out of the mmcblk0p1  folder???

someone would be awesome to make a youtube video on how they downgraded the firmware via the sd card method AND maybe even the SSH serial method which looks insanely complex to my ancient brain...a video would really help..

the sd card method im trying is this one
https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-(factory-reset)#via-sdcard

b1g1an, 2017-07-25:
If you're just going to use fang hacks why are you bothered about whether it works in Mi Home?

pointbob, 2017-07-25:
Well I actually would like to get it to work in mihome because i just need to basic ability for motion detection and then notifying me since i'm always on free wifi at work and do not have a cellular plan

samtap, 2017-07-25:
I'll say it once more: I've never recommended screwing around with firmware since there's no reliable way to restore to factory defaults. The info related to flashing on the wiki is not mine but contributed by another user. I've tried it, and it doesn't work on all occasions. Running the update using telnet/ssh instead of flashing at boot works, but requires ssh/telnet and doesn't do a complete flash update. Holding the reset button during boot simply makes the bootloader look for update files. There's no point in holding it for more than a couple of seconds. Mi Home is terrible and often doesn't work. I can't do anything about that, Xiaomi has their own support channels so go there instead.

There's so much misinformation in this thread I don't know where to begin. Anyone reading it gets the wrong idea and that's how even more incorrect info gets out. So from now on I'll start deleting posts that are not related to Fang Hacks and firmware V3.2.0.30. Please don't be offended, feel free to post on other issues more applicable (or create a new one).

pointbob, 2017-07-25:
samtap; it's ok though. This is how people learn. Stumbling thru the dark. We eventually get it - not all of us are linear minded.

SadMoustache, 2017-07-28:
Hello to all,

My camera just arrived yesterday and sadly updated to the latest firmware MiHome has offered.
I don't know which firmware is currently on the camera, where can I check?
Tried to connect to the camera in MiHome remote and Ap mode as well, but both shows the mainland china error. Shouldn't the camera work Ap mode?

martinbottanek, 2017-07-28:
> lso you say you copied the firmware to the root of the sd card - you mean you took it out of the mmcblk0p1 folder???

2. Extract the file and place the files **in the root of your SDCard** on the first (FAT) **partition** "mmcblk0p1"

rgs1001, 2017-07-29:
Hello to all,

I got 6 cameras, 4 working with fang hack, 2 not.
5 Cameras had old Factory Firmware on it, 1not. Also it looked used...(Package). All 34 MAC's. Not got the 3.2.0.30  working, i have to use a RS232 to serial FTDI which i just orderd, but!!!

1 Camera with old Firmware i did not hear the sound when insert the card so i opened the camera and found that one of the connectors from the SD-Card-Board to the Main Board was not fully fitted!!
I fit the connection correctly and then the sound appeared when inserting the SD-Card using Sandsik Ultra 16GB and Hack worked. So not trust to the quality of everything.

kulokoy, 2017-07-30:
@rgs1001 what is the old firmware of your 5 cam? If its 2.8.3.5, could you please dump an image for me to try it if it will work on my other camera. I have a problem on my 2 cameras, one keep disconnecting and not reconnecting the other is connecting but cant access outside network. Tried many suggestions and look around but seems like my hope is shrinking to find solution. Many thanks.

rgs1001, 2017-07-30:
With old version i mean 3.0.3.56. I use MyHome 4.1.26 with update disabled. My Cameras are all visible over LTE in Thailand. Very important set Locale to Mainland China. And Language to English. Add device with the second last Chinese Letters Camera. Works seamless here. But what i really want to do is to apply the hack to the camera and apply a hack to Synology Surveillance Station cos Xiaomi Cameras are not supported there, and add the rtsp feed to Synology. so i not have to deal with MyHome.

rgs1001, 2017-07-30:
I always reenter the password of the WIFI, when i deleted the camera. The Scan image is not always the same. This i think is a useful hint.

zhernovskiy, 2017-08-03:
I can confirm that hack can't be installed on 3.2.0.30 out of the box. After inserting card with hacks i can hear dang-dang sound, can access http://device-ip/cgi-bin/hello.cgi and cant access http://device-ip/cgi-bin/status - error 404. But if i try http://device-ip/cgi-bin/hello.cgi?name=record, i can see listing of record directory of my SDcard. So, for unknown circumstances snx_autorun wont run or run incorrcetly. The led is blinking blue and never become solid blue.

The only way is to downgrade to  3.0.3.56 according to [wiki](https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-%28factory-reset%29#via-sdcard). After that hacks installed and works without any problem on 3.0.3.56
Afther that, you can upgrade cam to 3.2.0.30 and hacks will still be working fine.

snoerenberg, 2017-08-03:
Hi,
I captured the firmware download of the actual 3.2.0.30. So when someone wants to have a look into it:
http://www5.zippyshare.com/v/x4gK3Yhp/file.html

There you can see that the whole "hotplug" functionality and autorun behaviour is removed.
Within the "hello.cgi" you can see the new parameters like "name=..." in IDA Disassembler.

FolderComp: 
[image]

pierangelof, 2017-08-03:
Hi Aleksandr,
I have tried to downgrade my camera following the WIKI but without success. Please, could you clarify a couple of things?
- how many partitions and which type of partitions (e.g. FAT or FAT32) were in the SD-card?
- which type of SD-card have you used?

Thanks a lot for your help.

zhernovskiy, 2017-08-03:
I've used a card that previously was written within latest fanghack image over dd. So the first partition is fat32 and i've done all the things as mentioned on wiki. 
The card is 2Gb microSD.

samtap, 2017-08-03:
@snoerenberg How did you extract the img file? I've upgraded multiple cams and all still have hotplug functionality.

pierangelof, 2017-08-03:
Just to be sure: what have you placed in the root of the sd-card: the
entire directory
_12c424a07178dceedb4b05130f736757_upd_isa.camera.isc5.bin.extracted or only
the files content in it?

Thanks for your help

zhernovskiy, 2017-08-03:
@pierangelof Only content of folder and rename 0.elf to FIRMWARE_660R.bin

pierangelof, 2017-08-03:
Thanks a lot but most probably I do some mistake and it does not work.

- I have used a 2 GB SD-Card that previously was used for the hack;
- I have copied in the root of the card the 3 files and the dir
(cramfs-root);
- I have renamed the 0.elf file to FIRMWARE_660R.bin;
- I have inserted the card and switched on the camera by pushing the setup
button for some seconds while the cable is connected;
- the LED is yellow for some seconds and then starts blinking (yellow)

I really do not understand where is my mistake.

zhernovskiy, 2017-08-03:
@pierangelof  
1. Hold setup button
2. Power on camera(put in power cable)
3. Do not release setup button for 10-15 seconds. I'm not sure, maybe you can hold it 5 second or should hold more than 15 sec, you should try )

[49 more replies]