samtap/fang-hacks #187: copying the image but file not starting the hack

ragaimeena, 2017-07-08:
I am a little confused
I am trying the hack the easy way which is the WIn32DiskImager way
I have a new SD card
I place in the computer
I download the Zip file from here https://github.com/samtap/fang-hacks/releases/tag/0.2.0
I unzip the file which brings up the image
I open WIn32DiskImager
I point it to the letter of the SD card
I point it to the image file
I write it to the SD card
I place it in the camera after booting blue light etc..
then nothing happens
My confusion:
The manual method calls for 2 partitions on the the SD. The automated method creates only one partition and theere is no DATA file
Tried the manual method but unsuccessful
Please advise thanks

ragaimeena, 2017-07-08:
Contents of /tmp/hacks.log:
Thu Jan  1 08:00:04 GMT 1970 - /etc/fang_hacks.sh: Executing script (enabled: 1)
Thu Jan  1 08:00:05 GMT 1970 - /etc/fang_hacks.sh: Waiting for cloud apps...
Thu Jan  1 08:00:06 GMT 1970 - /media/mmcblk0p1/snx_autorun.sh: Running (device: mmcblk0p1)
Thu Jan  1 08:00:06 GMT 1970 - /media/mmcblk0p1/snx_autorun.sh: Created /tmp/www/cgi-bin
Thu Jan  1 08:00:09 GMT 1970 - /etc/fang_hacks.sh: iCamera is running!
Sun Jul  9 05:20:18 GMT 2017 - /etc/fang_hacks.sh: Boa webserver is running!
Sun Jul  9 05:20:18 GMT 2017 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/action -> /tmp/www/cgi-bin/action
Sun Jul  9 05:20:18 GMT 2017 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/func.cgi -> /tmp/www/cgi-bin/func.cgi
Sun Jul  9 05:20:18 GMT 2017 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/network -> /tmp/www/cgi-bin/network
Sun Jul  9 05:20:18 GMT 2017 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/scripts -> /tmp/www/cgi-bin/scripts
Sun Jul  9 05:20:18 GMT 2017 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/status -> /tmp/www/cgi-bin/status
BusyBox v1.22.1 (2017-01-13 21:01:25 CET) multi-call binary.
BusyBox is copyrighted by many authors between 1998-2012.
Licensed under GPLv2. See source distribution for detailed
copyright notices.

Usage: busybox [function [arguments]...]
   or: busybox --list[-full]
   or: busybox --install [-s] [DIR]
   or: function [arguments]...

	BusyBox is a multi-call binary that combines many common Unix
	utilities into a single executable.  Most people will create a
	link to busybox for each function they wish to use and BusyBox
	will act like whatever it was invoked as.

Currently defined functions:
	[, [[, addgroup, adduser, arping, ash, awk, basename, blockdev,
	bunzip2, bzcat, bzip2, cat, chgrp, chmod, chown, chpasswd, chroot,
	clear, cp, crond, crontab, cut, date, dc, dd, delgroup, deluser,
	depmod, df, dhcprelay, diff, dirname, dmesg, dnsd, dnsdomainname, du,
	dumpleases, echo, egrep, env, expr, false, fdformat, fdisk, fgrep,
	find, flash_eraseall, free, fsck, fstrim, fsync, ftpd, ftpget, ftpput,
	getopt, getty, grep, groups, gunzip, gzip, halt, hd, hdparm, head,
	hexdump, hostid, hostname, httpd, hwclock, id, ifconfig, ifdown, ifup,
	inetd, init, insmod, install, iostat, ipcrm, ipcs, kill, killall,
	klogd, linuxrc, ln, logger, login, logname, logread, losetup, ls,
	lsmod, lsof, lspci, lsusb, lzcat, lzma, md5sum, mdev, mkdir, mkdosfs,
	mke2fs, mkfifo, mkfs.ext2, mkfs.vfat, mknod, mkswap, mktemp, modprobe,
	more, mount, mt, mv, nc, netstat, nice, nohup, nslookup, ntpd, passwd,
	patch, pgrep, pidof, ping, pkill, pmap, poweroff, printenv, printf, ps,
	pstree, pwd, pwdx, readlink, reboot, reset, rm, rmdir, rmmod, route,
	run-parts, sed, sendmail, setserial, sh, sha1sum, sha3sum, sleep, sort,
	start-stop-daemon, strings, stty, su, sulogin, swapoff, swapon, sync,
	sysctl, syslogd, tail, tar, tcpsvd, tee, telnet, telnetd, test, tftp,
	time, top, touch, tr, traceroute, true, tty, ubimkvol, ubirmvol,
	ubirsvol, ubiupdatevol, udhcpc, udhcpd, umount, uname, uniq, unlzma,
	unzip, uptime, users, usleep, uudecode, uuencode, vi, vlock, watch, wc,
	wget, which, who, whoami, whois, xargs, yes, zcat - /etc/fang_hacks.sh: Running startup scripts
BusyBox v1.22.1 (2017-01-13 21:01:25 CET) multi-call binary.
BusyBox is copyrighted by many authors between 1998-2012.
Licensed under GPLv2. See source distribution for detailed
copyright notices.

Usage: busybox [function [arguments]...]
   or: busybox --list[-full]
   or: busybox --install [-s] [DIR]
   or: function [arguments]...

	BusyBox is a multi-call binary that combines many common Unix
	utilities into a single executable.  Most people will create a
	link to busybox for each function they wish to use and BusyBox
	will act like whatever it was invoked as.

Currently defined functions:
	[, [[, addgroup, adduser, arping, ash, awk, basename, blockdev,
	bunzip2, bzcat, bzip2, cat, chgrp, chmod, chown, chpasswd, chroot,
	clear, cp, crond, crontab, cut, date, dc, dd, delgroup, deluser,
	depmod, df, dhcprelay, diff, dirname, dmesg, dnsd, dnsdomainname, du,
	dumpleases, echo, egrep, env, expr, false, fdformat, fdisk, fgrep,
	find, flash_eraseall, free, fsck, fstrim, fsync, ftpd, ftpget, ftpput,
	getopt, getty, grep, groups, gunzip, gzip, halt, hd, hdparm, head,
	hexdump, hostid, hostname, httpd, hwclock, id, ifconfig, ifdown, ifup,
	inetd, init, insmod, install, iostat, ipcrm, ipcs, kill, killall,
	klogd, linuxrc, ln, logger, login, logname, logread, losetup, ls,
	lsmod, lsof, lspci, lsusb, lzcat, lzma, md5sum, mdev, mkdir, mkdosfs,
	mke2fs, mkfifo, mkfs.ext2, mkfs.vfat, mknod, mkswap, mktemp, modprobe,
	more, mount, mt, mv, nc, netstat, nice, nohup, nslookup, ntpd, passwd,
	patch, pgrep, pidof, ping, pkill, pmap, poweroff, printenv, printf, ps,
	pstree, pwd, pwdx, readlink, reboot, reset, rm, rmdir, rmmod, route,
	run-parts, sed, sendmail, setserial, sh, sha1sum, sha3sum, sleep, sort,
	start-stop-daemon, strings, stty, su, sulogin, swapoff, swapon, sync,
	sysctl, syslogd, tail, tar, tcpsvd, tee, telnet, telnetd, test, tftp,
	time, top, touch, tr, traceroute, true, tty, ubimkvol, ubirmvol,
	ubirsvol, ubiupdatevol, udhcpc, udhcpd, umount, uname, uniq, unlzma,
	unzip, uptime, users, usleep, uudecode, uuencode, vi, vlock, watch, wc,
	wget, which, who, whoami, whois, xargs, yes, zcat - /etc/fang_hacks.sh: Finished
Back

samtap, 2017-07-09:
If you use the manual method make sure the symlinks are preserved (or just don't use Windows to clone the repo). Its easier to just use the image, which contains both partitions if you write it to sd correctly.

ragaimeena, 2017-07-09:
first thank you for your prompt response. Please forgive my ignorance:
When I used Win32DiskImager The image is created on one partition only then there is another partition created that windows is asking to format so I dont know where the other partition that contains the data files, the only the SD is the one partition that has the Bootstrab files and the other script but no partition with the data files. Please advise thank you

samtap, 2017-07-09:
The data partition is there, but Windows can't access it. So don't format it, just insert in the cam after Win32DiskImager completes.

ragaimeena, 2017-07-14:
thank you it worked

closed 2017-07-17 by samtap