samtap/fang-hacks #189: Just Purchased Camera - looking to record to the sd card.

x106x, 2017-07-12:
I just received my camera yesterday and through quite a bit of messing around I eventually setup my home router with an open VPN tunnel to Hong Kong.  It allowed me to see an error on the MiHome app that said the camera is only usable on mainland China.  However, using the VPN finally allowed me to get the camera on my WiFi.

I'd like to get this camera recording to the SD card.  It lists version 3.0.3.56 is installed currently and the RTSP stream works fine.  The MiHome app says the camera is offline but it lists the firmware version 3.0.3.56 as out of date.  I've seen other messages here about newer versions not working so I don't plan to update.

I get the impression that some of you are using the MiHome app even though that doesn't appear to be an option for me.

Verbank, 2017-07-12:
I also received my camera earlier this week - 
- New 34 MAC
Attempted Mi Home Versions 4.0.8 and 4.0.11
Only Blinking Blue, Stuck on 1/3 in Mi Home 
Ill have to attempt a VPN tunnel in the next few days. Hopefully that will gain me access and finally be able to actually attempt the SD script. 

If you have any other tips or tricks not posted in other threads please post ! 

Cheers

x106x, 2017-07-12:
Long story short, in all my testing I just simply used my router (which I run DD-WRT on) to get me to Hong Kong.  Once my phone was on that VPN as well as the camera it seemed to do the trick and I could now at least get the camera on my WiFi.  From that point I was immediately able to install the hack.

I think after reading some more, I may not be able to do exactly what I want.  On a side note, a 32gb sd card takes forever to resize, I let it go overnight.

x106x, 2017-07-12:
I haven't finished setting this up but I didn't run into this link before and it appears to have what I want.
http://bobbyromeo.com/technology/xiaomi-smart-1080p-wifi-ip-camera-rtsp-streaming-hack/

closed 2017-07-12 by x106x