samtap/fang-hacks #190: New firmware: 3.3.9_0121

Filipowicz251, 2017-07-12:
Hi

Yesterday i've received my "Xiaomi mijia 1080P Smart IP Camera" I've read about "Fang Hacks" - so i was sure i can enable its "full power"

First of all i've tried to connect camera to my wifi network. There were problem because i've tried to connect it to 5GHZ network and it cannot be connected this way. But it finally worked. At the begining i've tried to use APP, which was downloaded after i scan QR code for application (attached in Guide shipped with camera) It was not successful (because of 5GHZ - i didn't noticed it back then) - then i've tried to download another app called "YI home" But i cannot connect camera with this app either (but i realised it was wrong wifi type) Then i go back to this application from guide (on my iphone - application name is in china - but i've used "google translate" and it is saying it is just "MI" application)

I've moved to second step: "Installing hack to sd card". I've used Win32 Disk Imager (attaching PrtScr)

After that i've insert sd card to camera but... there is no effect. No sound of inserted sim card (as it has been mentioned) Nothing... In Mi application i see - that sd card is being "seen" (attaching PrtScr)

What am i doing wrong ? Is this a matter of new firmware ?

Attaching Printscreens:
1. Two application, that i've downloaded (only this with name in china is working)
[image]
2. Information about SD cards
[image]
3. Information about SD card details
[image]
4. Information before running Win32 Disk Imager
[image]
5. Information after clicking Validate (before saving image)
[image]
6 Informationa after clicking Validate (after saving image)
[image]

Verbank, 2017-07-13:
great info thank you

samtap, 2017-07-13:
Xiaomi MiJia is a different device! This project is about the Xiaomi XiaoFang camera. I don't have a MiJia so no idea if it even uses similar hardware. Would be interesting to find out, if it's similar to XiaoFang maybe we can support it.

Filipowicz251, 2017-07-13:
Yeah, i've just noticed it is a different device :/ Samtap - how can i assist you with "finding out" if it is similar ?

samtap, 2017-07-13:
First thing would be find out which hardware it uses. If it is similar to XiaoFang there's a SNX98600 SoC and maybe the software is compatible. But most likely it uses completely different hardware. You could take it apart and take pictures of all the chips etc, see if you can find serial console access and take it from there.

theDoc5655, 2017-07-13:
i have a mijia too, and i want so hard to get RTSP video out of it.

Filipowicz251, 2017-07-14:
samtap - i would love to help - but i'm no "hardware" man (more software like to be honest) If there is no screw and i cannot use screwdriver - i'm doomed :) And this is the case (there are no screws in this camera and i don't know how to dismantle it)

Maybe there is someone else with Mijia 1080P camera (beware - they have also Mijia 360 camera and Mijoa car DVR camera - this thread is about "pure" 1080P Smart camera IP)  with proper skills to dismantle it ? Me and theDoc5655 will be most welcome. :)

And really, samtap (and maybe someone who can help us with our camera) - consider some crowdfunding/patronate or plain donation portal - there will be many, who will want to reward you for your good work! ;)

theDoc5655, 2017-07-16:
don't let die this thread pls :vanuatu:

samtap, 2017-07-17:
I'll leave the thread open, I don't have a Mijia cam but maybe somebody else can contribute

theDoc5655, 2017-07-17:
I've some Mijia atm but same problem of Filipowicz251 here

icyleaf, 2017-07-20:
@Filipowicz251 

Mijia camera is also made by XiaoYi camera, maybe it helps for you:

https://github.com/fritz-smh/yi-hack
https://github.com/niclet/xiaomi_hack

Deteorek, 2017-07-31:
Hi,
I found this, http://www.hkvstar.com/product-news/tear-down-xiaomi-mijia-360-1080p-pan-tilt-ip-camera.html . I hope it will help a little.

theDoc5655, 2017-07-31:
if this can help to make a custom firmware for the mijia 1080p standard i'll build you a statue!

Filipowicz251, 2017-07-31:
But this is for mijia 360 1080p. It is not the same as just mijia 1080p. I suppose the hardware is different - so the hack would be also different :/

niighthawk, 2017-08-01:
I also want a hack for the sxj01zm ( Mijia 1080p ) to get acces to RTSP

ferdydek, 2017-08-04:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

let me know if you need more hardware pics.

Filipowicz251, 2017-08-04:
@samtap what are chances, that you could make use of it and create Mijia Xack ?:)

samtap, 2017-08-04:
Chances are slim since it's not the same SNX chip as in the Xiao Fang, but this one: http://www.grain-media.com/html/8136S_8135S.htm. So the software is completely different :(

Xiaomi sure is busy putting new cheap cameras on the market lately:
https://www.gizmochina.com/2017/08/03/xiaomi-launches-new-1080p-ptz-smart-camera-%C2%A5149-22/
https://www.gizmochina.com/2017/08/03/xiaomi-releases-new-infrared-camera-motion-detector/

snoerenberg, 2017-08-04:
@Filipowicz251 is there any firmware available to download? So that it can be analyzed?
I think @samtap was able to find the "autorun" possibility while he was on the serial console.

If there is no possibility to get into console, it would be needed to extract firmware and see if there is any chance to easily start scripts (maybe with root priviledges).

Filipowicz251, 2017-08-04:
@snoerenberg  - nothing that I'm aware of. firmware is being downloaded automatically by the application. I didn't see any url with firmware to download :/

snoerenberg, 2017-08-04:
@Filipowicz251 I triggered the FW download in the Mi app and captured on my router (Fritz.Box) the internet connection. So I was able to see the HTTP calls which have been issued by the webcam. Than I downloaded the FW on my PC after analyzing the packets with wireshark.

Filipowicz251, 2017-08-04:
@snoerenberg  - so you have the firmware? great news :) Maybe post it here - so, that anyone (who knows how to do it:) ) can "try" to find hack?

snoerenberg, 2017-08-05:
@Filipowicz251 no from the Xiaofang I could capture the FW download. I ordered now a Mijia as well... will take some time to Germany.

niighthawk, 2017-08-05:
Good

niighthawk, 2017-08-05:
We should start a new project only for the mijia 1080p camera.

snoerenberg, 2017-08-05:
@niighthawk we'll see if there is any easy entry point to create a hack.

What is urgently needed? Rtsp? The camera should work outside China with Mi app or?

niighthawk, 2017-08-05:
I think everyone want it's a RTSP and put cloud system off like others hack, also an option to put the blue light off and also the night mode off.

theDoc5655, 2017-08-05:
I urgently need the support to RTSP, thanks guys :D

andreq, 2017-08-08:
Just want to chime in that I'll receive a Mijia in a month or so and will do my best to help here. Meanwhile, was there any github project created for Mijia camera specificaly?

snoerenberg, 2017-08-09:
Hi,
when you google for "GM8136 SDK release v1.0.rar" you can find some useful stuff.
I dig deeper into this once the camera is delivered.

Can someone in the meanwhile make a port scan on the Mijia?
Thanks

Links:
- SDK http://pan.baidu.com/share/link?uk=2553453276&shareid=3324610145
- Bootloader? http://pan.baidu.com/share/link?shareid=1783156365&uk=2553453276&fid=275039869366609

Stephan

snoerenberg, 2017-08-10:
Within the SDK is a compiled "rtspd" at least.
\GM8136 SDK release v1.0\Software\Embedded_Linux\image\GM8136_2MP.NAND.jffs2.img -> "rtspd"

Filipowicz251, 2017-08-10:
Hi

I've created github project for hacks for Mijia 1080p. 

https://github.com/Filipowicz251/mijia-1080P-hacks

I've created also a topic for this:
https://github.com/Filipowicz251/mijia-1080P-hacks/issues/1

snoerenberg, 2017-08-13:
@ferdydek can you try to find the UART (if there is any)?
I've not got my Mijia device yet.
I've a CP2101 USB-UART Adapter for testing this later on.

Maybe you can check it as well, following a nice summary how to find the right pins (hopefully).
http://www.devttys0.com/2012/11/reverse-engineering-serial-ports/

From my point of view, suspects could be the following:
[image]

snoerenberg, 2017-08-13:
Hi,
most promising look thos two points. Can someone attach a USB-UART Adapter there?
Try RX/TX and vica versa,

[image]

When you look at the traces they go straight to the CPU legs.
The others look like grounded pads.
It's just what I could see from @ferdydek pictures he took.

Thanks
Stephan

tobbegutt, 2017-08-13:
thank you all guys. ill follow this with great optimism.

andreq, 2017-08-14:
@snoerenberg, good find! I'm 99% sure the UART is the top most 2 test point on your 2nd photo. They seem to be attached to pin 86 and 87 on the chip, which are UART0_SIN and UART0_SOUT.

If I'm not mistaken, the left test point would be IN and right one would be OUT.

kollaesch, 2017-08-15:
I don't know if this is any news for you, but on on the SD-Card (for saving videos) there are logs saved from camera as well. (linux kernel log)
Let me know if you could use them for checking as well.

snoerenberg, 2017-08-16:
@kollaesch would be nice. Mine is still in transit. No one with an USB-UART Adapter here to check if a writable root shell is present? :)

Thanks

willthrom, 2017-08-16:
Mine is in transit as well. As soon as it arrives I will give it a go.

kollaesch, 2017-08-16:
@snoerenberg I have an USB-TTL at home. 
How to connect? I couldn't open the cam myself yet. What's the trick? (howto somewhere?)
The log follows this evening.

kollaesch, 2017-08-16:
@ferdydek How did you open the cam?
(with or without breaking the snapin-hooks)

kollaesch, 2017-08-16:
the logs from the boot of the cam are listed here:
[pastebin](https://pastebin.com/raw/53hdb4x0)

snoerenberg, 2017-08-16:
Log looks good:

Jan  1 08:00:02 kernel: [    0.000000] Kernel command line: mem=128M gmmem=90M console=ttyS0,115200 user_debug=31 init=/squashfs_init root=/dev/mtdblock2 rootfstype=squashfs

So a console should be present on UART.

Seems also that the encrypted Miot traffic is dumped to this log. 👍

ferdydek, 2017-08-16:
@kollaesch simple plastic priming tool, wasn't difficult at all. It left super small wite trace in one place. No biggy.
There was a lose 1cm shrink tube inside (fun fact).

ferdydek, 2017-08-16:
@snoerenberg it may take me a while, my multimeter burned waiting for new one to arrive.

samtap, 2017-08-16:
I've pre-ordered one of these: http://www.gearbest.com/ip-cameras/pp_693217.html
Think there's a good change hardware is the same as in Xiao Fang...

andreq, 2017-08-16:
@samtap this "issue" is focusing on the mijia 1080P, you'll have better chance to create a new one for this specific camera.

Side note, should we all move over to https://github.com/Filipowicz251/mijia-1080P-hacks for further chat about the Mijia?

kollaesch, 2017-08-16:
@ferdydek I got it open. The plastic tube is the cover for the long led-glas-tube. Put it over/around again :)

snoerenberg, 2017-08-16:
@kollaesch maybe you can make a picture how to open it or point out where the clips are.

Thank you

kollaesch, 2017-08-16:
So, guys - thanks for all the previous hints! - I can confirm ... It's alive ;) 

_Referring to the picture last picture in snoerenberg's post_
The top right connection points are responding to: TX,RX (in that order from left to right)

pictures and boot-uart-log follow.

kollaesch, 2017-08-16:
I'm switching to "Filipowicz251/mijia-1080P-hacks" right now. Follow us/me there.

willthrom, 2017-08-17:
@Filipowicz251 close this issue so we don't carry on this discussion here :)

Filipowicz251, 2017-08-17:
You are right :) I'm closing it.

closed 2017-08-17 by Filipowicz251