samtap/fang-hacks #193: Is there any infrared sensor pir500BP?

1043717432, 2017-07-18:
Is there any infrared sensor pir500BP?
[image]
[image]
Join the module, how to achieve normal use?

samtap, 2017-07-18:
Maybe it's hooked up to one of the gpio's? You'll have to figure out yourself but I imagine you could use gpio tools (gpio_ms1, gpio_aud) to read the sensor.

1043717432, 2017-07-22:
Do you know if hack Fang supports this sensor after you add this sensor?

samtap, 2017-07-22:
Nope nobody knows until somebody tries it the first time

MarcoFilius, 2017-11-16:
This will not be possible. At least on the 5.3 rev board as the board layout is different and misses the holes to solder a pir500BP.
[image]

ril3y, 2017-11-16:
I think you would need to trace the pins back from the PIR sensor placement
to the CPU / ad converter / voltage divider etc.  If its wired to the CPU
or other a/d hardware then I fully expect you could get this working.  I am
pretty sure this is just and analog read on a GPIO to the cpu which can all
be setup.

I will see if my cams have it.  I am a it busy on the WYZE firmware right
now however.

Riley

ril3y, 2017-11-16:
The WYZE camera I have does not have the PIR sensor FYI.
WYZEC1 that is.

Riley

MarcoFilius, 2017-11-17:
I havent got a camera with a board as in the initial picture. So, I wouldnt know where to start tracing.
Perhaps when redesigning the newer boards they may even omited the vias.